Screenly/Anthias #1119: Automaticly "End date"

madsrh, 2019-05-16:
This is a feature request / enhancement 

#### Overview of the Issue

When uploading a new asset, the dates (start and end) are by default set to the current date. When changing the "Start date" you get a notification that `end date should be after start date` but the end date is unchanged. It obviously isn't possible to set the end date before the start date, so why not link the two, so the end date always will show the same date as the starte date. 

[image]

It's not much trouble to change the end date for the user, but it would be a nice usability improvement if the two were linked :)

vpetersson, 2019-05-16:
Thanks for reporting!

rusko124, 2019-06-12:
Should be resolved here #1156.
I think we can close this.

closed 2019-06-12 by vpetersson