Screenly/Anthias #1127: connecting to the wifi

yogeshsethi, 2019-05-18:
Hi, 
I am new to this website and also new to raspberry pi.
I have installed a screenly ose on my raspberry pi 2. It has a wifi dongle connected to it.

When i boot the raspberry pi without the wifi dongle and connected to my laptop with a lan cable everything works fine. I am able to access the screenly page to manage the assets.
the problem is when i connect the WIFI dongle and disconnect the lan cable, the raspberry boots to the screen where it says, " no network detected and has opened a access point to connect a laptop or mobile device. It shows the SSID, password and also the screenly.io/wifi to access the wifi settings, first of all the connection is not very stable and when i try to access the screenly.io/wifi page, the resin.io page opens up but it does not displays any SSID or WIFI router name. So i do not know where to go next . Request your help
thanks

rusko124, 2019-05-23:
Hi @yogeshsethi
Discussion with same issue #1098

rusko124, 2019-05-28:
@yogeshsethi 
This problem should be solved in the latest development version, try to upgrade. Please let me know if it helped you.

yogeshsethi, 2019-06-02:
Thanks Rusko,
how do I upgrade?
thanks

vpetersson, 2019-06-02:
See "Manual upgrade" on https://www.screenly.io/ose

rusko124, 2019-06-12:
I think we can close it.
@yogeshsethi If there's any issues reopen please.

closed 2019-06-12 by vpetersson