Screenly/Anthias #1128: Cant start presentation cycle

ShukhratK, 2019-05-20:
Hi,

I cant get into presentation cycle. It is stuck on cycling through the splash screen with instructions pointing to the ip address. It also keeps resetting it and here is the error from balena terminal:
"OCI runtime exec failed: exec failed: cannot exec a container that has stopped: unknown	"

rusko124, 2019-05-23:
Hi @ShukhratK!
Did you try using the latest version of Screenly?(development)

ShukhratK, 2019-05-23:
@rusko124 
I am using the latest version of master.

rusko124, 2019-05-28:
@ShukhratK 
[image]
At the moment, the interface should look something like this.
You can compare the differences or you can provide some screenshots.

ShukhratK, 2019-05-28:
[image]

ShukhratK, 2019-05-28:
28.05.19 08:06:41 (-0400)  screenly-viewer    RAN: '/usr/bin/git rev-parse --abbrev-ref HEAD'
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDOUT:
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDERR:
28.05.19 08:06:41 (-0400)  screenly-viewer  fatal: Not a git repository (or any of the parent directories): .git
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:42 (-0400) Service exited 'screenly-viewer sha256:5cec82650016f63b85b04256ac75180dc5970341fd75721452ffd95a261228de'
28.05.19 08:06:43 (-0400) Restarting service 'screenly-viewer sha256:5cec82650016f63b85b04256ac75180dc5970341fd75721452ffd95a261228de'
28.05.19 08:06:41 (-0400)  screenly-viewer  Viewer crashed.
28.05.19 08:06:41 (-0400)  screenly-viewer  Traceback (most recent call last):
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 551, in <module>
28.05.19 08:06:41 (-0400)  screenly-viewer      main()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 546, in main
28.05.19 08:06:41 (-0400)  screenly-viewer      asset_loop(scheduler)
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 421, in asset_loop
28.05.19 08:06:41 (-0400)  screenly-viewer      check_update()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 375, in check_update
28.05.19 08:06:41 (-0400)  screenly-viewer      git_branch = sh.git('rev-parse', '--abbrev-ref', 'HEAD').strip()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 726, in __call__
28.05.19 08:06:41 (-0400)  screenly-viewer      return RunningCommand(cmd, call_args, stdin, stdout, stderr)
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 291, in __init__
28.05.19 08:06:41 (-0400)  screenly-viewer      self.wait()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
28.05.19 08:06:41 (-0400)  screenly-viewer      self._handle_exit_code(self.process.wait())
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
28.05.19 08:06:41 (-0400)  screenly-viewer      self.process.stderr
28.05.19 08:06:41 (-0400)  screenly-viewer  ErrorReturnCode_128: 
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    RAN: '/usr/bin/git rev-parse --abbrev-ref HEAD'
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDOUT:
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDERR:
28.05.19 08:06:41 (-0400)  screenly-viewer  fatal: Not a git repository (or any of the parent directories): .git
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer  Traceback (most recent call last):
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 551, in <module>
28.05.19 08:06:41 (-0400)  screenly-viewer      main()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 546, in main
28.05.19 08:06:41 (-0400)  screenly-viewer      asset_loop(scheduler)
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 421, in asset_loop
28.05.19 08:06:41 (-0400)  screenly-viewer      check_update()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/data/screenly/viewer.py", line 375, in check_update
28.05.19 08:06:41 (-0400)  screenly-viewer      git_branch = sh.git('rev-parse', '--abbrev-ref', 'HEAD').strip()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 726, in __call__
28.05.19 08:06:41 (-0400)  screenly-viewer      return RunningCommand(cmd, call_args, stdin, stdout, stderr)
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 291, in __init__
28.05.19 08:06:41 (-0400)  screenly-viewer      self.wait()
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
28.05.19 08:06:41 (-0400)  screenly-viewer      self._handle_exit_code(self.process.wait())
28.05.19 08:06:41 (-0400)  screenly-viewer    File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
28.05.19 08:06:41 (-0400)  screenly-viewer      self.process.stderr
28.05.19 08:06:41 (-0400)  screenly-viewer  sh.ErrorReturnCode_128: 
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    RAN: '/usr/bin/git rev-parse --abbrev-ref HEAD'
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDOUT:
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:41 (-0400)  screenly-viewer    STDERR:
28.05.19 08:06:41 (-0400)  screenly-viewer  fatal: Not a git repository (or any of the parent directories): .git
28.05.19 08:06:41 (-0400)  screenly-viewer  
28.05.19 08:06:43 (-0400)  screenly-viewer  Still continue...
[... cut ...]
28.05.19 08:06:48 (-0400)  screenly-viewer  /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
28.05.19 08:06:48 (-0400)  screenly-viewer    utils.CryptographyDeprecationWarning
28.05.19 08:06:49 (-0400)  screenly-viewer  Loading browser...
28.05.19 08:06:49 (-0400)  screenly-viewer  Browser loading None. Running as PID 43.
28.05.19 08:06:50 (-0400)  screenly-viewer  current url is http://127.0.0.1:8080/splash_pag

rusko124, 2019-05-28:
Hmm, judging by the logs this should have been fixed in the last commits(#1133), try to update again, I think this will solve your problem.

ShukhratK, 2019-05-28:
@rusko124 thanks, will give that a try.

ShukhratK, 2019-05-28:
@rusko124 it's working now. thx. having new issues with it randomly restarting but i will search for that through the threads.

rusko124, 2019-06-10:
Can we close this?

closed 2019-06-10 by vpetersson