Screenly/Anthias #1129: screenly-viewer.service won't run

Steffen75, 2019-05-21:
Today I upgraded the packages of my raspi running Screenly. After the reboot Screenly got stuck at the splash screen with screenly-viewer.service in the lower left corner.

Ok, the installation was rather old. So why not do a clean install?

I made a fresh SD card with the latest lite image from the Raspberry Pi homepage, applied all pending updates and then installed Screenly again using the provided bash script.
I chose the stable version of Screenly and the script ran through fine. But after the reboot Screenly got stuck at screenly-viewer.service again.

When checking the System Info page of Screenly you can see that the viewer tries to start every about every 5 seconds but somehow fails.

When I checked the status of the screenly-viewer.service using systemctl I once got the following error:

> /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!

What went wrong here? A problem with Python?

rusko124, 2019-05-21:
Hi @Steffen75 !
I currently tried to reproduce your issue.
Device: Raspberry Pi 3b+
Steps:
1.Fresh SD card with the latest image https://github.com/Screenly/screenly-ose/releases/tag/0.17
2.Run upgrade to development version w/out a full system upgrade.
3.Made sure everything was working.
4.Run upgrade to development version with a full system upgrade.
5.Again made sure everything was working.
In my case I haven't issue which you described.

Could you provide the steps in more detail, or provide any other information please?

Steffen75, 2019-05-21:
1. I made a fresh SD card with the lastet image from here: https://downloads.raspberrypi.org/raspbian_lite_latest (April 8th 2019)
2. Made all the normal adjustments afterwards (enlarge file system, set keyboard and locale to Germany, enable SSH, setup wifi, change hostname and give it a fixed IP).
3. Reboot
4. apt-get update && apt-get upgrade
5. Reboot again
6. bash <(curl -sL https://www.screenly.io/install-ose.sh)
7. y/n/n/n/y

And it hangs at the screenly-viewer.service as mentioned above.

Idan2cohen, 2019-05-21:
Same over here, It's stuck on 

> 1. I made a fresh SD card with the lastet image from here: https://downloads.raspberrypi.org/raspbian_lite_latest (April 8th 2019)
> 2. Made all the normal adjustments afterwards (enlarge file system, [...]

I had the same problem,
After i did everything you said, the splash screen disappear and now the screen became black. I can get into the web but nothing is displayed on the monitor.

rusko124, 2019-05-21:
Got it, I will try to reproduce this problem once again and resolve the issue ASAP.

rusko124, 2019-05-22:
Okay, i could reproduce it. 
https://github.com/Screenly/screenly-ose/issues/1102#issuecomment-490717071
I can offer you the following solutions:
1. Manually install network-manager
2. Update to development version(by default don't install network-manager, but you can not use Balena network configuration)
3. You could made a fresh SD card with the lastet Screenly image https://github.com/Screenly/screenly-ose/releases/download/0.17/2019-04-27-Screenly-OSE.zip

bitos2002, 2019-05-23:
Will be possible resolve the problem for OSE instalation on Balena.

Thanks !!!!

vpetersson, 2019-05-24:
> Will be possible resolve the problem for OSE instalation on Balena.
> I had used de git repository to upload services to Balena. Will be resolved on githaub repository?
> https://github.com/Screenly/screenly-ose.git
> That will resolve the problem on device in Balena?
> 
> Thanks !!!!

Please move this to another issue as it is unrelated to this issue, but in short yes, we've already fixed the Balena support.

Steffen75, 2019-05-31:
>     3. You could made a fresh SD card with the lastet Screenly image https://github.com/Screenly/screenly-ose/releases/download/0.17/2019-04-27-Screenly-OSE.zip

Used this option.

closed 2019-05-31 by Steffen75