Screenly/Anthias #1134: OpenGL Driver on experimental?

danpoltawski, 2019-05-24:
Is it possible to use Screenly-OSE on experimental with the opengl drivers? 

I'd really like to see if with the experimental drivers, a page we want to display with some CSS animation would render..

### Environment

* Raspberry Pi Hardware Version: Pi 3 Model B+
* Screenly OSE Version: OSE experimental

vpetersson, 2019-05-24:
My understanding is that we're already using hardware accelleration in the experimental branch (at least to the extent supported by QT). I don't know if that translates to full OpeNGL support or not.

danpoltawski, 2019-05-28:
When I switch to the opengl driver with raspi-config, the system boots up into a black screen - just trying to debug how to work out where the problem is

vpetersson, 2019-05-28:
Interesting. Could potentially be some docker flag or `/dev/` mapping we need to make.

danpoltawski, 2019-05-28:
I'm trying to understand the architecture in order to try and debug - is Xorg running in the viewer container?

vpetersson, 2019-05-28:
It's actually not using xorg. It's a QT app, so it's talking directly to frame buffer.

closed 2019-08-03 by stale[bot]