Screenly/Anthias #1137: API viewer_current_asset not updating after asset switch through API

burnnat, 2019-05-29:
#### Overview of the Issue

When using the `/assets/control` API to switch to an asset by ID, the displayed asset switches but is not reflected in later API calls to `viewer_current_asset`. I would expect the call to return the asset that was just switched to, but it appears to return the same as was previously displayed. Note that in my testing, the `next` and `previous` operations did appear to update the `viewer_current_asset` properly, so this seems to be specific to the `asset&asset_id` case.

#### Reproduction Steps

* Switch to a particular out-of-order asset via API call to `/api/v1/assets/control/asset&asset_id`.
* Wait for the display to update.
* Make an API call to `/api/v1/viewer_current_asset` and note the returned asset details do not match the displayed asset.

### Environment

* Raspberry Pi Hardware Version: Zero W
* Screenly OSE Version: 0.17 (production@618864d)

vpetersson, 2019-05-29:
@burnnat Fix landed. Please try with the master branch.

burnnat, 2019-06-01:
I've confirmed, the API is working as expected in master now. Thanks for the quick fix!

closed 2019-06-01 by burnnat