Screenly/Anthias #1140: Random restarting

ShukhratK, 2019-05-30:
After the last update, the whole thing started restarting at random intervals. Here is the balena log:

30.05.19 10:56:40 (-0400)  screenly-viewer  Sleeping for 4
30.05.19 10:56:44 (-0400)  screenly-viewer  Last update: 2019-05-29 17:10:14.649406
30.05.19 10:56:45 (-0400)  screenly-viewer  get_next_asset
30.05.19 10:56:45 (-0400)  screenly-viewer  refresh_playlist
30.05.19 10:56:45 (-0400)  screenly-viewer  refresh: counter: (1) deadline (2019-05-28 19:47:00) timecur (2019-05-30 14:56:45.164532)
30.05.19 10:56:45 (-0400)  screenly-viewer  update_playlist
30.05.19 10:56:45 (-0400)  screenly-viewer  Generating asset-list...
30.05.19 10:56:45 (-0400)  screenly-viewer  generate_asset_list deadline: 2019-05-28 19:47:00
30.05.19 10:56:45 (-0400)  screenly-viewer  update_playlist done, count 4, counter 0, index 0, deadline 2019-05-28 19:47:00
30.05.19 10:56:45 (-0400)  screenly-viewer  get_next_asset after refresh
30.05.19 10:56:45 (-0400)  screenly-viewer  get_next_asset counter 0 returning asset 1 of 4
30.05.19 10:56:45 (-0400)  screenly-viewer  Showing asset SignPro Solar.mp4 (video)
30.05.19 10:56:45 (-0400)  screenly-viewer  Asset URI /data/screenly_assets/948dc79749ab4cfa816c02bf4ae25234
30.05.19 10:56:45 (-0400)  screenly-viewer  Displaying video /data/screenly_assets/948dc79749ab4cfa816c02bf4ae25234 for 64 
30.05.19 10:56:46 (-0400)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
30.05.19 10:57:54 (-0400)  screenly-viewer  Last update: 2019-05-29 17:10:14.649406
30.05.19 10:57:54 (-0400)  screenly-viewer  get_next_asset
30.05.19 10:57:54 (-0400)  screenly-viewer  refresh_playlist
30.05.19 10:57:54 (-0400)  screenly-viewer  refresh: counter: (0) deadline (2019-05-28 19:47:00) timecur (2019-05-30 14:57:54.867811)
30.05.19 10:57:54 (-0400)  screenly-viewer  update_playlist
30.05.19 10:57:54 (-0400)  screenly-viewer  Generating asset-list...
30.05.19 10:57:54 (-0400)  screenly-viewer  generate_asset_list deadline: 2019-05-28 19:47:00
30.05.19 10:57:54 (-0400)  screenly-viewer  update_playlist done, count 4, counter 0, index 1, deadline 2019-05-28 19:47:00
30.05.19 10:57:54 (-0400)  screenly-viewer  get_next_asset after refresh
30.05.19 10:57:54 (-0400)  screenly-viewer  get_next_asset counter 0 returning asset 2 of 4
30.05.19 10:57:54 (-0400)  screenly-viewer  Showing asset MakingSignLetters_IMG_3230.jpg (image)
30.05.19 10:57:54 (-0400)  screenly-viewer  Asset URI /data/screenly_assets/bb0d858c77b44656b7fd044b40e16aa5
30.05.19 10:57:54 (-0400)  screenly-viewer  Already showing file:///tmp/screenly_html/black_page.html, reloading it.
30.05.19 10:57:54 (-0400)  screenly-viewer  Sleeping for 4
30.05.19 10:57:58 (-0400)  screenly-viewer  Last update: 2019-05-29 17:10:14.649406
30.05.19 10:57:59 (-0400)  screenly-viewer  get_next_asset
30.05.19 10:57:59 (-0400)  screenly-viewer  refresh_playlist
30.05.19 10:57:59 (-0400)  screenly-viewer  refresh: counter: (0) deadline (2019-05-28 19:47:00) timecur (2019-05-30 14:57:59.623657)
30.05.19 10:57:59 (-0400)  screenly-viewer  update_playlist
30.05.19 10:57:59 (-0400)  screenly-viewer  Generating asset-list...
30.05.19 10:57:59 (-0400)  screenly-viewer  generate_asset_list deadline: 2019-05-28 19:47:00
30.05.19 10:57:59 (-0400)  screenly-viewer  update_playlist done, count 4, counter 0, index 2, deadline 2019-05-28 19:47:00
30.05.19 10:57:59 (-0400)  screenly-viewer  get_next_asset after refresh
30.05.19 10:57:59 (-0400)  screenly-viewer  get_next_asset counter 0 returning asset 3 of 4
30.05.19 10:57:59 (-0400)  screenly-viewer  Showing asset NewSign190508b029.jpg (image)
30.05.19 10:57:59 (-0400)  screenly-viewer  Asset URI /data/screenly_assets/3813986d7a50413f957b0abf1609e63d
30.05.19 10:57:59 (-0400)  screenly-viewer  Already showing file:///tmp/screenly_html/black_page.html, reloading it.
30.05.19 10:57:59 (-0400)  screenly-viewer  Sleeping for 4
30.05.19 10:59:07 (-0400) Supervisor starting
30.05.19 10:58:55 (-0400)  screenly-viewer  Still continue...
30.05.19 10:58:57 (-0400)  screenly-viewer  Still continue...
30.05.19 10:58:58 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:02 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:04 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:05 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:06 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:07 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:08 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:09 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:10 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:10 (-0400)  screenly-server  /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
30.05.19 10:59:10 (-0400)  screenly-server    RequestsDependencyWarning)
30.05.19 10:59:10 (-0400)  screenly-server  /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
30.05.19 10:59:10 (-0400)  screenly-server    utils.CryptographyDeprecationWarning
30.05.19 10:59:11 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:12 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:13 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:14 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:15 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:16 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:17 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:18 (-0400)  screenly-viewer  Still continue...
30.05.19 10:59:19 (-0400)  screenly-viewer  Still continue...
[... 255 lines cut ...]
30.05.19 11:07:33 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:34 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:35 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:36 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:37 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:38 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:39 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:40 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:41 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:42 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:43 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:44 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:45 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:46 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:47 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:49 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:50 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:51 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:52 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:53 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:54 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:55 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:56 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:57 (-0400)  screenly-viewer  Still continue...
30.05.19 11:07:58 (-0400)  screenly-viewer  Still continue...

vpetersson, 2019-05-31:
@rusko124 any clues here?

rusko124, 2019-05-31:
It's weird because Balena restarts all services at once!. In my case, the latest version works for several days without restarts. 
Need more information to I can reproduce it.
Perhaps something causes this behavior in the settings of the device(Balena) .

ShukhratK, 2019-05-31:
@rusko124 do you want a screenshot of the fleet configuration?

rusko124, 2019-06-05:
@ShukhratK yes, please

ShukhratK, 2019-06-05:
@rusko124 
[image]

rusko124, 2019-06-10:
@ShukhratK Have you used 192MB gpu memory for special occasions?
Screenly enough 64MB.

ShukhratK, 2019-06-10:
@rusko124 I think i was following some tutorial on balena where they mentioned to make that change. should i set it back to 64?

rusko124, 2019-06-11:
@ShukhratK  Yes, please.

rusko124, 2019-06-12:
Any progress here?

ShukhratK, 2019-06-12:
@rusko124 i adjusted the memory to 64. still restarting. longest it lasted was 18 hours.

ShukhratK, 2019-06-12:
@rusko124 actually the output screen went completely blank about 20 minutes ago. balena showing the device as online however logs are not updating. i can go into the terminal via balena's dashboard. this happened after about 19 hours of being online. 
i don't know if it makes a difference or not but it is on development mode in balena.

rusko124, 2019-06-13:
@ShukhratK I'm a little confused. What device do you have and what Balena image installed?

ShukhratK, 2019-06-13:
@rusko124 i have a raspberry pi 3 b+ and using balena image for raspberry pi 3 development. (balenaOS 2.32.0+rev1
)

ShukhratK, 2019-06-13:
by the way: still black screen showing on tv connected to raspberry pi but balena is showing it being online for 2 days. here is the last logs available:

12.06.19 00:43:47 (-0400)  screenly-viewer  Already showing file:///tmp/screenly_html/black_page.html, reloading it.
12.06.19 12:08:46 (-0400)  screenly-server  sudo: unable to resolve host fe32e6c
12.06.19 12:08:46 (-0400)  screenly-server  Failed to get D-Bus connection: Unknown error -1

rusko124, 2019-06-14:
```
12.06.19 12:08:46 (-0400) screenly-server sudo: unable to resolve host fe32e6c
12.06.19 12:08:46 (-0400) screenly-server Failed to get D-Bus connection: Unknown error -1
```
This error is related to the system information page on Balena. But this should not be related to your issue.

ShukhratK, 2019-06-14:
@rusko124 That's what I figured. I know they had issues with their service a few days ago.

ShukhratK, 2019-06-24:
https://github.com/Screenly/screenly-ose/issues/1163 this threads sounds similar to mine. do you think it could be the video file?

rusko124, 2019-08-30:
@ShukhratK I finally caught the problem. It's not really related to Screenly. The solution in my case was to change the SD card.

closed 2019-11-05 by stale[bot]