Screenly/Anthias #1151: Unable to run tvservices command

singh87, 2019-06-10:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

I am trying to setup Screenly with balena. The problem is that HDMI signal is not showing up on the display. I want to be able to get the EDID information of the connected monitor/TV

I thought I would just be able to run the tvservices commands in the balena host os terminal but the command was not found.

Any help would be appreciated

vpetersson, 2019-06-11:
@singh87 We've been looking to add support for `tvservice` on Balena, but are yet to get there (as we had more pressing issues to deal with). 

For the time being, I would suggest you use the regular disk image to make sure everything works (as it includes `tvservice`).

closed 2019-08-17 by stale[bot]

altontoth, 2019-10-08:
So,I just updated my install of Screenly,and I assume Docker causes a similar issue? I'm getting the exact same problem on one device, and it's the only one I've applied an update to as of today.

vpetersson, 2019-10-09:
@altontoth Yes, i believe the dockerized version broke `tvservice` for now. It should be possible to work around this in the future.