Screenly/Anthias #1152: Upgrading to Experimental Branch partially fails

coderforlife, 2019-06-10:
Installed most recent OSE image (from April I believe). Then upgraded after setting up some assets since UBZL was doing very poorly. Then startup would be stuck on screenly-web.service. SSHing in showed that screenly-viewer was failing to start. A little more investigation showed that dbus was being imported and failing. So I did

```
sudo apt-get install python-dbus
```

Then doing a `python viewer.py` in the screenly folder caused a `CommandNotFound` exception when trying to do `sh.Command('ScreenlyWebview')`.

Seems like the viewer is still the old viewer and half-uninstalled...

coderforlife, 2019-06-10:
Looking further back, during the upgrade process the following error showed up, likely causing this partial upgrade:

```
TASK [screenly : Run database migration] *******************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:00.301803", "end": "2019-06-10 17:57:56.040121", "msg": "non-zero return code", "rc": 1, "start": "2019-06-10 17:57:55.738318", "stderr": "  File \"/home/pi/screenly/bin/migrate.py\", line 75\n    print 'Column (' + col + ') already present'\n                   ^\nSyntaxError: invalid syntax", "stderr_lines": ["  File \"/home/pi/screenly/bin/migrate.py\", line 75", "    print 'Column (' + col + ') already present'", "                   ^", "SyntaxError: invalid syntax"], "stdout": "", "stdout_lines": []}
```

Seems like the `screenly/bin/migrate.py` has a Python syntax error however when I look at the line in question there is nothing wrong with it for Python 2.7 syntax but it is incorrect syntax in Python 3 and is the first such line like that. However, doing `/usr/bin/env python` like the she-bang line says gives me a Python 2.7.13 terminal.

vpetersson, 2019-06-11:
@coderforlife Screenly OSE is still on Python 2.7. We are intending to upgrade to Python 3, but we don't have any ETA for this right now. Hence, stick to python 2.7 and you should be good.

closed 2019-08-17 by stale[bot]