Screenly/Anthias #1155: Add "Integration tab" extension for WoTT

vpetersson, 2019-06-11:
We need to add a new section to the "Integration tab" if WoTT is detected and installed and populate it with the "Device ID" from WoTT.

eugenosm, 2019-06-17:
jut to be sure. 
only that neede in WoTT extension?
[image]
or something else to add?

And one else question.  When start it in docker there is no sudo. And cause of:

```python
def run(ping=True, debug=False, dev=False):
    with Locker('ping'):
```

All actions need to run wott-agent as sudo. 
and as soon as i get WoTT Device ID like that:
``` python
result = os.popen('sudo wott-agent whoami').read()
...
```
this is not work in docker.

So what about balena? Is there sudo?

a-martynovich, 2019-06-18:
@eugenosm Because our device id is stored in certificate which is in a protected folder `get_device_id` would still fail even after you removed Locker.

You can run bash in Docker as root like this:
`docker exec --user=root -ti screenly-ose bash`.

However Screenly is running as user pi, and we shouldn't change this. This user shouldn't have sudo privileges either.

Instead we should think of a way to interact with agent without superuser (but only for the publicly available data).

eugenosm, 2019-06-18:
> You can run bash in Docker as root like this: ...

yes of course, but i mean that when i run this:
``` python
result = os.popen('sudo wott-agent whoami').read()
```
in docker it receives empty string and smoething like `sudo not found`  in stderr

a-martynovich, 2019-06-18:
@eugenosm In our recent discussion we've decided to make agent certs publicly readable. Then you can get device id this way:
```
import agent
id = agent.get_device_id()
```

You can do this because agent and Screenly are installed on the same machine and should both be available for import.

eugenosm, 2019-06-19:
so, as we disscuss in slack. there different python versions.

so i return to shell wott-agent execution. and in raspberry hardware all works fine.
but when in docker it did not find sudo and send to stderr this:
`sh: 1: sudo: not found`

if try to run without sudo,  then rises an exception about that `PermissionError: [Errno 13] Permission denied: '/opt/wott/ping-wott.lock'`

so:
would it be run in docker? (for dev we can mount external storage with modified rights)
wouldn't the same troubles with balena?

PR #1166 (2019-06-21): add integration tab extension for wott
closes #1155
closed 2019-06-21 by vpetersson