Screenly/Anthias #1161: How do I "test" my changes in development

w1nterl0ng, 2019-06-12:
I am working on some API changes, and I understand the server side, I am able to load the docker container, make changes and see them on the server.  However, I am not sure how to see my changes in a "viewer" short of pushing changes from every "save" to the actual raspberry pi itself.
I looked in the wiki and did not see anything about setting up a development environment.
How are other developers managing their environment?

vpetersson, 2019-06-12:
@w1nterl0ng It largely depends on what you're working on. If you're working on strict back-end code, then the docker container is usually fine. If it's front-end changes however, it usually requires a Pi. That said, in *theory* it is possible to run everything on x86 (on Linux), but it is largely untested.

w1nterl0ng, 2019-06-12:
I am working on extending the functionality of /assets/control  I will attempt this Linux, if that fails I'll just write a script to deploy files to the Pi.  Thanks.

closed 2019-08-18 by stale[bot]