Screenly/Anthias #1164: seamless loop

albanoalfredo, 2019-06-16:
#### Lack of seamless loop

I've used screenly in the past for digital signage for playing a video playlist with seamless loop and it was de perfect solution.

Now, everytime the video ends the player goes to black before starting again.
What happened in these newer versions?

How can I optimize for a seamless loop.

##### hardware

raspberry pi 3 b+

##### video: 

mp4 h264 low bit rate (2 to 6 mbps)

tried with different rpi, different cards and several videos.

vpetersson, 2019-06-17:
@albanoalfredo There has always been a gap there. I know some contributor (can't find the ticket) was working on reducing this for this particular edge case, but I don't think it's in master yet. In short, `viewer.py` needs to be patched to pass on custom parameters to `omxplayer` such that it simply loops a single video. The black screen that you are seeing is from `omxplayer` loading.

albanoalfredo, 2019-06-17:
thanks for your help.

[adafruit's video looper](https://github.com/adafruit/pi_video_looper) also uses omxplayer and there's no lag, the loop is seamsless. Can we somehow use the same method they use?

vpetersson, 2019-06-17:
Technically speaking, it's not super hard. There's a built-in functionality for it in `omxplayer`. It just requires an update to the logic in `viewer.py`.

closed 2019-08-23 by stale[bot]

ealmonte32, 2019-09-24:
@albanoalfredo 
maybe if you still need to do this seamless loop you can try my workaround that works perfect for me so far, edit 2 lines of code:
https://forums.screenly.io/t/looping-video-for-a-virtual-fireplace/145/11

nkle, 2020-01-20:
> @albanoalfredo
> maybe if you still need to do this seamless loop you can try my workaround that works perfect for me so far, edit 2 lines of code:
> https://forums.screenly.io/t/looping-video-for-a-virtual-fireplace/145/11

Thank you so much! This worked perfectly for me.

unique311, 2020-03-03:
Was a fix ever found for the gapless playback?
I gave the code above a shot .. but it won't play multiple assets automatically,.  It just loops one asset.

vpetersson, 2020-03-03:
If you're talking about proper gap-less playback for videos, that will likely never happen, due to technical limitations in the software used. It is however something that we already support in the commercial version.