Screenly/Anthias #1167: I have been trying to use screenlyOSE on balena cloud . After ive successfully performed balena push, and got the unicorn image, i have been waiting for a really long time but screenly doesnt appear in my page. I tried to access my public URL and it gives off an error that “Device does not have registered vpn tunnel endpoint” ive been following the tutorial from this link: https://www.balena.io/blog/deploy-free-digital-signage-software-screenly-ose/

car-thee-ca, 2019-06-21:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:

car-thee-ca, 2019-06-21:
[image]
This is the exact error message that i am receiving.

I did get build finished. I unzipped the github screenly-ose-master and performed balena push. Ive been waiting ever since yesterday evening but the screenly page does not appear on the monitor connected to my raspberry pi
[image]

vpetersson, 2019-06-21:
@remingtona - can you see if you can reproduce this?

RemingtonA, 2019-06-21:
Can do!

novasea, 2019-06-26:
I got the exact same issue. When I check the balena device trying to run screenly, through the balena web interface, the logs says it cant find some command, and that there's a syntax error near unexpected token.

`26.06.19 11:19:02 (+0200)  screenly-server  /tmp/screenly/bin/start_balena.sh: line 2: $'\r': command not found
26.06.19 11:19:02 (+0200)  screenly-server  /tmp/screenly/bin/start_balena.sh: line 3: syntax error near unexpected token `$'{\r''
'6.06.19 11:19:02 (+0200)  screenly-server  /tmp/screenly/bin/start_balena.sh: line 3: `run_setup () {
26.06.19 11:19:03 (+0200) Service exited 'screenly-server sha256:ca1430a370504fa7e178f951562e8490d5bb6ab5dc08c77ddd1f1854db666128'
26.06.19 11:19:14 (+0200) Restarting service 'screenly-viewer sha256:d60e6aed47c824e8917b0cc0bf3d088d0c909c772678c9cca79568d4d06cd8c4'
26.06.19 11:19:14 (+0200)  screenly-viewer  /tmp/screenly/bin/start_balena.sh: line 2: $'\r': command not found
26.06.19 11:19:14 (+0200)  screenly-viewer  /tmp/screenly/bin/start_balena.sh: line 3: syntax error near unexpected token `$'{\r''
'6.06.19 11:19:14 (+0200)  screenly-viewer  /tmp/screenly/bin/start_balena.sh: line 3: `run_setup () {
26.06.19 11:19:14 (+0200)  screenly-websocket  /tmp/screenly/bin/start_balena.sh: line 2: $'\r': command not found
26.06.19 11:19:14 (+0200)  screenly-websocket  /tmp/screenly/bin/start_balena.sh: line 3: syntax error near unexpected token `$'{\r''
'6.06.19 11:19:14 (+0200)  screenly-websocket  /tmp/screenly/bin/start_balena.sh: line 3: `run_setup () {`

vpetersson, 2019-06-26:
@novasea I'm not sure that's the same error. That looks very much like a Windows formatting error (i.e. a file edited on Windows, that broke the newlines).

vpetersson, 2019-06-26:
@RemingtonA were you able to reproduce this?

rusko124, 2019-07-01:
[image]
[image]

Works good for me, `9bdd4e8` release version

homerosbart2, 2019-07-10:
> [image]
> [image]
> 
> Works good for me, `9bdd4e8` release version

**Is this in Windows?**

rusko124, 2019-07-16:
> > [image]
> > [image]
> > Works good for me, `9bdd4e8` release version
> 
> **Is this in Windows?**

Ubuntu 18, but I also reproduced all steps on Windows 8.
[image]

closed 2019-09-21 by stale[bot]