Screenly/Anthias #1170: Audio in URL is not playing

y4nn1ckx, 2019-06-27:
#### Overview of the Issue

A web page using the <audio> tag or the <embed> tag does not play audio files. When using the <audio> tag, the player is displayed and act as if it was playing but the sound does not come out on the 3.5mm jack. 

I did configure sound to point to 3.5mm jack in the Screenly settings and if I upload a video to the Raspberry Pi, the sounds plays well. But not on URLs. 

### Environment

* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3 Ram: 1 GB Sony UK
* Screenly OSE Version: production@618864d from image 2019-04-27-Screenly-OSE.zip

vpetersson, 2019-06-28:
@y4nn1ckx I don't know if the browser used actually supports audio. Could you try upgrading to the experimental version, as it is using a different browser.

y4nn1ckx, 2019-07-02:
Is there an official way to upgrade my Sprint9 image to experimental?  

Tried using the run_upgrade.sh file from /home/pi/screenly/bin  but after that I get stuck at Screenly loading screen written "user@1000.service"

vpetersson, 2019-07-02:
@y4nn1ckx sounds like a broken SD card actually. Try replacing it. Random freezes and odd behavior tend to come from that.

y4nn1ckx, 2019-07-02:
Tried another MicroSD card, results are similar. Right now, it is stuck at Screenly loading screen written "screenly-web.service"

I didn't do any changes to the config of the Sprint9 image after initial imaging process except enabling SSH. 

During the run_upgrade.sh, I did get the following error

TASK [screenly : Run database migration] **********************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:00.153941", "end": "2019-07-02 16:38:48.040442", "msg": "non-zero return code", "rc": 1, "start": "2019-07-02 16:38:47.886501", "stderr": "Traceback (most recent call last):\n  File \"/home/pi/screenly/bin/migrate.py\", line 163, in <module>\n    migrate_drop_filename()\n  File \"/home/pi/screenly/bin/migrate.py\", line 153, in migrate_drop_filename\n    with open_db_get_cursor() as (cursor, _):\n  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__\n    return self.gen.next()\n  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor\n    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:\nsqlite3.OperationalError: unable to open database file", "stderr_lines": ["Traceback (most recent call last):", "  File \"/home/pi/screenly/bin/migrate.py\", line 163, in <module>", "    migrate_drop_filename()", "  File \"/home/pi/screenly/bin/migrate.py\", line 153, in migrate_drop_filename", "    with open_db_get_cursor() as (cursor, _):", "  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__", "    return self.gen.next()", "  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor", "    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:", "sqlite3.OperationalError: unable to open database file"], "stdout": "", "stdout_lines": []}

PLAY RECAP ****************************************************************************************************************************************************
localhost                  : ok=36   changed=15   unreachable=0    failed=1    skipped=7    rescued=0    ignored=0

vpetersson, 2019-07-02:
This is very odd, but that still looks like another SD card failure. These things are fragile and I had a number of them break over the years. Here's what I recommend:

 * Re-download the disk image (and check the MD5 sum)
 * Use [SD Formatter](https://www.sdcard.org/downloads/formatter/) to perform an Overwrite format
 * Reflash the disk image using [Etcher](https://www.balena.io/etcher/)  (since it does a verification)

y4nn1ckx, 2019-07-02:
I will be trying this shortly. 

In the mean time, my issue regarding the upgrade to experimental looks very similar to https://github.com/Screenly/screenly-ose/issues/1152

y4nn1ckx, 2019-07-02:
Redid it all, the upgrade went fine this time. (I suspect that run_upgrade.sh should NOT be run using sudo?) 

Regarding the main topic, my attempts to have audio on a webpage so far have not worked. I tried the EMBED and AUDIO html tag and none of them worked. Will continue testing.

vpetersson, 2019-07-03:
> I suspect that run_upgrade.sh should NOT be run using sudo?

Correct. That will mess up the permission. `sudo` is invoked in the script when needed.

closed 2019-09-08 by stale[bot]