Screenly/Anthias #1171: Hello,

bitos2002, 2019-06-28:
#### Overview of the Issue
 
 Screenly-OSE and wifi-connect
 
#### Reproduction Steps

Hello,
I want to use screenly-ose on balena container, but I want to use wifi-connect.
Is wifi-connect include in screenly-ose last version.
I have tried to merge on one container screenly-ose an wifi-container, but i have not exit on it.
Anyone has done this and can help me or share the project.

Thanks !!!!

rusko124, 2019-07-12:
Hey @bitos2002 !
I don't think it's that obvious. A possible solution can be found here I think https://www.balena.io/docs/reference/cli/

madsrh, 2019-08-06:
Can we close this issue?

vpetersson, 2019-08-07:
@madsrh Yeah i think so. 

Closing this out.

closed 2019-08-07 by vpetersson