Screenly/Anthias #1172: How to use web server inside screenly for use personal satic web page

bitos2002, 2019-06-28:
Hello,
Anyone can say me the path inside screenly directories to put a personal static web page (index.html) , and use it in screenly schedule app. as a local resource.
Thanks

vpetersson, 2019-06-28:
You can't because it's an app server, not a web server. You'd have to re-configure Nginx for this.

bitos2002, 2019-06-29:
Thank you!!!
Can you say me the path of .conf file for configure nginx on another port for no disturb the other services, and how can configure for automatic star on init.
Thanks !!!

vpetersson, 2019-06-29:
Nginx is already enabled and running as a reverse proxy.

You'd have to add your own config in `/etc/nginx/sites-enabled/`, but this is not Screenly specific. The config Screenly uses looks like [this](https://github.com/Screenly/screenly-ose/blob/master/ansible/roles/nginx/files/nginx.conf).

closed 2019-09-04 by stale[bot]

simmondp, 2020-09-30:
OK, has the same issue, here is how I solved it:

Running a lightweight webserver on the Raspberry Pi (lighttpd)
==============================================================
       Using PUTTY get a terminal window in Screenly OSE

	Refresh database of available packages
	=========================
	Ensure that the package manager has up-to-date information about which packages are available:
	* sudo apt update

	Installing lighttpd
	===================

	To install the lighttpd web server issue the command.
	* sudo apt install lighttpd

	This will install the web server and also pull in any other packages (called dependencies) that are required. 
	The server will be automatically started and set to start by default after a reboot.
	[ ok ] Starting web server: lighttpd.

	Change the default port number of Lighttpd to 82 (from 80) so as not to conflict with Screenly management interface

	Check the version of lighttpd as follows:
	* lighttpd -version

	Check the status of lighttpd service
	* systemctl status lighttpd

	Check the listening port number of lighttpd using netstat
	* netstat -tulpn | grep lighttpd

	Edit the configuration file of lighttpd
	* sudo nano /etc/lighttpd/lighttpd.conf
		edit server port = 80 to 82
		edit serve.document-root = "/var/www2/html"
		Save <crtl>O and exit <ctrl>X

	Create, then set permissions on the web directory /var/www2/
	* cd /var
	* sudo mkdir www2
	* cd www2
	* sudo mkdir html

	Change the directory owner and group
	* sudo chown www-data:www-data /var/www2
	allow the group to write to the directory
	* sudo chmod 775 /var/www2
	* sudo chmod 777 /var/www2/html
	Add the pi user to the www-data group
	* sudo usermod -a -G www-data pi

	Copy the default file from 
	* sudo cp /var/www/html/index.lighttpd.html /var/www2/html/index.lighttpd.html

	Restart the service of Lighttpd as follows
	* sudo systemctl restart lighttpd.service

	Verify the listening port number of lighttpd
	* sudo netstat -tulpn | grep lighttpd

	Now go to the web root for html pages
	* cd /var/www2/html
	* ls (you should see the index.lighttpd.html file)

	Now upload the static pages etc to the PI
	* FTP to 192.168.xxx.xxx with username=pi and password=your-password port=22
	Reverse up the tree to the root, then go to var/www2/html
	Copy files for the html directory.

Verify the webserver and your files are working by entering into a browser:
http://192.168.xxx.xxx:82/my-html-file.html

Now in screenly enter your asset URL as:
http://127.0.0.1:82/my-html-file.html etc.