Screenly/Anthias #1177: Cannot play WebGL 3D Websites

carlesjavierre, 2019-07-04:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Cannot play 3D enabled websites ("Enable WebGL to explore in 3D")

#### Reproduction Steps

Add an asset with this URL (for example): https://my.matterport.com/show/?m=gCq5iFJrJXG

### Environment

* Raspberry Pi 3B
* Screenly OSE Version: Checkout from master today. Built with balena.

vpetersson, 2019-07-05:
This is expected. WebGL isn’t really well supported on the Pi to my knowledge. Perhaps this will change with the Pi 4, but I doubt that is well supported yet in any browser.

carlesjavierre, 2019-07-05:
I've seen it's quite messy, but I was hoping for someone to know more than me and have it working in a more fluid way. Anyhow, let's see how the new RBP holds it.

Speaking of the Pi 4, is there any expected/imminent support for it? (Couldn't find it on the next sprints)

vpetersson, 2019-07-08:
> Speaking of the Pi 4, is there any expected/imminent support for it? 

We need to build a new image for this, but in theory, it should work if you install Raspbian Stretch on the Pi 4 and do a manual installation of Screenly.

closed 2019-09-13 by stale[bot]