Screenly/Anthias #1178: Feature Request: ADD CEC Control...

unmanagedtn, 2019-07-06:
Add CEC Control over HDMI

1. sudo apt-get install cec-utils

Once everything is installed you should be able to control the tv using the command below:

**Turn tv on: echo on 0 | cec-client -s -d 1**
**Turn tv off: echo standby 0 | cec-client -s -d 1**
**Tv status: echo pow 0 | cec-client -s -d 1**

#### Troubleshooting Tips:

* Make sure your tv supports cec and that it is enabled. Tv manufactures call [CEC by different names](http://hdmicec.blogspot.com/2012/05/how-to-find-out-if-your-device-is-hdmi.html) so you may have to do some research depending on your brand.
* Make sure you are using a [new hdmi cable](http://amzn.to/1OniG9v) that is at least HDMI 1.2a

#### Different names for HDMI CEC

* Samsung –  **Anynet+**
* Sony –  **BRAVIA Link**  or  **BRAVIA Sync**
* Sharp –  **Aquos Link**
* Hitachi –  **HDMI-CEC**
* AOC –  **E-link**
* Pioneer –  **Kuro Link**
* Toshiba –  **Regza Link**  or  **CE-Link**
* Onkyo –  **RIHD**  (Remote Interactive over HDMI)
* LG –  **SimpLink**
* Panasonic –  **VIERA Link**  or  **HDAVI Control**  or  **EZ-Sync**
* Philips –  **EasyLink**
* Mitsubishi –  **NetCommand**  for HDMI
* Runco International –  **RuncoLink**
 

Credits:raspberrypi.stackexchange.com/questions/7054/cec-wake-up-command

vpetersson, 2019-07-08:
Nice documentation @unmanagedtn. Somewhat of a duplicate of #108, but will close out the other one in favor of this as it has more info.

ealmonte32, 2019-11-13:
@unmanagedtn Hello, just wondering, can you remind me why you need these CEC controls? besides turning screen on & off since we can already do this via `vcgencmd display_power 0/1`

vpetersson, 2019-11-13:
I'm curious if CEC would yield a different result than `vcgencmd display_power`. In my experience `vcgencmd` has been flaky.

ealmonte32, 2019-11-13:
I'll test now, tvservice -o turns off the screen, but when i turn it back on with tvservice -p , the viewer or any image does not show, i need to restart screenly-viewer.service for the images to come back.

vpetersson, 2019-11-13:
> I'll test now, tvservice -o turns off the screen, but when i turn it back on with tvservice -p , the viewer or any image does not show, i need to restart screenly-viewer.service for the images to come back.

Interesting. Good find. 

Could you compare that to the cec command?

ealmonte32, 2019-11-13:
Will do

ealmonte32, 2019-11-13:
spent some time trying to figure out what this would be useful for in terms of screenly-specific use cases, and i dont see many use cases that require this, changing volume from screenly dashboard? that is about the only one that i could see being used.. standby doesn't work on TV's or monitors, so if it is to turn off the display vcgencmd display_power 0 accomplishes this without a problem..

```
pi@screenly400w:~ $ echo pow 0 | cec-client -s -d 1
opening a connection to the CEC adapter...
power status: on
pi@screenly400w:~ $ echo standby 0 | cec-client -s -d 1
opening a connection to the CEC adapter...

pi@screenly400w:~ $ echo mute | cec-client -s -d 1
opening a connection to the CEC adapter...
mute: 7F

pi@screenly400w:~ $ echo unmute | cec-client -s -d 1
opening a connection to the CEC adapter...

pi@screenly400w:~ $ echo volup | cec-client -s -d 1
opening a connection to the CEC adapter...
volume up: 7F

pi@screenly400w:~ $ echo poll 1 | cec-client -s -d 1
opening a connection to the CEC adapter...
POLL message not sent

pi@screenly400w:~ $ echo poll 0 | cec-client -s -d 1
opening a connection to the CEC adapter...
POLL message sent

pi@screenly400w:~ $ echo lad | cec-client -s -d 1
opening a connection to the CEC adapter...
listing active devices:
logical address 0
logical address 1

pi@screenly400w:~ $ echo scan | cec-client -s -d 1
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Unknown
osd string:    TV
CEC version:   1.4
power status:  on
language:      eng

device #1: Recorder 1
address:       2.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)

```

vpetersson, 2019-11-14:
There are some really cool things you can do with CEC that you cannot do with `vcgencmd`. 

Here are some examples:

 * You're able to change the channel on the monitor, such that you can change it to display the input Screenly is on.
 * You can "name" a channel (such that "Screenly" shows up in the input menu).

ealmonte32, 2019-11-20:
I see, so like if screenly is at a store display and someone mistakenly changed the input, your going to program cec so that a certain setting from the WebUI will always switch back the channel input to the named "Screenly" one?

vpetersson, 2019-11-20:
Yep. Or perhaps you turn off the TV at night. You could then use CEC to both turn on the monitor but also ensure that it switches to the right input.

escoand, 2019-12-05:
Are there already plans to integrate this? What's the best way to do so?

I would say the most flexible and powerful way would be to have some kind of special asset for `screen off` which could be scheduled on week days and time. This would be simple for non-experienced users to configure.

Btu in the meanwhile how to add a cronjob in a simple way?

vpetersson, 2019-12-05:
> Are there already plans to integrate this? What's the best way to do so?
> 
> I would say the most flexible and powerful way would be to have some kind of special asset for `screen off` which could be scheduled on week days and time. This would be simple for non-experienced users to configure.
> Update: This could be also nice for every asset, something like `show on every monday 8 to 10`.

I'm not too keen on this approach to be honest.

> Btu in the meanwhile how to add a cronjob in a simple way?

Depends on how you're running it. If you're using the disk image, just doing `crontab -e` should allow you to add the cronjob.

escoand, 2019-12-05:
> Depends on how you're running it.

Yes, off course. I'm using balena-cloud.

vpetersson, 2019-12-05:
That complicate things. You can't really use a regular cronjob (easily) on Balena. You'd have to wait (or contribute) the code required to do this as a celery task.

vpetersson, 2021-01-08:
Calling for testers on this. I've written a [script](https://github.com/Screenly/screenly-ose/blob/master/bin/cec_test.sh) that gathers some diagnostics that will help us assess how well CEC is implemented across monitors.

To participate, please take the following steps:

* Run `$HOME/screenly/bin/run_upgrade.sh` and select the developer version
* Run `$HOME/screenly/bin/cec_test.sh` and post the result

ealmonte32, 2021-01-12:
> Calling for testers on this. I've written a [script](https://github.com/Screenly/screenly-ose/blob/master/bin/cec_test.sh) that gathers some diagnostics that will help us assess how well CEC is [...]

Just to confirm, are we talking about TVs and computer monitors that are connected through HDMI only correct?

vpetersson, 2021-01-12:
Yes, this is HDMI only (part of that standard). It will **only** work on CEC enabled devices. I have not run across any computer monitors with CEC support, but I'm sure they exist. Most modern TVs will support CEC, but what we want to collect information on here is *how* well it works across a larger set of monitors.

escoand, 2021-01-23:
I need at least `v4l-utils` as `cec-compliance` is in this package at debian. And could you post a command how to check it on a docker hist? I tried something like this
```
docker run -it --rm --device /dev/vchiq --device /dev/cec0 --privileged debian:buster-slim sh -c 'apt update -qq && apt install -yqq curl v4l-utils && curl -o test.sh https://raw.githubusercontent.com/Screenly/screenly-ose/master/bin/cec_test.sh && bash test.sh'
```

vpetersson, 2021-01-23:
So if you're on the developer/master branch on either Balena or natively, everything is already installed. You should be able to simply run `./bin/cec_test.sh`.

ealmonte32, 2021-01-26:
I believe the users need to run the command with sudo because it requires installing other packages, see below:

```
pi@screenlypi4:~ $ ./screenly/bin/cec_test.sh 
What is the TV brand/make? Element
What is the TV model? N/A (non-Smart TV)

Starting diagnostics. This will take a few minutes...

E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?
./screenly/bin/cec_test.sh: line 20: cec-client: command not found
./screenly/bin/cec_test.sh: line 23: cec-client: command not found
Is the TV on right now? yes
./screenly/bin/cec_test.sh: line 29: cec-client: command not found
Is the TV off right now? ^C
```

then when ran with sudo:
```
pi@screenlypi4:~ $ sudo ./screenly/bin/cec_test.sh 
What is the TV brand/make? Element            
What is the TV model? N/A (non-Smart TV)

Starting diagnostics. This will take a few minutes...

Selecting previously unselected package libp8-platform2:armhf.
(Reading database ... 42401 files and directories currently installed.)
Preparing to unpack .../libp8-platform2_2.1.0.1+dfsg1-2_armhf.deb ...
Unpacking libp8-platform2:armhf (2.1.0.1+dfsg1-2) ...
Selecting previously unselected package libxrandr2:armhf.
Preparing to unpack .../libxrandr2_2%3a1.5.1-1_armhf.deb ...
Unpacking libxrandr2:armhf (2:1.5.1-1) ...
Selecting previously unselected package libcec4:armhf.
Preparing to unpack .../libcec4_4.0.4+dfsg1-2+rpi1+rpt1_armhf.deb ...
Unpacking libcec4:armhf (4.0.4+dfsg1-2+rpi1+rpt1) ...
Selecting previously unselected package cec-utils.
Preparing to unpack .../cec-utils_4.0.4+dfsg1-2+rpi1+rpt1_armhf.deb ...
Unpacking cec-utils (4.0.4+dfsg1-2+rpi1+rpt1) ...
Selecting previously unselected package pastebinit.
Preparing to unpack .../pastebinit_1.5-2_all.deb ...
Unpacking pastebinit (1.5-2) ...
Setting up pastebinit (1.5-2) ...
Setting up libp8-platform2:armhf (2.1.0.1+dfsg1-2) ...
Setting up libxrandr2:armhf (2:1.5.1-1) ...
Setting up libcec4:armhf (4.0.4+dfsg1-2+rpi1+rpt1) ...
Setting up cec-utils (4.0.4+dfsg1-2+rpi1+rpt1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Is the TV on right now? yes
Is the TV off right now? no
Is the TV on right now? yes

Please share this URL with us:
http://paste.debian.net/hidden/a109d85d/
```

vpetersson, 2021-01-27:
@ealmonte32 Thanks! So it looks like the monitor/TV you are using doesn't support CEC. We need a good way to detect this.

augustodinizl, 2021-02-26:
> Calling for testers on this. I've written a [script](https://github.com/Screenly/screenly-ose/blob/master/bin/cec_test.sh) that gathers some diagnostics that will help us assess how well CEC is [...]

count on me, already installing, will post the results ASAP