Screenly/Anthias #1181: Lack of developer documentation and difficulty running tests

gtrak, 2019-07-08:
I'm attempting to adapt screenly for a custom use-case, learning the code, etc.  I hit a wall running the tests on a local ubuntu 19.04 install.  Eventually, through looking through the travis-ci config, I was able to install enough dependencies to be able to run the tests, but hit a few roadblocks.  I'd like to highlight what those are for the next person and see what is some low-hanging fruit.

Ideally, I'd like a way to reproduce the Travis-CI test environment in a local docker image, which can serve as documentation for what's needed to run the tests locally, outside of travis's own infra.

I would be happy to contribute to documenting or improving these issues, and I'm not sure if it is desirable or inline with the project direction, so seeking some guidance.  What should be split out into a separate issue, what can be documented better, etc.

  - Lack of an isolated dev environment: In my brief foray into python, I've noticed there are a number of tools that are partial solutions to dependency management. Pipenv seems the best of these, which brings together pyenv, virtualenv, and deterministic dependency management similar to npm lock files.  I would propose switching from the requirements.txt to a Pipfile, and have generated one locally using pipenv's CLI with some success.  It's a necessity for developers working on more than one project at a time to minimize the shared dependencies needed and their interaction.

  - Not clear what python version the project requires.  I see the project is still on 2.7 (noticed the use of print statements), but adding a .pyenv-version or Pipenv file would make it clearer.

  - The tests are really slow: specifically the splinter/selenium based tests require extra setup (I tried using a newer version of splinter, which tipped me to needing geckodriver, which I had to install manually), and the test running seems to wait for over a minute after the browser exits on an up-to-date firefox 66 (travis references 45).  I can't tell if I'm doing something wrong, or if I just need to turn off those tests, but I see it's much faster on travis-ci.  The issues were not googleable.  The specific issue is too much dependency on an old OS and its deps (travis-ci and ubuntu 14.04).  IMO, the fix for this is more isolation, allowing the tests to be run in docker, with a headless firefox or XVFB setup.  This dockerfile can also serve as documentation for the minimal dependencies needed to work on the project. I attempted to alter the supplied Dockerfile, but hit so many problems that I wanted to ask here first.

  - The tests require files to be copied into paths in the user's home directory.  IMO the fix for this is to use env-vars to override the $HOME-relative paths to in-repo files only during a test scenario.

  - The tests require a daemon to be running. I would normally expect calling the nosetests CLI command to spin up whatever services are needed and tear them down at the end. This seems especially easy given the DB is just a sqlite file.

vpetersson, 2019-07-09:
@gtrak This is unfortunately a known problem. As with many open source projects, documentation is what suffers. Having more documentation and a better development environment is absolutely in the projects interest. It is largely a bandwidth issue.

Generally speaking, when developing the server component, using the [Docker setup](https://github.com/Screenly/screenly-ose#dockerized-development-environment) is the easiest way. It should work out-of-the-box.

Doing end-to-end testing on the other hand is a bit more challenging, as it requires X11, and rather specific environment setup. As such, when I was actively working on OSE myself, I usually had a Raspberry Pi with Screenly OSE installed where I `rsync`d in my changes to for testing. This of course sub-par, but it was the path of least resistance.

I know that we have had multiple contributors over the years that have used either a Linux VM or a Linux workstation, where they got the entire stack working natively. Unfortunately I don't think this is documented anywhere.

I hope that makes sense, and we'd love to get some contributions around lowering the barrier for getting the development environment running.

vpetersson, 2019-07-09:
Also, ping @RemingtonA when you're back from vacation.

gtrak, 2019-07-09:
I think I can alter the Dockerfile to mirror the Travis CI tests with a little effort. I'll try down that path.  Since it's separate from the balena/deploy dockerfile, it might be suitable for a development environment?

I'm still not sure what the right path is for pipenv/python3 etc., but it's less of a priority once docker works for testing.

vpetersson, 2019-07-09:
> Since it's separate from the balena/deploy dockerfile, it might be suitable for a development environment?

The main Dockerfile is currently used for the [demo site](http://ose.demo.screenlyapp.com) and for local development environment. It's fine to keep building on this as long as we don't introduce any issues.

> I'm still not sure what the right path is for pipenv/python3 etc., but it's less of a priority once docker works for testing.

There's an open PR for the migration to Python 3 [here](https://github.com/Screenly/screenly-ose/pull/1180), but it is work-in-progress.

closed 2019-09-14 by stale[bot]