Screenly/Anthias #1182: Feature Request: Download Video from URL

gtrak, 2019-07-08:
When evaluating this project, I noticed the ability to play youtube URLs and that it would actually download the video file locally.

Upon digging into the code, I saw that the architecture to do so is a mix of message-queues and shelling out into a youtube-dl script, but expected it to be more general.

It appears as if I have the option of locally syncing youtube vids, and streaming URLs, but not locally syncing arbitrary URLs.  I would like to be able to sync video from arbitrary URLs.  For my specific use-case, it is fine if it's just a http file download, not syncing against any streaming protocol.

I'm happy to contribute code for this if it's desirable, but would like some guidance on the architecture.

I guess it would look like: 
  - a checkbox during URL paste
  - Some validation on the URL
  - Mapping of the downloaded content URL to local storage (however the youtube work is doing it)
  - Actually downloading the file (with curl or something that allows auth headers)

The backend parts are the highest priority on my end.

vpetersson, 2019-07-09:
@gtrak my understanding is that a YouTube URL can never change the content (i.e. you can't replace it with a new video). Hence having a syncing feature is somewhat moot. 

We do support proper streaming URLs too (see [release notes](https://www.screenly.io/blog/2017/07/05/screenly-ose-0-10-is-out/)).

I doubt streaming YouTube is realistic on the Raspberry Pi, unless we can extract an RTSP feed or similar)

gtrak, 2019-07-09:
To be clear, I mean for this feature to locally download videos that are not youtube, eg hosted as a regular file somewhere on a CDN.  Currently, it seems it will play the URL directly, and that's not always going to be as reliable or quick enough depending on the network connection.

vpetersson, 2019-07-09:
Oh, I see. Well, honestly, i think it's easier to do that using the API instead.

gtrak, 2019-07-09:
I can't seem to find documentation on the API, but it seems to fit the use-case according to https://www.screenly.io/api/ .  

I was also hoping for more of a pull model to sync assets with an upstream server, since it's not easy to call into a raspberry pi box from the internet.  Is that possible with either OSE or the commercial version?

vpetersson, 2019-07-09:
In the commercial version of Screenly, you can fire off API calls and add/remove assets and have them being updated on all the devices. 

The API docs for Screenly OSE can be found [here](http://ose.demo.screenlyapp.com/api/docs/), and the commercial API is the one you linked to.

gtrak, 2019-07-09:
OK, it's not clear to me if this is desirable for OSE's scope, when there are a couple of ways to address it including commercial. Happy to close this unless I'm missing something.

closed 2019-07-09 by gtrak

vpetersson, 2019-07-09:
Generally speaking, we want to move as much as possible to the API rather than building in everything (as it adds a lot of complexity). From the sound of the description, a 15-20 line Python script should be able to both check for an updated version and add/remove the asset to Screenly (regardless of version).

gtrak, 2019-07-09:
Yea, for my use-case, I'm considering whether I'd add a separate daemon to be deployed alongside on the raspberry pi, then I have the option of interacting with the screenly API or reaching into the sqlite file at my own risk.  It sounds like if the API has specific gaps, then I should open up issues/PRs for those.

vpetersson, 2019-07-09:
Yeah, having a side-loaded service/daemon should do the job just fine. And yeah, if you run into issues with the API, file a separate issue for that. Also, I'd strongly discourage altering the sqlite directly.