Screenly/Anthias #1183: ERR_CONNECTION_REFUSED after update (see #361) - rerunning upgrade not working.

WTPLJim, 2019-07-08:
#### Overview of the Issue:
    Ran upgrade today, splash screen comes up and then TV goes blank. I tried accessing the 8080 page, and it refused connection. I assume at this time, the screen going blank after showing commands after splash screen is simply that there is no active assets in the 8080 page.

#### Reproduction Steps
Upgraded, see splash screen, see some commands, screen blanks, attempt login to 8080 via another browser, told connection refused.

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 2
* Screenly OSE Version: Jessie

WTPLJim, 2019-07-08:
**Note: Service is failing to load!!

pi@raspberrypi:~ $ sudo systemctl
  UNIT                      LOAD   ACTIVE     SUB          DESCRIPTION
  proc-sys-fs-binfmt_misc.automount loaded active     waiting      Arbitrary Exe
  sys-devices-platform-soc-3f201000.uart-tty-ttyAMA0.device loaded active     pl
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0
  sys-devices-platform-soc-3f980000.usb-usb1-1\x2d1-1\x2d1.1-1\x2d1.1:1.0-net-et
  sys-devices-virtual-sound-card0.device loaded active     plugged      /sys/dev
  sys-devices-virtual-tty-ttyprintk.device loaded active     plugged      /sys/d
  sys-module-configfs.device loaded active     plugged      /sys/module/configfs
  sys-subsystem-net-devices-eth0.device loaded active     plugged      SMSC9512/
  -.mount                   loaded active     mounted      /
  boot.mount                loaded active     mounted      /boot
  dev-mqueue.mount          loaded active     mounted      POSIX Message Queue F
  run-user-1000.mount       loaded active     mounted      /run/user/1000
  sys-kernel-config.mount   loaded active     mounted      Configuration File Sy
  sys-kernel-debug.mount    loaded active     mounted      Debug File System
  systemd-ask-password-plymouth.path loaded active     waiting      Forward Pass
  systemd-ask-password-wall.path loaded active     waiting      Forward Password
  session-c2.scope          loaded active     running      Session c2 of user pi
  asplashscreen.service     loaded active     exited       LSB: Show custom spla
  avahi-daemon.service      loaded active     running      Avahi mDNS/DNS-SD Sta
  console-setup.service     loaded active     exited       LSB: Set console font
lines 1-23
  UNIT                      LOAD   ACTIVE     SUB          DESCRIPTION
  proc-sys-fs-binfmt_misc.automount loaded active     waiting      Arbitrary Executable
  sys-devices-platform-soc-3f201000.uart-tty-ttyAMA0.device loaded active     plugged      /sys/devices/platform
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p1.device loaded active     plug
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0-mmcblk0p2.device loaded active     plug
  sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0001-block-mmcblk0.device loaded active     plugged      /
  sys-devices-platform-soc-3f980000.usb-usb1-1\x2d1-1\x2d1.1-1\x2d1.1:1.0-net-eth0.device loaded active     plugged
  sys-devices-virtual-sound-card0.device loaded active     plugged      /sys/devices/virtual/
  sys-devices-virtual-tty-ttyprintk.device loaded active     plugged      /sys/devices/virtual/
  sys-module-configfs.device loaded active     plugged      /sys/module/configfs
  sys-subsystem-net-devices-eth0.device loaded active     plugged      SMSC9512/9514 Fast Et
  -.mount                   loaded active     mounted      /
  boot.mount                loaded active     mounted      /boot
  dev-mqueue.mount          loaded active     mounted      POSIX Message Queue F
  run-user-1000.mount       loaded active     mounted      /run/user/1000
  sys-kernel-config.mount   loaded active     mounted      Configuration File Sy
  sys-kernel-debug.mount    loaded active     mounted      Debug File System
  systemd-ask-password-plymouth.path loaded active     waiting      Forward Password Requ
  systemd-ask-password-wall.path loaded active     waiting      Forward Password Requ
  session-c2.scope          loaded active     running      Session c2 of user pi
  asplashscreen.service     loaded active     exited       LSB: Show custom spla
  avahi-daemon.service      loaded active     running      Avahi mDNS/DNS-SD Sta
  console-setup.service     loaded active     exited       LSB: Set console font
  cron.service              loaded active     running      Regular background pr
  dbus.service              loaded active     running      D-Bus System Message
  dhcpcd.service            loaded active     running      dhcpcd on all interfa
  dphys-swapfile.service    loaded active     exited       LSB: Autogenerate and
  fake-hwclock.service      loaded active     exited       Restore / save the cu
  getty@tty1.service        loaded active     running      Getty on tty1
  kbd.service               loaded active     exited       LSB: Prepare console
  keyboard-setup.service    loaded active     exited       LSB: Set preliminary
  kmod-static-nodes.service loaded active     exited       Create list of requir
  matchbox.service          loaded active     running      X11
  ModemManager.service      loaded active     running      Modem Manager
  networking.service        loaded active     exited       LSB: Raise network in
  ntp.service               loaded active     running      LSB: Start NTP daemon
  polkitd.service           loaded active     running      Authenticate and Auth
  raspi-config.service      loaded active     exited       LSB: Switch to ondema
  rc-local.service          loaded active     exited       /etc/rc.local Compati
  rsyslog.service           loaded active     running      System Logging Servic
● screenly-viewer.service   loaded failed     failed       Screenly Viewer
  screenly-web.service      loaded activating auto-restart Screenly Web UI
[... cut ...]
  timers.target             loaded active     active       Timers
  systemd-tmpfiles-clean.timer loaded active     waiting      Daily Cleanup of Temp

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

98 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.

WTPLJim, 2019-07-08:
Some more...

pi@raspberrypi:~ $ sudo service screenly-viewer start
pi@raspberrypi:~ $ sudo service screenly_net_mgr start
Job for screenly_net_mgr.service failed. See 'systemctl status screenly_net_mgr.                   service' and 'journalctl -xn' for details.
pi@raspberrypi:~ $ journalctl -xn
No journal files were found.
pi@raspberrypi:~ $ systemctl status screenly_net_mgr.service
● screenly_net_mgr.service - LSB: Screenly network config tool
   Loaded: loaded (/etc/init.d/screenly_net_mgr)
   Active: failed (Result: exit-code) since Mon 2019-07-08 13:09:17 EDT; 5min ag                   o
  Process: 8916 ExecStart=/etc/init.d/screenly_net_mgr start (code=exited, statu                   s=203/EXEC)
pi@raspberrypi:~ $

vpetersson, 2019-07-09:
@rusko124

rusko124, 2019-07-12:
Hey @WTPLJim !
`screenly_net_mgr.service` looks like a old service.
Good and fast solution for you I think will be installing the latest version from https://github.com/Screenly/screenly-ose/releases/tag/0.17.
If you still want to keep the current version then run this command `cd ~/screenly && ./bin/run_upgrade.sh` and after that provide me logs of updating, please.

WTPLJim, 2019-07-12:
Okay - is there a command path via SSH to upgrade to the newest Sprint?
- Pi is in an engineered "security" cage that can be a bit rough to access.

I'll start the upgrade tomorrow if I need to access the physical device.

Thanks, running update on current version now.

WTPLJim, 2019-07-12:
Log of upgrade:

pi@raspberrypi:~ $ cd ~/screenly && ./bin/run_upgrade.sh
   _____                           __         ____  _____ ______
  / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
  \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
 ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
/____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                 /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Would you like to use the experimental branch? It contains the last major changes, such as the new browser and migrating to Docker (y/N)

Do you want Screenly to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to install the WoTT agent to help you manage security of your Raspberry Pi? (y/N)

Would you like to perform a full system upgrade as well? (y/N)
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt-get update
Get:1 http://mirrordirector.raspbian.org jessie InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org jessie InRelease [22.9 kB]
Get:3 http://mirrordirector.raspbian.org jessie/main armhf Packages [9,539 kB]
Get:4 http://archive.raspberrypi.org jessie/main armhf Packages [170 kB]
Get:5 http://archive.raspberrypi.org jessie/ui armhf Packages [58.9 kB]
Ign http://archive.raspberrypi.org jessie/main Translation-en_US
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_US
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,806 kB in 42s (230 kB/s)
Reading package lists... Done
+ sudo apt-get purge -y python-setuptools python-pip python-pyasn1
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'python-pip' is not installed, so not removed
Package 'python-pyasn1' is not installed, so not removed
Package 'python-setuptools' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 256 not upgraded.
+ sudo apt-get install -y python-dev git-core libffi-dev libssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
git-core is already the newest version.
libffi-dev is already the newest version.
libssl-dev is already the newest version.
python-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 256 not upgraded.
+ curl -s https://bootstrap.pypa.io/get-pip.py
+ sudo python
Traceback (most recent call last):
  File "<stdin>", line 21492, in <module>
  File "<stdin>", line 197, in main
[... 42 lines cut ...]
  File "/usr/lib/python2.7/multiprocessing/__init__.py", line 84, in <module>
    import _multiprocessing
ImportError: /usr/lib/python2.7/lib-dynload/_multiprocessing.arm-linux-gnueabihf.so: undefined symbol: PyFloat_AsDouble
+ sudo apt-get autoclean
Reading package lists... Done
Building dependency tree
Reading state information... Done
+ sudo apt-get clean
+ sudo find /usr/share/doc -depth -type f '!' -name copyright -delete
+ sudo find /usr/share/doc -empty -delete
+ sudo rm -rf /usr/share/man /usr/share/groff /usr/share/info /usr/share/lintian /usr/share/linda /var/cache/man
+ sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -exec rm -r '{}' ';'
+ cd /home/pi/screenly
+ git rev-parse HEAD
++ git rev-parse --abbrev-ref HEAD
++ git rev-parse --short HEAD
++ lsb_release -a
No LSB modules are available.
+ echo -e 'Screenly version: production@85732dd\nDistributor ID:        Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie'
+ set +x
Installation completed.

WTPLJim, 2019-07-12:
Upgraded to Sprint9. 
I have a splash screen; cannot connect to web interface (IP:8080).
Nor can I access via SSH.

I am able to ping on IP (172.29.53.102)

WTPLJim, 2019-07-12:
Got it... removed the Port 8080 on flash!

WTPLJim, 2019-07-12:
Up and working again, thanks!

WTPLJim, 2019-07-12:
Before closing.... SSH is no longer connecting on new version.
Is this now on a non standard port?

vpetersson, 2019-07-12:
> Before closing.... SSH is no longer connecting on new version.
> Is this now on a non standard port?

SSH has been disabled by default for some time in Raspbian (and Screenly OSE). 

You need to manually enable it using `sudo raspi-config`.

closed 2019-07-12 by vpetersson