Screenly/Anthias #1187: Upgrade to Raspbian Buster

vpetersson, 2019-07-10:
The Raspberry Pi 4 Model B does not seem to boot on Jessie. We need to upgrade to Buster.

klarkizim, 2019-10-16:
Hi Victor, still learning how to follow on Git, 
where will i find the 'buster' image / instructions on upgrading ?

vpetersson, 2019-10-16:
@klarkizim No need to know anything about `git`. You can find images [here](https://github.com/Screenly/screenly-ose/releases). Just do a backup and then restore. That's the easiest way to do it.

klarkizim, 2019-10-17:
thanks so much, will give it a try

vpetersson, 2019-10-17:
Correct - the Pi 4 is not fully supported yet. The GPU is completely different in the Pi 4, such that `ompxlayer` now breaks for video playback.

alastairbattrick, 2019-10-17:
> Correct - the Pi 4 is not fully supported yet. The GPU is completely different in the Pi 4, such that `ompxlayer` now breaks for video playback.

Is there an ETA on it being resolved?

vpetersson, 2019-10-17:
It looks like `omxplayer` now supports the Pi 4 (but not in 4k). Hence we could in theory add support for it once we've unblocked some other things. Tl;dr is that the reason why it's taken a while is because Buster + Pi 4 broke a few components that were out of our control. Check the roadmap for more details.

codmdu, 2019-10-21:
Is there a ETA for full Pi 4 Support ? (i cant find a ETA on the roadmap)

vpetersson, 2019-10-22:
@codmdu This is the ticket for it, so yes, it is on the roadmap (i.e. [Sprint 11](https://github.com/Screenly/screenly-ose/projects/4)).

hoerms, 2020-05-11:
Hi. Do you have any idea when sprint 11 will be finished. Regards Hermann S.

vpetersson, 2020-05-13:
No update on this as of right now, unfortunately. COVID as forced us to tighten our budgets.

ealmonte32, 2020-10-02:
https://github.com/Screenly/screenly-ose/issues/1355#issuecomment-696388611

closed 2020-12-20 by vpetersson