Screenly/Anthias #1193: Balena and ScreenlyOSE video not playing performance issues with .GIF

WMcClureInzata, 2019-07-15:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Video does not play. Blank screen for default time (10 sec) no matter video length. Then switch to next asset. Images load fine. GIF files load but play poorly. If in full resolution, choppy and erratic. Loads fine if small resolution (640x480)

I apologize if I lack any particular detail. I am very new at this. If any other information needed to assist, please let me know. 

#### Reproduction Steps

Deleted all assets, reloaded new assets making sure the resolution and formats matched.
On Balena Log Screen:
```
15.07.19 08:17:13 (-0400)  screenly-viewer  Generating asset-list...
15.07.19 08:17:13 (-0400)  screenly-viewer  Showing asset ezgif.com-gif-to-mp4.mp4 (video)
15.07.19 08:17:23 (-0400)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
15.07.19 08:17:23 (-0400)  screenly-viewer  Resource URI is not correct, remote host is not responding or request was rejected.
15.07.19 08:17:24 (-0400)  screenly-viewer  Generating asset-list...
15.07.19 08:17:24 (-0400)  screenly-viewer  Showing asset Inflow2.gif (image)
15.07.19 08:17:34 (-0400)  screenly-viewer  Sleeping for 11
15.07.19 08:17:45 (-0400)  screenly-viewer  Generating asset-list...
15.07.19 08:17:45 (-0400)  screenly-viewer  Showing asset My Promo video 4.mp4 (video)

```

State 0x12000a [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
Model 3B Revision: 1.2 Ram: 1 GB Sony UK
balenaOS 2.38.0+rev1
release : d856ab07e72a2128bdeda8356d1f2923

vpetersson, 2019-07-15:
What is the codec of the video? It needs to be encoded using h264 for it to work. The Raspberry Pi is very picky on this.

closed 2019-09-20 by stale[bot]

MydKnight, 2019-09-23:
I'm also having this issue. I tried using a video off of Youtube as well as your asset.mov and neither played with the same error as above. I'm currently running a video through an h264 encoder to see if that resolves it. I'll update one way or the other.

developer-bit-2020, 2019-12-05:
I had the same issue where my Screenly set up on Rpi with Balena Cloud would not play any of the H264 or normal videos.

This is what I tried to fix it:

Adjust the GPU memory of your Raspberry Pi in order to optimize content playback.
To do this, click the “Fleet Configuration” tab on the left-side menu on the balenaCloud interface. On the Fleet Configuration menu, scroll down to find the RESIN_HOST_CONFIG_gpu_mem configuration item. Then, click “activate” to enable the item. Next, click the pencil icon on the right-hand side and change the value to 192. Restart the Pi to make this config work.
Hope that is helpful!