Screenly/Anthias #1194: 504 gateway time-out

TheCazz, 2019-07-15:
Hi
I have install the latest version of screenly OSE and I have got it running for a friend

At home I have add som rtsp streaming and got it to work and I have now done that on for my friends screenly but now I get 504 gateway time-out and not sure how to see how to fix the problem.
I have look at journalctl but it does not say so much when I do not know what to look after.

The only think I can think of that is different between his and my Raspberry Pi is that my is a little new then his (My is 3+ and he is 3) other then that is not so much change.

vpetersson, 2019-07-16:
@TheCazz Try upgrading to the latest development version to see if that resolves the issue.

TheCazz, 2019-07-16:
hmm sorry if it sound stupid but how do I do that?

I guess if I use this

`bash <(curl -sL https://www.screenly.io/install-ose.sh)`

then it not the dev it trying to install?

vpetersson, 2019-07-16:
That's the right steps - just make sure to answer `Y` to if you want to install the development version.

TheCazz, 2019-07-16:
After a update now I get 502 Bad Gateway

rusko124, 2019-07-17:
Hey @TheCazz !
[image]

Could your provide logs(`journalctl -u screenly-web.service`), please?
Also would be good if you said all steps to reproduce the problem, and what device do you use?

TheCazz, 2019-07-17:
Well first I did install the image to SD card and Power on
It looks like it works so I have to after that force HDMI (Have VGA adapter) and then rotate the screen
Everything looks fine so I did add a rstp address that I know is working (Have try it on VLC) and after that I get the error. so I did run the manual update and did say yes to almost everything (but no to update the raspberry system). I did take sometime but after a reboot I get now 502 error.

When I run the journal command it say nothing
`-- No enteries --`

So I guess the screenly does not start?

when I look for the model I get
`Raspberry Pi Model B Plus Rev.12`
so is not so old version of raspberry pi

My friend want to have like me rstp streaming but I was thinking if I try to remove the rstp and see if it works but I can't do that, Maybe is possible att edit from the terminal but not sure.

rusko124, 2019-07-18:
@TheCazz As far as I understand you got the error right after adding the rstp asset?
Then could you provide the full logs?`journalctl`

TheCazz, 2019-07-18:
[image]

Hmm well no but I maybe do something wrong?

rusko124, 2019-07-18:
@TheCazz I mean without any unit. Just `journalctl` :)

TheCazz, 2019-07-18:
ahh ok but what are I going to look after. is going to be many line of info :)
All I can see now is 
[image]
wlan info (I going to use wifi later but now is ethernet connected)

rusko124, 2019-07-18:
@TheCazz Please put the logs to somewhere(online text paste service or textfile)

TheCazz, 2019-07-18:
ok here it is but it does not say so much.
https://www.c-media.se/log.txt

rusko124, 2019-07-22:
@TheCazz could you update one more time and show the logs at the end of the updating, please ?

TheCazz, 2019-07-22:
Sure it going to take a little time but I going to do it
Just run the update and again run  journalctl

TheCazz, 2019-07-22:
ok after I came back after have been out I did see it was finnish and did run the command but it gave me same info as before.
So I did reboot the raspberry and again run the command and it gave me this info
https://www.c-media.se/log2.txt

rusko124, 2019-07-23:
@TheCazz Judging by the logs, no one of the services Screenly does not start. Have you had any problems with the update?Any errors?

TheCazz, 2019-07-23:
No error what I can see but I think I have to reinstall it again when I'm at his place again.

rusko124, 2019-07-30:
Hey @TheCazz !
How is it going?

closed 2019-10-05 by stale[bot]