Screenly/Anthias #1196: Black page displays at end of asset list

at565, 2019-07-19:
Running on a Pi 3 Model B. Can't say the version of OSE as it doesn't say anywhere in the UI and there is no SSH access to this device, which I don't have physical access to at present. I can say that it was whatever version was available back in February of this year, as that was when it the device was built and installed.

OSE is throwing up the black screen for 30 seconds, seemingly at the end of every cycle through the asset list. This is an issue as the asset list only has a few items, so a black screen is displaying about 20% of the time. It may also be hanging on this screen, as we've had complaints that it's just blank. When I had someone check it today, the screen was black when they got there. After waiting a while they power cycled the TV (and thus the Pi), so they couldn't say for sure if it was definitely hung at that point.

II had them watch it for a while and could verify there wasn't a loss of video signal by checking the display link status on the TV, and the logs also corroborate what I'm seeing, pasted below. I also confirmed that all energy saving settings are off on the TV so it isn't the display. 

```

Jul 19 17:30:54 raspberrypi python[1035]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 1115.
--
Jul 19 17:30:55 raspberrypi python[1035]: Sleeping for 30
Jul 19 17:31:25 raspberrypi python[1035]: Last update: 2019-07-19 06:18:24.545121
Jul 19 17:31:25 raspberrypi python[1035]: get_next_asset
Jul 19 17:31:25 raspberrypi python[1035]: refresh_playlist
Jul 19 17:31:25 raspberrypi python[1035]: refresh: counter: (0) deadline (2019-03-25 20:26:00) timecur (2019-07-19 17:31:25.697081)
Jul 19 17:31:25 raspberrypi python[1035]: update_playlist
Jul 19 17:31:25 raspberrypi python[1035]: Generating asset-list...
Jul 19 17:31:25 raspberrypi python[1035]: generate_asset_list deadline: 2019-03-25 20:26:00
Jul 19 17:31:25 raspberrypi python[1035]: get_next_asset after refresh
Jul 19 17:31:25 raspberrypi python[1035]: get_next_asset counter 0 returning asset 1 of 4
Jul 19 17:31:25 raspberrypi python[1035]: Showing asset Spring Colours Logo Swap_1080x1920.gif (image)
Jul 19 17:31:25 raspberrypi python[1035]: Asset URI /home/pi/screenly_assets/f8d5473f6e7d4641851a0017752ea116
Jul 19 17:31:25 raspberrypi python[1035]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Jul 19 17:31:25 raspberrypi python[1035]: Sleeping for 60
.....

Jul 19 17:37:29 raspberrypi python[1035]: generate_asset_list deadline: 2019-03-25 20:26:00
--
Jul 19 17:37:29 raspberrypi python[1035]: get_next_asset after refresh
Jul 19 17:37:29 raspberrypi python[1035]: get_next_asset counter 0 returning asset 2 of 4
Jul 19 17:37:29 raspberrypi python[1035]: Showing asset Fossellos_Now_Open.jpg (image)
Jul 19 17:37:29 raspberrypi python[1035]: Asset URI /home/pi/screenly_assets/68887e6487c64c1e9f35656b281b5bcc
Jul 19 17:37:29 raspberrypi python[1035]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Jul 19 17:37:29 raspberrypi python[1035]: Sleeping for 30
Jul 19 17:37:59 raspberrypi python[1035]: Last update: 2019-07-19 06:18:24.545121
Jul 19 17:37:59 raspberrypi python[1035]: get_next_asset
Jul 19 17:37:59 raspberrypi python[1035]: refresh_playlist
Jul 19 17:37:59 raspberrypi python[1035]: refresh: counter: (0) deadline (2019-03-25 20:26:00) timecur (2019-07-19 17:37:59.613853)
Jul 19 17:37:59 raspberrypi python[1035]: update_playlist
Jul 19 17:37:59 raspberrypi python[1035]: Generating asset-list...
Jul 19 17:37:59 raspberrypi python[1035]: generate_asset_list deadline: 2019-03-25 20:26:00
Jul 19 17:37:59 raspberrypi python[1035]: get_next_asset after refresh
Jul 19 17:37:59 raspberrypi python[1035]: get_next_asset counter 0 returning asset 3 of 4
Jul 19 17:37:59 raspberrypi python[1035]: Showing asset Amni-50PC_offGraphic_v4-01.jpg (image)
Jul 19 17:37:59 raspberrypi python[1035]: Asset URI /home/pi/screenly_assets/5a7af0e394374a9daf3212aa4b091970
Jul 19 17:37:59 raspberrypi python[1035]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Jul 19 17:37:59 raspberrypi python[1035]: Sleeping for 30

```

at565, 2019-07-19:
To add to this, it appeared to us that the black screen was being displayed at the end of each cycle through the asset list. To offset this a bit, I increased the display duration of the assets from 30 to 60 seconds each. The display time for the black screen has now also increased to 60 seconds, and seems to be happening at random rather than just at the end of the asset list. More logs below from observing this:

```

Jul 19 17:55:49 raspberrypi python[1581]: Asset URI /home/pi/screenly_assets/6e1d6467a7a64de99553e897b4ef5f94
--
Jul 19 17:55:49 raspberrypi python[1581]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Jul 19 17:55:49 raspberrypi python[1581]: browser found dead, restarting
Jul 19 17:55:49 raspberrypi python[1581]: Loading browser...
Jul 19 17:55:49 raspberrypi python[1581]: killing previous uzbl 1594
Jul 19 17:55:49 raspberrypi python[1581]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 1641.
Jul 19 17:55:50 raspberrypi python[1581]: Sleeping for 60
Jul 19 17:56:50 raspberrypi python[1581]: Last update: 2019-07-19 06:18:24.545121
Jul 19 17:56:50 raspberrypi python[1581]: get_next_asset
Jul 19 17:56:50 raspberrypi python[1581]: refresh_playlist
Jul 19 17:56:50 raspberrypi python[1581]: refresh: counter: (0) deadline (2019-03-25 20:26:00) timecur (2019-07-19 17:56:50.594541)
Jul 19 17:56:50 raspberrypi python[1581]: update_playlist
Jul 19 17:56:50 raspberrypi python[1581]: Generating asset-list...
Jul 19 17:56:50 raspberrypi python[1581]: generate_asset_list deadline: 2019-03-25 20:26:00
Jul 19 17:56:50 raspberrypi python[1581]: get_next_asset after refresh
Jul 19 17:56:50 raspberrypi python[1581]: get_next_asset counter 0 returning asset 1 of 4
Jul 19 17:56:50 raspberrypi python[1581]: Showing asset Spring Colours Logo Swap_1080x1920.gif (image)
Jul 19 17:56:50 raspberrypi python[1581]: Asset URI /home/pi/screenly_assets/f8d5473f6e7d4641851a0017752ea116
Jul 19 17:56:50 raspberrypi python[1581]: Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Jul 19 17:56:50 raspberrypi python[1581]: Sleeping for 60

```

rusko124, 2019-07-22:
Hey, @at565 !
Is `Swap_1080x1920.gif` display correctly?

at565, 2019-07-23:
It is displaying normally.

rusko124, 2019-07-30:
@at565 
Could you upgrade Screenly? 
`cd screenly && ./bin/run_upgrade.sh`
I believe that It could help.
Let me know if problem still have, we will continue to try resolve this.

at565, 2019-08-08:
Unfortunately the Pi is in a secured enclosure with security bolts, etc, and SSH was not enabled when it was created. If there were a way of doing it via the web interface that would be a huge help. As there isn't at present (to my knowledge), I will update this thread when we can gain access to the device to update it (and enable ssh...)

at565, 2019-08-16:
An update on this issue:

I've been testing this on another Pi running the current (Experimental) version of Screenly OSE and have found what seems to be the cause - everything is fine, until the process hits an animated .gif. It does occasionally display it as expected, but most of the time it throws the black screen. Checking the logs, it looks like the browser is crashing:

```
● screenly-viewer.service - Screenly Viewer
--
Loaded: loaded (/etc/systemd/system/screenly-viewer.service; enabled; vendor preset: enabled)
Active: active (running) since Fri 2019-08-16 18:30:40 UTC; 10s ago
Process: 3989 ExecStartPost=/bin/rm -f /tmp/screenly_html/* (code=exited, status=0/SUCCESS)
Process: 3987 ExecStartPost=/bin/rm -f /tmp/uzbl_* (code=exited, status=0/SUCCESS)
Process: 3982 ExecStartPre=/usr/bin/xset s noblank (code=exited, status=0/SUCCESS)
Process: 3979 ExecStartPre=/usr/bin/xset -dpms (code=exited, status=0/SUCCESS)
Process: 3977 ExecStartPre=/usr/bin/xset s off (code=exited, status=0/SUCCESS)
Main PID: 3986 (python)
CGroup: /system.slice/screenly-viewer.service
├─3986 /usr/bin/python /home/pi/screenly/viewer.py
├─4018 uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html --print-events --connect-socket /home/pi/.cache/uzbl/event_daemon
└─4023 /usr/bin/python /usr/bin/uzbl-event-manager -va start
 
Aug 16 18:30:39 raspberrypi systemd[1]: Starting Screenly Viewer...
Aug 16 18:30:40 raspberrypi systemd[1]: Started Screenly Viewer.
Aug 16 18:30:43 raspberrypi python[3986]: browser found dead, restarting
Aug 16 18:30:43 raspberrypi python[3986]: Loading browser...
Aug 16 18:30:43 raspberrypi python[3986]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 4018.
Aug 16 18:30:45 raspberrypi python[3986]: current url is file:///tmp/screenly_html/black_page.html
Aug 16 18:30:45 raspberrypi python[3986]: Generating asset-list...
Aug 16 18:30:46 raspberrypi python[3986]: Generating asset-list...
Aug 16 18:30:46 raspberrypi python[3986]: Showing asset tv-1000.gif (image)
Aug 16 18:30:46 raspberrypi python[3986]: Sleeping for 10

....

Aug 16 18:32:58 raspberrypi python[3986]: Showing asset Amni-50PC_offGraphic_v4-01 (1).jpg (image)
--
Aug 16 18:32:58 raspberrypi python[3986]: Sleeping for 10
Aug 16 18:33:08 raspberrypi python[3986]: Generating asset-list...
Aug 16 18:33:08 raspberrypi python[3986]: Showing asset Logo Swap_1080x1920 (1).gif (image)
Aug 16 18:33:08 raspberrypi python[3986]: browser found dead, restarting
Aug 16 18:33:08 raspberrypi python[3986]: Loading browser...
Aug 16 18:33:08 raspberrypi python[3986]: killing previous uzbl 4172
Aug 16 18:33:08 raspberrypi python[3986]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 4218.
Aug 16 18:33:09 raspberrypi python[3986]: Sleeping for 10
Aug 16 18:33:19 raspberrypi python[3986]: Generating asset-list...
Aug 16 18:33:19 raspberrypi python[3986]: Showing asset tv-1000.gif (image)

```

Note that in the above logs, the tv-1000.gif displays without issue every time, but it isn't animated in any way, so I'm not sure if the browser has an issue with gifs that actually have multiple frames in them or not, but that's how it appears.

rusko124, 2019-08-19:
Hey @at565 !
Thank you for your contribution in finding the root of the problem. We will take this into account.

closed 2019-10-25 by stale[bot]