Screenly/Anthias #1199: Raspian Buster on RPi 4 - NOT RUNNING

hahni, 2019-08-07:
After upgrading from RPi 3 to RPi 4 I made a new Buster-Image and run a manual installation with $ bash <(curl -sL https://www.screenly.io/install-ose.sh)

The System installed without any issues and after reboot I could connect to the Web-Backend to add some screens - bit the display is black.

I read through the Buster comments and found some details in 'pi-gen' but for me its actually not possible to create a new image on my own (based on not so professional with linux).

So when a working RPi 4 image will be available or how could I make the image on my own?

vpetersson, 2019-08-12:
@rusko124 Were you able to get your hands on a pi4?

rusko124, 2019-08-13:
@vpetersson Not yet, I believe that will have it soon

hahni, 2019-08-13:
> @rusko124 Were you able to get your hands on a pi4?

Yes of course - that would pe possible

shaqaruden, 2019-08-20:
I had this problem as well but I just press Ctrl+Alt+F1, logged in and restarted (sudo reboot) and then it worked fine.

hahni, 2019-08-20:
I will try it tomorrow and report here

shaqaruden, 2019-08-20:
@hahni I also did a full update and upgrade before starting the installation and used the beta. I wanted to use the experimental but couldn’t get it to install

TheRictus, 2019-09-03:
I have the same problem - brand new RPi4, replacing the Pi3 we've been using for quite a while with Screenly. Downloaded the latest Screenly OSE. System wouldn't boot. Installed Raspian Buster, Pi boots to desktop no problem. Completed prerequisites, then installed Screenly using bash <(curl -sL https://www.screenly.io/install-ose.sh)

Opted for no dev features, install completed successfully and prompted to restart. Booted back to Raspian desktop. Attempted repeat of install, rechecked prerequisites. Same result. Wiped SD card a second time, reflashed Raspian Buster from fresh, booted up, prereqs, curl install, same result...

vpetersson, 2019-09-03:
@TheRictus You must use the development version (i.e. answer yes to that question) if you want to use it on Buster. If you do that it should in theory work.

TheRictus, 2019-09-03:
Hi @vpetersson 

Thanks - sorry I realised the same and solved it just before lunch, you beat me to replying to say that works!
Thanks :)

closed 2019-09-03 by vpetersson