Screenly/Anthias #1200: Kibana 7.0 dashboards displayed as blank page

Andr3wid, 2019-08-16:
#### Overview of the Issue

I heavily use screenly as a solution to display Kibana dashboards. Since the Kibana update to version 7.0, all Kibana-assets get displayed as a white, blank screen.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter an asset-URL that points to Kibana (you can use [this](https://demo.elastic.co) official Kibana demo if you don't have an installation in place)
* Press "On"
* Wait for the asset to show up on the screen
* As you will see, only a white screen shows up instead of the Kibana dashboard or login-page

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 3 Model B

Tried with different screenly versions. Don't know what the version I tried with was before the update. Right now running:
* experimental@1215534 (Raspbian 9.6 'stretch')

Will append other version numbers as I try throughout the day.

vpetersson, 2019-08-16:
@Andr3wid Try reverting to the development version. The experimental version is running a different browser. We are however keen on getting this to work with the experimental version. 

@rusko124 can you try to reproduce this?

Andr3wid, 2019-08-16:
@vpetersson Thanks for the fast reply. Happens in the newest production version as well. (production@618864d) Updated issue description.

vpetersson, 2019-08-16:
Can you try the development version?

Andr3wid, 2019-08-16:
Same behavior in development version (master@3c8645c).

rusko124, 2019-08-19:
I confirm that uzbl cannot display this page. In the near future, the experimental browser will become the main. Make fix for uzbl I do not see sense.

vpetersson, 2019-08-19:
@rusko124 right, but as per @Andr3wid's comment above, it doesn't seem to work in the experimental branch either.

rusko124, 2019-08-19:
I confirm that the experimental browser also does not show the page. I will look for the cause.

bummzulu, 2019-08-19:
Same issue with the pro version

gery0815, 2019-08-21:
I use screenly as primary software display kibana dashboards on big screens. Thank's to @Andr3wid to report the issue. I have this issue too since kibana 7.0. I already comb through the forums in the internet and finally i found this bug report. I urgently need a solution.

vpetersson, 2019-08-21:
@rusko124 can you prioritize this?

asussitz, 2019-08-22:
Finally, someone looks into this, I came across this problem in our office multiple times, thank you so much for looking into this!!!!

vpetersson, 2019-08-22:
Just an update on this: the issue appears to stem from that Kibana returns a 308 HTTP status. Presumably, this is a new thing. Since it is a somewhat rare redirect status, this has not been fully implemented.

mrauter, 2019-08-23:
We run into the same problem after upgrade to Elastic Stack 7.0, all our monitoring screens a white now. I think it has nothing to do with the redirect because we also get a blank screen if we configure the final Kibana URL after redirection.

vpetersson, 2019-08-23:
@mrauter, @rusko124 is experimenting with this now. Hope to have an update soon.

kendall-link, 2019-08-28:
Also running into this issue.  Glad I'm not alone!  I'll keep checking this post for an update.  Thx!

bummzulu, 2019-09-10:
can we please get an upate in this matter? My company is also having licenses of the Pro version and we cannot use Screenly for all of hour Kibana dashboards.

vpetersson, 2019-09-10:
@bummzulu with regards to the commercial version of Screenly, please get in touch with [support.screenly.io](https://support.screenly.io). We're looking at this for OSE but cannot commit to a timeline.

bummzulu, 2019-09-10:
> @bummzulu with regards to the commercial version of Screenly, please get in touch with [support.screenly.io](https://support.screenly.io). We're looking at this for OSE but cannot commit to a timeline.

thanks for the hint. Will do.

SIM0N-F, 2019-10-07:
Hi, i have same trouble with my Metabases dashboards (blank page). Have you workarround for this ? 
Many Thanks in advance.

bummzulu, 2019-10-08:
Hi,

We did not get a workaround so far. Our screens on the shopfloor stay blank and we had to find a workaround with a different solution. We also have an active subscription for the pro version and opened a case to get help. So far no progress - so don’t spend the money on the pro version. We learned out of it.

vpetersson, 2019-10-09:
@bummzulu Sorry to hear that. I know the guys are looking at it for the commercial version, but I don't know what the timeline is right now. As far as OSE goes, it should be something we can look into after [Sprint 11](https://github.com/Screenly/screenly-ose/projects/4) when we've changed the browser. 

@SIM0N-F did you try the experimental branch?

Andr3wid, 2019-12-09:
Bump. Any update on this ?

qudy2001, 2019-12-19:
I have the same problem, any work around here?

Andr3wid, 2020-01-22:
Saw that "Make the experimental browser the default browser" in sprint 11is marked as done. Does that impact this issue @vpetersson? Anything new to try on our side ? I'm still interested in this issue.

vpetersson, 2020-01-22:
@Andr3wid Yes that will hopefully solve most browser rendering issues. Feel free to switch to the experimental branch and provide feedback.

closed 2020-03-29 by stale[bot]