Screenly/Anthias #1201: "Unable to resolve address"

jacobhweeks, 2019-08-26:
I am unable to resolve the webpage that I am trying to use with Screenly-OSE. the webpage is internal only. Could this be the reason? It loads on the devices in the network.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL "http://webclock._domain_.local
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version: 2B
* Screenly OSE Version: production@618864d

vpetersson, 2019-08-27:
@jacobhweeks Judging a project because you don't get a reply within 23 hours on a weekend? That feels a bit harsh. Screenly OSE is definitely not dead as you can see in the commit history (and activity in other tickets).

jacobhweeks, 2019-08-30:
Thanks, but Screenly links and weblinks didn't work and with Screenly being
as big as it is supposed to be and for professionals, one would think that
it would be able to load local network resources. However, it appears that
web pages must be made available to the public which would compromise my
security for what I am trying to do. From a security point of view, this is
a fail.

Thank you for your response.

vpetersson, 2019-08-30:
> weblinks didn't work and with Screenly being
as big as it is supposed to be and for professionals

Screenly OSE has a big user base and it's community driven. If you want commercial support, we do offer a commercial version of Screenly with commercial support (see [this page](https://www.screenly.io/screenly-ose-vs-pro/) for details).

> However, it appears that web pages must be made available to the public

No, this is not true. As long as the device itself can talk to the web server over the local network, that's sufficient. This is true for both the commercial version and for OSE. We have plenty of users using Screenly this way.

Re-reading the original comment, it is likely related to DNS. Try SSH'ing into the device and run the following:

```
$ http://webclock.domain.local
```

and 

```
$ curl http://webclock.domain.local
```

jacobhweeks, 2019-08-31:
I'll consider trying again. Thanks.