Screenly/Anthias #1212: Convert installation script into something more user friendly

vpetersson, 2019-09-11:
We've long relied on a regular text input for the installer. Perhaps we should migrate it to [whiptai](https://linux.die.net/man/1/whiptail) to make it more user friendly.

rchadgray, 2020-02-26:
I started this migration to whiptail.  I am still testing the install.sh file.  

I have done the fork correctly that you can get my edited file once i am happy with it?  I have not done much team development in GitHub.  I just wanted to check before i went any further with edits.

vpetersson, 2020-02-27:
Splendid. Thanks @rchadgray. Take a look at [these instructions](https://gist.github.com/Chaser324/ce0505fbed06b947d962) for a good crash course on how to collaborate on GitHub.