Screenly/Anthias #1222: Black screen after a day and a half

billgod, 2019-09-18:
I have looked around and found many black screen issues but non of them seem to match what I have.  My screenly would run for about a day and half sometimes 2 days then black screen.  I just downloaded the latest 2 days ago, new sd card, different pi 3 and a different power supply.  After about a day and a half screen went black again.  The screen is on and back lit.  just black.  Has anyone else suffered this?  I find it odd that 2 version, 2 pies same problem?

I am only displaying an image and 2 urls to google calendars.

ALSO.. PLEASE add a reboot option to the web interface!

When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version:
* Screenly OSE Version:

rusko124, 2019-09-23:
Hey @billgod !
Could you provide the logs, please? 
Command: `journalctl -u screenly-viewer.service`

mobilebob, 2019-09-23:
I am having the same problem.  Running the command for logs shows no logs.  A reboot fixes the problem.

Started out with a fresh copy and ran the upgrade process on 9-20-19

ealmonte32, 2019-09-23:
@billgod 

Yeah i agree with rusko, without running the `journalctl` command and showing logs, there is no way to tell what is causing the black screen.. could be so many things..

Regarding the web interface and reboot button..
Screenly has a reboot option, have you installed the latest version?

[image]

ealmonte32, 2019-09-27:
@billgod 
I am seeing this as well on production branch, I have logs from my own Pi and will troubleshoot and see what i find.. the only thing i think i've done to my Pi differently is i use ZRAM to try and optimize the memory usage on the Pi, but i noticed an out of memory error on the log and that could maybe be the cause on mine.. maybe, maybe not, just saying this because it said it closed the parent process of uzbl browser (uzbl-core)..

```
Sep 24 08:47:15 rpi300we kernel: Out of memory: Kill process 1111 (uzbl-core) score 518 or sacrifice child
Sep 24 08:47:15 rpi300we kernel: Killed process 1111 (uzbl-core) total-vm:1155412kB, anon-rss:234688kB, file-rss:0kB, shmem-rss:0kB
```

will see what i find through the logs.

TehBeef, 2019-09-27:
I'm having this same issue, I have it deployed on BalenaCloud and I get a black screen. The log gives me this:

`27.09.19 15:32:47 (-0400)  screenly-viewer  Showing asset 1204743365.mp4 (video)
27.09.19 15:32:47 (-0400)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
27.09.19 15:32:48 (-0400)  screenly-viewer  Resource URI is not correct, remote host is not responding or request was rejected.`

I've re-installed everything and PNG's and a few url's seem to work, but other then that most things seem to not want to load and I'm stuck. I've also had this issue installing screenly directly on the Pi as the sole software. I'm on the Pi 3 1gb ram.

ealmonte32, 2019-09-27:
> `27.09.19 15:32:47 (-0400) screenly-viewer Showing asset 1204743365.mp4 (video) 27.09.19 15:32:47 (-0400) screenly-viewer current url is file:///tmp/screenly_html/black_page.html 27.09.19 15:32:48 [...]

Just to confirm:
So if you dont add the video it doesn't go black?
Can you try and look through the logs more for any errors that are not explicitly from screenly since it does interact with other services of the raspbian OS.. maybe we might have some in common..

I'm assuming you are looking at the System info logs right? if so, can you ssh into the Pi and run `journalctl` , you will most likely need to press space bar to go through the logs one page at a time, but once you reach the end, you should be able to use a mouse to scroll up and see if you can find any errors or something before it started doing it..

TehBeef, 2019-09-28:
Looked through all of logs and I couldn't find anything. It just seems like it's not pulling from the right directory when I add the video. Maybe it's me being stupid and not configuring something right but so far it doesn't seem like that. (It's been a year since the last time I played with anything like this or linux)

billgod, 2019-09-29:
Sorry I have been out of town.  Can you please provide me with a link on how to ssh into the box?  There is no SSH port open and not sure how to enable it.

rusko124, 2019-09-30:
Hey @TehBeef !
Did you increase gpu_mem to 64(min) in device settings ?
[image]

rusko124, 2019-09-30:
> @billgod
> I am seeing this as well on production branch, I have logs from my own Pi and will troubleshoot and see what i find.. the only thing i think i've done to my Pi differently is i use ZRAM to [...]

Good find!
I think It's also related to https://github.com/Screenly/screenly-ose/issues/1211

TehBeef, 2019-09-30:
> Hey @TehBeef !
> Did you increase gpu_mem to 64(min) in device settings ?
> [image]

I had it set to 124 at one point and I had the same issues, I'm not that education in the full extent of that does so I thought I'd leave it off as of now for the sake of troubleshooting.

ealmonte32, 2019-09-30:
gpu_mem just assigns the amount of memory to the graphics card of the Pi, 124 is not really a good size to assign because it is not an exponent of 2..I suggest 128 as rusko mentions, this can be easily done either via the boot config file or through the memory split section raspi-config..

ealmonte32, 2019-09-30:
@TehBeef 
I see you are using balena, the steps to change gpu_mem are different when running a docker container.. here are the instructions:
https://www.balena.io/docs/reference/OS/advanced/

Also, I didnt see any memory errors on your logs, which is interesting because that would rule out it being a memory error.. hmm..

---

@billgod 
here are instructions for SSH into a Pi:
https://github.com/Screenly/screenly-ose/wiki#enabling-ssh

ealmonte32, 2019-09-30:
> Looked through all of logs and I couldn't find anything. It just seems like it's not pulling from the right directory when I add the video. Maybe it's me being stupid and not configuring something [...]

What do you mean pulling from the right directory, how are you uploading/assigning the video assets?

TehBeef, 2019-09-30:
> gpu_mem just assigns the amount of memory to the graphics card of the Pi, 124 is not really a good size to assign because it is not an exponent of 2..I suggest 128 as rusko mentions 64 is minimum, this can be easily done either via the boot config file or through the memory split section raspi-config..

In all honestly I think it was 128, I just can't remember for sure. I'm out of town so my mind is elsewhere.

rusko124, 2019-10-01:
@vpetersson 
A little bit of research:
1.
[image]
2.
[image]

For 3 hours uzbl-core ate 100MB

Guys, This issue should be resolved after this https://github.com/Screenly/screenly-ose/issues/1093
This is the main task of the current sprint https://github.com/Screenly/screenly-ose/projects/4

vpetersson, 2019-10-01:
Yeah, I guess we could in theory add a memory watchdog, but given that we're deprecating UZBL i don't think it's worth the effort. Good research tho @rusko124.

TehBeef, 2019-10-02:
I'm a bit uneducated here, mind pointing me in the right direction on how to switch to the experimental browser? I'll see if that was the issue once switched out.

ealmonte32, 2019-10-02:
> I'm a bit uneducated here, mind pointing me in the right direction on how to switch to the experimental browser? I'll see if that was the issue once switched out.

One way, you can just re-run the screenly install script and when it asks if you want to install the experimental branch you say yes by typing y..
```
bash <(curl -sL https://www.screenly.io/install-ose.sh)
```

ealmonte32, 2019-10-02:
@rusko124 
After 4 days and removed zram use there has still been out of memory errors, but no black screen so far... although I did notice rabbitmq showing process taking up 127M of memory and the time spent in kernel to be very long (time in red)... any ideas?

[image]

[image]

rusko124, 2019-10-03:
@ealmonte32 
Yes, I also ran into a memory issue with celery.
In search found that at people solved the issue by celery updating to 4.3.0. Currently I testing this, I'll let you know if it helps.

ealmonte32, 2019-10-03:
```
Oct 02 08:36:42 screenlyatrium kernel: Out of memory: Kill process 11531 (uzbl-core) score 567 or sacrifice child
Oct 02 23:25:41 screenlyatrium kernel: [<8023c358>] (oom_kill_process) from [<8023cc88>] (out_of_memory+0x134/0x36c)
Oct 02 23:25:41 screenlyatrium kernel: [<8023cc88>] (out_of_memory) from [<802430e0>] (__alloc_pages_nodemask+0x1024/0x1178)
Oct 02 23:25:41 screenlyatrium kernel: Tasks state (memory values in pages):
Oct 02 23:25:41 screenlyatrium kernel: Out of memory: Kill process 23997 (uzbl-core) score 569 or sacrifice child
Oct 03 06:55:26 screenlyatrium celery[1062]: [2019-10-03 06:55:25,971: ERROR/ForkPoolWorker-4] Task server.cleanup[5df510bb-e7b9-4826-a463-c75db831ba82] raised unexpected: OSError(12, 'Cannot allocate memory')
Oct 03 06:55:26 screenlyatrium celery[1062]: OSError: [Errno 12] Cannot allocate memory
pi@screenlyatrium:~ $
```

Now the actual celery worker reported not being able to allocate enough memory..
I'll update to see..

---

@TehBeef try the following command to see if it also helps you:

`sudo pip install --upgrade --no-deps celery`

ealmonte32, 2019-10-03:
updated celery to 4.3.0 and still getting memory eaten up by uzbl-core:

```
pi@screenly600w:~ $ journalctl | grep memory
Oct 03 11:52:35 screenly600w kernel: Virtual kernel memory layout:
Oct 03 11:52:35 screenly600w kernel: vc-sm: Videocore shared memory driver
Oct 03 11:52:35 screenly600w kernel: Freeing unused kernel memory: 1024K
Oct 03 11:52:35 screenly600w kernel: bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Oct 03 13:39:34 screenly600w celery[1107]: [2019-10-03 13:39:32,688: ERROR/ForkPoolWorker-5] Task server.cleanup[76971243-fdaa-455f-b04b-8c13906af81f] raised unexpected: OSError(12, 'Cannot allocate memory')
Oct 03 13:39:34 screenly600w celery[1107]: OSError: [Errno 12] Cannot allocate memory
Oct 03 14:37:25 screenly600w celery[1107]: [2019-10-03 14:37:25,612: ERROR/ForkPoolWorker-5] Task server.cleanup[182921c6-1f3b-46b5-bd42-c37ef019b808] raised unexpected: OSError(12, 'Cannot allocate memory')
Oct 03 14:37:25 screenly600w celery[1107]: OSError: [Errno 12] Cannot allocate memory
Oct 03 15:37:25 screenly600w celery[1107]: [2019-10-03 15:37:25,543: ERROR/ForkPoolWorker-5] Task server.cleanup[30e1c42f-9639-437e-ad02-b97a543d8469] raised unexpected: OSError(12, 'Cannot allocate memory')
Oct 03 15:37:25 screenly600w celery[1107]: OSError: [Errno 12] Cannot allocate memory
Oct 03 16:37:25 screenly600w celery[1107]: [2019-10-03 16:37:25,809: ERROR/ForkPoolWorker-5] Task server.cleanup[c5ac6d38-992e-4a5f-9922-b4dadcf5d79c] raised unexpected: OSError(12, 'Cannot allocate memory')
Oct 03 16:37:25 screenly600w celery[1107]: OSError: [Errno 12] Cannot allocate memory
```

```
pi@screenly600w:~ $ pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package                 Version    
----------------------- -----------
celery                  4.3.0    
```

ealmonte32, 2019-10-30:
Just an update, the Xorg process seems to also be affected which indicates issue with video drivers or video rendering, and I thought, hmm.. since uzbl does not do GPU rendering by default, maybe it needs to be enabled in the variables by adding:
`set enable_3d_acceleration = 1` in the ./.config/uzbl/config-screenly file..
Maybe this can help?

vpetersson, 2019-11-04:
ping @rusko124

ealmonte32, 2019-11-05:
I thought about it and maybe you're right, there shouldn't be time spent on tweaking uzbl anyway..
I think for anyone that wants to have a quick workaround a simple cron job of
`systemctl restart screenly-viewer.service` will keep it running throughout the day/week..
I have been testing and have all screenly pi's restarting the service every 6 hours and I have not seen a black screen since.

I was also seeing rabbitmq(amqp) causing high memory usage so I disabled that service as well..

rusko124, 2019-11-05:
@vpetersson I'm waiting #1239
About RabbitMQ I will try to replace RabbitMQ in favor of Redis and set the expire time for results to half a day.
What do you think?

vpetersson, 2019-11-05:
@rusko124 I prefer redis over rabbit due to it's smaller footprint in this environment.

rusko124, 2019-11-07:
Hey @ealmonte32 
In recent changes Screenly uses Redis instead of Rabbitmq.
After 2 days, I doesn't notice any memory leak.

ealmonte32, 2019-11-07:
@rusko124 thanks, will begin testing today..

dingetje, 2019-11-11:
For what it's worth, just want to report the same issue. Black screen after random amount of time. The journal log shows **frequent** Celery errors:

`Nov 08 18:27:25 screenly celery[20560]: [2019-11-08 18:27:25,127: ERROR/MainProcess] consumer: Cannot connect to amqp://guest:**@127.0.0.1:5672//: [Errno 111] Connection refused.`
`Nov 08 18:27:25 screenly celery[20560]: Trying again in 32.00 seconds...`

For now I've added a cronjob to restart the service every 6 hours

vpetersson, 2019-11-11:
@dingetje Are you running on latest master changes?

dingetje, 2019-11-11:
@vpetersson no, I'm running the latest release version. Not sure where to find the version number...

vpetersson, 2019-11-11:
I see. While it might break things, you might want to switch to the development version as we are working on resolving this issue. However, if you're on a Pi Zero, there is a known issue right now so you might need to wait a bit.

ealmonte32, 2019-11-12:
@rusko124 
When users re-run the install screenly-ose script, and they are running on top of previous install, maybe we need to remind them to `sudo apt purge rabbitmq-server -y` so that it removes the previous service..

This is of course low priority but I am trying to think of a way that re-running the install script via terminal or upgrading via web can remove previous services that were changed to something else and are no longer used.. but then again this will require constant keeping up with packages removed ... ugh no easy way to do this..  =[

rusko124, 2019-11-12:
Hey @ealmonte32 !
Yes, that's a good point. But it is already provided :)
https://github.com/Screenly/screenly-ose/blob/master/ansible/roles/system/tasks/main.yml#L177

ealmonte32, 2019-11-12:
@rusko124 
But when I ran the install script and kept the production version, the rabbitmq service was still running (after installation finished and reboot).. so something didnt happen during the ansible task..

ealmonte32, 2019-11-12:
I could test by manually installing the rabbitmq-server package now, and re-running the `./screenly/bin/run_upgrade.sh` file to show you that after installation finishes, upon reboot you can still find the rabbitmq.service running..

ealmonte32, 2019-11-13:
ugh, i keep getting confused with version named production vs one named development aka master branch... doh, can we add the version name of master next to the development question?

when it asks if we want to use the latest development version, can we add in parentheses (master branch)?

vpetersson, 2019-11-13:
Sensible. I've pushed a fix.

ealmonte32, 2019-11-13:
@dingetje 
If you run the install script again and select yes for `development version`, it has been tested, and it fixes the memory error.

Maybe we can close this issue now.. unless you notice that it breaks any of the other assets you have?..

P.S. I notice you said you have old rpi, can i suggest you install [ZRAM](https://github.com/novaspirit/rpi_zram) on it?
___
Running install script again:
`./screenly/bin/install.sh`

ealmonte32, 2019-11-13:
@rusko124 

the master branch still shows uzbl-core memory errors:
```
pi@screenlypi:~ $ journalctl | grep -i memory
Nov 13 13:05:17 screenlypi kernel: Memory: 886300K/917504K available (8192K kernel code, 653K rwdata, 2224K rodata, 1024K init, 822K bss, 23012K reserved, 8192K cma-reserved)
Nov 13 13:05:17 screenlypi kernel: Virtual kernel memory layout:
Nov 13 13:05:17 screenlypi kernel: vc-sm: Videocore shared memory driver
Nov 13 13:05:17 screenlypi kernel: Freeing unused kernel memory: 1024K
Nov 13 13:05:18 screenlypi kernel: bcm2835_vc_sm_cma_probe: Videocore shared memory driver
Nov 13 13:40:16 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 13 13:40:16 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 13 13:40:16 screenlypi kernel: Tasks state (memory values in pages):
Nov 13 13:40:16 screenlypi kernel: Out of memory: Kill process 625 (uzbl-core) score 657 or sacrifice child
Nov 13 14:32:45 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 13 14:32:45 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 13 14:32:45 screenlypi kernel: Tasks state (memory values in pages):
Nov 13 14:32:45 screenlypi kernel: Out of memory: Kill process 1764 (uzbl-core) score 660 or sacrifice child
Nov 13 15:07:11 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 13 15:07:11 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 13 15:07:11 screenlypi kernel: Tasks state (memory values in pages):
Nov 13 15:07:11 screenlypi kernel: Out of memory: Kill process 3041 (uzbl-core) score 638 or sacrifice child
Nov 13 15:38:30 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 13 15:38:30 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 13 15:38:30 screenlypi kernel: Tasks state (memory values in pages):
Nov 13 15:38:30 screenlypi kernel: Out of memory: Kill process 4501 (uzbl-core) score 632 or sacrifice child
```

The production version with removed celery service does not show me any memory errors at all, which i have set this way in 3 more Pi's.
I will test now by disabling the celery worker from this one: 
```
pi@screenlypi:~ $ cat ./version.md
Screenly version: master@bc64b08
```
which will then determine if the celery worker is in fact causing the memory error, but what does celery worker have to do with uzbl? 🤦‍♂

ealmonte32, 2019-11-14:
@rusko124 

on this screenshot Xorg is invoking the oom-killer because of low memory, and of course uzbl-core is always going to be the process that is using the most memory and this will be why it gets killed.. i assume that when celery was also consuming all the memory, the system did the same and killed uzl-core..

i am comparing two screenly on the same pi but with different branches, and the Xorg (same package version) is spending a lot of time in the CPU, whether on the master branch
or production.. in the master branch i get the following:
```
Nov 13 18:34:58 screenlypi kernel: Xorg invoked oom-killer: gfp_mask=0x6000c0(GFP_KERNEL), nodemask=(null), order=0, oom_score_adj=0
Nov 13 18:34:58 screenlypi kernel: Xorg cpuset=/ mems_allowed=0
Nov 13 18:34:58 screenlypi kernel: CPU: 0 PID: 461 Comm: Xorg Tainted: G         C        4.19.83-v7+ #1277
Nov 13 18:34:58 screenlypi kernel: Hardware name: BCM2835
Nov 13 18:34:58 screenlypi kernel: [<801120c0>] (unwind_backtrace) from [<8010d5f4>] (show_stack+0x20/0x24)
Nov 13 18:34:58 screenlypi kernel: [<8010d5f4>] (show_stack) from [<80844280>] (dump_stack+0xd4/0x118)
Nov 13 18:34:58 screenlypi kernel: [<80844280>] (dump_stack) from [<8026124c>] (dump_header+0x80/0x250)
Nov 13 18:34:58 screenlypi kernel: [<8026124c>] (dump_header) from [<802605b4>] (oom_kill_process+0x358/0x3a8)
Nov 13 18:34:58 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 13 18:34:58 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 13 18:34:58 screenlypi kernel: [<8026734c>] (__alloc_pages_nodemask) from [<802674c4>] (__get_free_pages+0x24/0x70)
Nov 13 18:34:58 screenlypi kernel: [<802674c4>] (__get_free_pages) from [<80347824>] (proc_pid_cmdline_read+0x1d0/0x3c8)
Nov 13 18:34:58 screenlypi kernel: [<80347824>] (proc_pid_cmdline_read) from [<802cfd9c>] (__vfs_read+0x48/0x16c)
Nov 13 18:34:58 screenlypi kernel: [<802cfd9c>] (__vfs_read) from [<802cff5c>] (vfs_read+0x9c/0x168)
Nov 13 18:34:58 screenlypi kernel: [<802cff5c>] (vfs_read) from [<802d05a4>] (ksys_read+0x74/0xe8)
Nov 13 18:34:58 screenlypi kernel: [<802d05a4>] (ksys_read) from [<802d0630>] (sys_read+0x18/0x1c)
Nov 13 18:34:58 screenlypi kernel: [<802d0630>] (sys_read) from [<80101000>] (ret_fast_syscall+0x0/0x28)
Nov 13 18:34:58 screenlypi kernel: Exception stack(0xb4ef7fa8 to 0xb4ef7ff0)
Nov 13 18:34:58 screenlypi kernel: 7fa0:                   7ea83af0 011c5550 00000019 7ea83af0 00001001 00000000
Nov 13 18:34:58 screenlypi kernel: 7fc0: 7ea83af0 011c5550 76f40ce8 00000003 7ea83af8 00000001 00000019 00201a00
Nov 13 18:34:58 screenlypi kernel: 7fe0: 00000000 7ea83ae0 00000000 76a86414
Nov 13 18:34:58 screenlypi kernel: Mem-Info:
Nov 13 18:34:58 screenlypi kernel: active_anon:30398 inactive_anon:30435 isolated_anon:0
                                    active_file:328 inactive_file:1071 isolated_file:32
                                    unevictable:3 dirty:0 writeback:0 unstable:0
                                    slab_reclaimable:2372 slab_unreclaimable:4770
                                    mapped:48389 shmem:48029 pagetables:1164 bounce:0
                                    free:4221 free_pcp:34 free_cma:791
Nov 13 18:34:58 screenlypi kernel: Node 0 active_anon:121592kB inactive_anon:121740kB active_file:1312kB inactive_file:4284kB unevictable:12kB isolated(anon):0kB isolated(file):128kB mapped:193556k
Nov 13 18:34:58 screenlypi kernel: Normal free:16884kB min:16384kB low:20480kB high:24576kB active_anon:121592kB inactive_anon:121740kB active_file:612kB inactive_file:4452kB unevictable:12kB write
Nov 13 18:34:58 screenlypi kernel: lowmem_reserve[]: 0 0
Nov 13 18:34:58 screenlypi kernel: Normal: 200*4kB (UMEHC) 251*8kB (UMEHC) 197*16kB (UMEHC) 86*32kB (UMEHC) 53*64kB (UEHC) 17*128kB (UEHC) 3*256kB (UEH) 3*512kB (UE) 1*1024kB (C) 0*2048kB 0*4096kB
Nov 13 18:34:58 screenlypi kernel: 49511 total pagecache pages
Nov 13 18:34:58 screenlypi kernel: 40 pages in swap cache
Nov 13 18:34:58 screenlypi kernel: Swap cache stats: add 1278737, delete 1278657, find 8904/603264
Nov 13 18:34:58 screenlypi kernel: Free swap  = 0kB
Nov 13 18:34:58 screenlypi kernel: Total swap = 895504kB
Nov 13 18:34:58 screenlypi kernel: 229376 pages RAM
Nov 13 18:34:58 screenlypi kernel: 0 pages HighMem/MovableOnly
Nov 13 18:34:58 screenlypi kernel: 5497 pages reserved
Nov 13 18:34:58 screenlypi kernel: 2048 pages cma reserved
Nov 13 18:34:58 screenlypi kernel: Tasks state (memory values in pages):
Nov 13 18:34:58 screenlypi kernel: [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Nov 13 18:34:58 screenlypi kernel: [    103]     0   103     3112        2    18432      102             0 systemd-journal
Nov 13 18:34:58 screenlypi kernel: [    290]     0   290     1918        0    16384      771             0 haveged
Nov 13 18:34:58 screenlypi kernel: [    294]   100   294     4300        0    20480      120             0 systemd-timesyn
Nov 13 18:34:58 screenlypi kernel: [    349]   105   349     1640        2    18432      137          -900 dbus-daemon
Nov 13 18:34:58 screenlypi kernel: [    354]     0   354     6900        0    18432       56             0 rngd
Nov 13 18:34:58 screenlypi kernel: [    361]     0   361    13256        0    38912      347             0 ModemManager
Nov 13 18:34:58 screenlypi kernel: [    362]     0   362     1845        0    16384      121             0 systemd-logind
[... cut ...]
but on the production i do not get the oom-killer invoked, but the Xorg does stay in CPU for long time and becomes red.

[image]
___

On a raspberry pi that is not running screenly, i have it running chromium on the solaredge site, and i just noticed the GPU flag process on chromium spending time on CPU as well.. so this is all probably GPU related.. but that rpi is running buster and not stretch, the xserver-xorg version is just +b8 of difference.. i am trying to think of how to best approach this issue since it is not specifically screenly related.. mostly to do with handling of memory.. 🤦‍♂ 

[image]

vpetersson, 2019-11-15:
What memory split are you using?

ealmonte32, 2019-11-15:
so for testing I changed from my normal default of 128 to 256, it crashed even faster, went down to 64, has been much stabler... still not seen black screen in master branch.

getting python invoking oom-killer but the kernel takes care of restarting all processes so the user would not see anything unless they go looking through logs..
___

i am pasting the following logs just for documenting, i dont expect anyone to read through them all but just for documentation.. 

```
Nov 14 20:30:59 screenlypi python[621]: current url is https://docs.google.com/presentation/d/e/XXXXXXXX/pub?start=true
Nov 14 20:30:59 screenlypi python[621]: Sleeping for 60
Nov 14 20:32:20 screenlypi kernel: python invoked oom-killer: gfp_mask=0x7080c0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), nodemask=(null), order=1, oom_score_adj=0
Nov 14 20:32:20 screenlypi kernel: python cpuset=/ mems_allowed=0
Nov 14 20:32:20 screenlypi kernel: CPU: 3 PID: 621 Comm: python Tainted: G         C        4.19.83-v7+ #1277
Nov 14 20:32:20 screenlypi kernel: Hardware name: BCM2835
Nov 14 20:32:20 screenlypi kernel: [<801120c0>] (unwind_backtrace) from [<8010d5f4>] (show_stack+0x20/0x24)
Nov 14 20:32:20 screenlypi kernel: [<8010d5f4>] (show_stack) from [<80844280>] (dump_stack+0xd4/0x118)
Nov 14 20:32:20 screenlypi kernel: [<80844280>] (dump_stack) from [<8026124c>] (dump_header+0x80/0x250)
Nov 14 20:32:20 screenlypi kernel: [<8026124c>] (dump_header) from [<802605b4>] (oom_kill_process+0x358/0x3a8)
Nov 14 20:32:20 screenlypi kernel: [<802605b4>] (oom_kill_process) from [<80260ee4>] (out_of_memory+0x134/0x37c)
Nov 14 20:32:20 screenlypi kernel: [<80260ee4>] (out_of_memory) from [<8026734c>] (__alloc_pages_nodemask+0x1024/0x1178)
Nov 14 20:32:20 screenlypi kernel: [<8026734c>] (__alloc_pages_nodemask) from [<8011e008>] (copy_process.part.5+0x1f8/0x1b08)
Nov 14 20:32:20 screenlypi kernel: [<8011e008>] (copy_process.part.5) from [<8011fae0>] (_do_fork+0xd8/0x438)
Nov 14 20:32:20 screenlypi kernel: [<8011fae0>] (_do_fork) from [<8011ff64>] (sys_clone+0x34/0x3c)
Nov 14 20:32:20 screenlypi kernel: [<8011ff64>] (sys_clone) from [<80101000>] (ret_fast_syscall+0x0/0x28)
Nov 14 20:32:20 screenlypi kernel: Exception stack(0xb61ddfa8 to 0xb61ddff0)
Nov 14 20:32:20 screenlypi kernel: dfa0:                   76f5d068 7ea11638 01200011 00000000 00000000 00000000
Nov 14 20:32:20 screenlypi kernel: dfc0: 76f5d068 7ea11638 00000000 00000078 00000001 76e14000 00000000 7ea11674
Nov 14 20:32:20 screenlypi kernel: dfe0: 76f5d000 7ea11638 76d752e8 76d75178
Nov 14 20:32:20 screenlypi kernel: Mem-Info:
Nov 14 20:32:20 screenlypi kernel: active_anon:28454 inactive_anon:28460 isolated_anon:0
                                    active_file:119 inactive_file:195 isolated_file:27
                                    unevictable:3 dirty:0 writeback:0 unstable:0
                                    slab_reclaimable:2321 slab_unreclaimable:4982
                                    mapped:11248 shmem:39916 pagetables:1175 bounce:0
                                    free:6037 free_pcp:424 free_cma:1439
Nov 14 20:32:20 screenlypi kernel: Node 0 active_anon:113816kB inactive_anon:113840kB active_file:476kB inactive_file:780kB unevictable:12kB isolated(anon):0kB isolated(file):0kB mapped:44992kB dir
Nov 14 20:32:20 screenlypi kernel: Normal free:24148kB min:16384kB low:20480kB high:24576kB active_anon:113816kB inactive_anon:113840kB active_file:44kB inactive_file:384kB unevictable:12kB writepe
Nov 14 20:32:20 screenlypi kernel: lowmem_reserve[]: 0 0
Nov 14 20:32:20 screenlypi kernel: Normal: 319*4kB (UMEHC) 626*8kB (UMEHC) 181*16kB (UMEHC) 63*32kB (UMEHC) 33*64kB (UMEHC) 10*128kB (UEH) 5*256kB (UHC) 3*512kB (UHC) 2*1024kB (UE) 0*2048kB 1*4096k
Nov 14 20:32:20 screenlypi kernel: 40321 total pagecache pages
Nov 14 20:32:20 screenlypi kernel: 53 pages in swap cache
Nov 14 20:32:20 screenlypi kernel: Swap cache stats: add 1165333, delete 1165248, find 8615/520494
Nov 14 20:32:20 screenlypi kernel: Free swap  = 0kB
Nov 14 20:32:20 screenlypi kernel: Total swap = 960464kB
Nov 14 20:32:20 screenlypi kernel: 245760 pages RAM
Nov 14 20:32:20 screenlypi kernel: 0 pages HighMem/MovableOnly
Nov 14 20:32:20 screenlypi kernel: 5641 pages reserved
Nov 14 20:32:20 screenlypi kernel: 2048 pages cma reserved
Nov 14 20:32:20 screenlypi kernel: Tasks state (memory values in pages):
Nov 14 20:32:20 screenlypi kernel: [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Nov 14 20:32:20 screenlypi kernel: [    104]     0   104    10315        0    24576      108             0 systemd-journal
Nov 14 20:32:20 screenlypi kernel: [    274]     0   274     1918        2    16384      771             0 haveged
Nov 14 20:32:20 screenlypi kernel: [    276]   100   276     4300        0    18432      111             0 systemd-timesyn
Nov 14 20:32:20 screenlypi kernel: [    322]     0   322     5939        0    22528      241             0 rsyslogd
Nov 14 20:32:20 screenlypi kernel: [    325]   105   325     1624        2    16384      114          -900 dbus-daemon
Nov 14 20:32:20 screenlypi kernel: [    350]  1000   350    14954      253    40960     2249             0 python
Nov 14 20:32:20 screenlypi kernel: [    351]     0   351     1324        0    12288       57             0 cron
Nov 14 20:32:20 screenlypi kernel: [    355]     0   355     1845        0    16384      112             0 systemd-logind
[... cut ...]
Nov 14 20:32:26 screenlypi python[4354]:   RequestsDependencyWarning)
Nov 14 20:32:28 screenlypi python[4354]: Generating asset-list...
Nov 14 20:32:28 screenlypi python[4354]: browser found dead, restarting
Nov 14 20:32:28 screenlypi python[4354]: Loading browser...
Nov 14 20:32:28 screenlypi python[4354]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 4372.
Nov 14 20:32:30 screenlypi python[4354]: current url is file:///tmp/screenly_html/black_page.html
Nov 14 20:32:31 screenlypi python[4354]: Generating asset-list...
```

Jrdnkasparek, 2019-11-15:
This is also happening with our 3 Raspberry Pi 3s we have at our district

ealmonte32, 2019-11-15:
@Jrdnkasparek  A few questions:

1. when you look at the logs, are you looking after the black screen happens, or since you are saying it becomes unresponsive and cant ssh, are you manually rebooting and then checking the logs? because as far as i know unless you are doing something different, that would of course not show previous logs due to reboot.

_(i can suggest a simple checkwifi.sh script that would check if the pi has internet connectivity, and if not, it restarts the networking.service and screenly-viewer.service so that it sort of tries to repair itself)_

2. did you try at least on one to run screenly install (`./screenly/bin/install.sh`) and select the development branch/version?
3. if you run a `systemctl status` do you see any of the following there: rabbitmq, celery, running?

thx

ealmonte32, 2019-11-18:
Black screen on master branch still happening..

Pressing Reboot in WebUI did not restart screenly viewer which should have temporarily resolved the issue, and checking syslog only showed this:
```
Nov 18 08:59:23 screenlypi sudo[30305]:       pi : TTY=unknown ; PWD=/home/pi/screenly ; USER=root ; COMMAND=/bin/systemctl status screenly-viewer.service -n 20
Nov 18 08:59:23 screenlypi sudo[30305]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 18 08:59:23 screenlypi sudo[30305]: pam_unix(sudo:session): session closed for user root
Nov 18 09:00:08 screenlypi sudo[30347]:       pi : TTY=unknown ; PWD=/home/pi/screenly ; USER=root ; COMMAND=/bin/systemctl status screenly-viewer.service -n 20
Nov 18 09:00:11 screenlypi sudo[30347]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 18 09:00:31 screenlypi sudo[30347]: pam_unix(sudo:session): session closed for user root
```

Will try disabling swap memory to see if system flushes out when needed.

[image]

Gifford47, 2019-11-22:
is there already a fix for the stable version?

vpetersson, 2019-11-22:
@Gifford47 No, we don't do releases until the end of the sprint, but please verify that it solves the issue for you by switching to the development version.

Gifford47, 2019-11-25:
i have the version master@e301484 and after one day i still get a black screen... the ram is also continuously fully described. free ram at screenly start: 724mb
free ram after 1 day: 84mb ram

Gifford47, 2019-11-25:
i am at the development branch.
here is the log
[Uploading screenly_log_25.11.19.txt…](screenly log)
[Uploading memory.txt…](and the memory list)

rusko124, 2019-11-26:
@Gifford47 
Interestingly, most of the memory is occupied by docker and uzbl, which looks contradictory. Since one must exclude the other. I assume we are dealing with 2 viewers at once. Cause of this is switching between the master and the experimental branch.

Also I see that the Screenly ansible-playbook doesn't have a cleanup of Docker when choose the development branch after experimental.

Not sure we need to create PR to fix that so the Screenly will be migrate to experimental browser as default.
Could you one more install the last Screenly image to SD and upgrade to the development branch without touching the experimental branch, please?

Gifford47, 2019-11-26:
Sure! I'll try and post the results this afternoon.

Gifford47, 2019-12-01:
thanks a lot for the answer, now it works for me too! but screenly reserves about 230mb in total. is that normal? from time to time it also drops 150mb (e.g. when restarting screenly).
is that normal and can I still optimize it?

[image]

ealmonte32, 2019-12-04:
> thanks a lot for the answer, now it works for me too! but screenly reserves about 230mb in total. is that normal? from time to time it also drops 150mb (e.g. when restarting screenly).
> is that normal and can I still optimize it?

Remember the viewer is always running and calls upon the uzbl-core to display the assets and omxplayer for videos etc, so the memory will vary a lot depending on your assets and the page load.. i think the memory issue is worse when it cant be swapped and oom-killer kills the browser but for some reason the viewer is not restarting the dead browser automatically which makes the screen either stuck or black, so the viewer needs to also restart itself instead of just uzbl via `systemctl restart screenly-viewer.service`..

@rusko124 the screenshot below was taken on latest screenly image production with manual celery upgrade.. results are bad, look at locked processes, never seen that before... i will upgrade to development and test that branch to see if better memory management is handled..

[image]

ealmonte32, 2019-12-04:
update:
zram modified too high (1.5x built-in memory) is not working well on the RPi 3...
it slows the pi to a crawl after some time.. and shows all these locked processes..

ealmonte32, 2019-12-05:
so testing development branch with load heavy sites like news.google.com and videos and google slides.. and without swap memory the browser just stops, this time instead of black screen it's white, if the viewer somehow knew it was in a "dead" or "failed" state itself, restarting the viewer fixes the issue and no need to reboot the Pi.. when swap is used, and all memory is allocated, the Pi functions less and every command in terminal crawls to execute..

hmm... i thought the viewer was capable of finding the browser dead and restarting it, seems like it is not doing it..
___

screenshots for documentation:

[image]

[image]

[image]

vpetersson, 2019-12-05:
> and without swap memory the browser just stops, this time instead of black screen it's white, if the viewer somehow knew it was in a "dead" or "failed" state itself,

A white screens means that UZBL started, but for whatever reason, `viewer.py` was unable to instruct UZBL to switch to the default black screen. This is likely because there is no more memory available. 

So taking a step back, it's likely that because we've now added more components (celery, rabbit/redis), we are now running out of memory.  My gut feeling tells me that this is either redis or rabbit that is eating up the memory (given that they are databases). With the move to Docker, we can simply confine the memory using Docker metadata.

ealmonte32, 2019-12-19:
@vpetersson 
After much testing and trying to deal with the issue being the memory, to prove it, I tested by creating a script that runs chrome-browser like this:

```
#!/bin/bash

while [ ! -f /home/pi/screenly/STOP ]
do

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://news.google.com &
sleep 20;
pkill -f chromium-browser;
sleep 0.5;

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://news.ycombinator.com &
sleep 20;
pkill -f chromium-browser;
sleep 0.5;

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://www.nytimes.com & 
sleep 60;
pkill -f chromium-browser;
sleep 0.5;

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://www.cnn.com &
sleep 60;
pkill -f chromium-browser;
sleep 0.5;

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://weather.srly.io &
sleep 30;
pkill -f chromium-browser;
sleep 0.5;

DISPLAY=:0 chromium-browser --start-fullscreen -app=https://clock.srly.io &
sleep 30;
pkill -f chromium-browser;

done
#end

```
Yes, bad, archaic, but simple and good for testing this rotation of assets in order to generate memory consumption... chrome-browser as we know is a huge memory hog on the Pi, and some of these assets are very memory consuming, when I tried this same asset list on uzbl, it would crash the browser and make the Pi crawl to unusable state within an hour..

So..the results are.. that if killing the process before the next asset kept this working without black screen or freezing browser page, etc, means all we have to do is put in viewer.py a `pkill -f uzbl-core` after the end of every Asset loop (not after every asset, just once after the loop), that way every iteration has more memory to work with as oppose to how it's handled now.. besides that, as of right now, I cant think of anything else..

[4 more replies]

closed 2020-07-11 by stale[bot]