Screenly/Anthias #1223: Video failing to play

MydKnight, 2019-09-23:
#### Overview of the Issue

I just installed Screenly OSE on my BalenaOS RaspberryPI and video assets will not play. I read a bit online about codec issues possibly causing this and reencoded the video im using to be H264 and it is still not working. The log files in Balena state the following:

22.09.19 17:56:40 (-0700)  screenly-viewer  Showing asset Commander2018.png (image)
22.09.19 17:56:40 (-0700)  screenly-viewer  Sleeping for 10
22.09.19 17:56:51 (-0700)  screenly-viewer  Showing asset DeadMansParty264.mp4 (video)
22.09.19 17:56:51 (-0700)  screenly-viewer  current url is file:///tmp/screenly_html/black_page.html
22.09.19 17:56:52 (-0700)  screenly-viewer  Resource URI is not correct, remote host is not responding or request was rejected.
22.09.19 17:56:53 (-0700)  screenly-viewer  Showing asset Commander2018.png (image)
22.09.19 17:56:53 (-0700)  screenly-viewer  Sleeping for 10

The video log immediately moves on to the next step, never trying to play

#### Reproduction Steps

* Add Any MP4 to BalenaOS running Screenly
* Activate Asset
* Screen stays black

### Environment

* Raspberry Pi Hardware Version: BalenaOS v2.43.0
* Screenly OSE Version: Celery

rusko124, 2019-09-23:
Hey @MydKnight !
Omxplayer has issues playing video with the latest version of BalenaOS. Please try with BalenaOS v2.38.0

ealmonte32, 2019-09-23:
@rusko124 
thanks rusko, I was going to tell him that I tested without balena and screenly production and development version play h264 videos perfectly fine after I tested both branches today.

closed 2019-09-24 by vpetersson