Screenly/Anthias #1228: "Play for" options on Start/End Date not on Experimental

ealmonte32, 2019-09-26:
#### Overview of the Issue

The "Play for Week/Month/Forever" options are not available on the Experimental branch.

#### Reproduction Steps

Steps to reproduce this issue:

* Press Add Asset
* Enter the Asset URL or File upload ... Save.
* Click Edit to check on the "Play for Start/End Date" options

### Environment

* Screenly OSE Version: Experimental

ealmonte32, 2019-09-26:
* this is current experimental edit of asset options:

[image]

---

* this is production and development edit of asset options:

[image]

rusko124, 2019-09-27:
Hey @ealmonte32 !
In the current sprint our main point is to close the experimental branch. To do this, we need to make all features of the experimental branch as default.
https://github.com/Screenly/screenly-ose/issues/1093

ealmonte32, 2019-09-27:
gotcha.. sorry it's just I have all 3 branches in different SD cards in order to test everything on whichever one the issue needs to be tested on, and during one of my tests I saw that experimental branch didn't have this option, and i thought it would be a quick edit of the code, but all good then, i'll close this.

closed 2019-09-27 by ealmonte32