Screenly/Anthias #1232: Create with ID

gtrak, 2019-09-29:
#### Overview of the Issue

I have a use-case for creating assets with my own identifiers.  I see in https://github.com/Screenly/screenly-ose/issues/814 it was identified as something we _should not_ be doing, but I don't see another way to get what I want.  Essentially, I'm looking for a way to sync a playlist with an external relational database, so I can and wish to supply own identifiers.  I am diffing two lists and generating DELETE or POST calls according to the API to add or remove what I need.

I think this is all that's needed.  In server.py:
```
@@ -528,8 +528,18 @@ def prepare_asset_v1_2(request_environ, asset_id=None, unique_name=False):
         if not validate_url(uri):
             raise Exception("Invalid URL. Failed to add asset.")
 
+    # TODO: Custom hack, allow ourselves to create with an identifier
+    # so we can link back to our own playlists
+    # if not asset_id:
+    #     asset['asset_id'] = uuid.uuid4().hex
+    #     if uri.startswith('/'):
+    #         rename(uri, path.join(settings['assetdir'], asset['asset_id']))
+    #         uri = path.join(settings['assetdir'], asset['asset_id'])
+
+    # It seems to be abusing whether the ID is present to check if it needs to update
     if not asset_id:
-        asset['asset_id'] = uuid.uuid4().hex
+        if not asset['asset_id']:
+            asset['asset_id'] = uuid.uuid4().hex
         if uri.startswith('/'):
             rename(uri, path.join(settings['assetdir'], asset['asset_id']))
             uri = path.join(settings['assetdir'], asset['asset_id'])
```

I'm asking here because I'd like to avoid maintaining any custom patches.  Is there a better way to implement this over the existing API?

vpetersson, 2019-09-30:
@ghholm It's generally considered bad API design by not having some kind of UUID for an object like this (provided by the system). This is to avoid issues like collisions etc., so I don't think I want to go down that path. 

However, what I would suggest that you do is either to store the foreign ID (i.e. the asset id) in some some temporary sync database (or in the remote database). Alternatively, you could look it up by giving it a unique name, and then look up the UUID/asset id based on the name. Sure, it will involve another API call, but that's low price to pay instead of added complexity.

gtrak, 2019-09-30:
I think the name field gets overwritten by the youtube downloader code, so if I were to use the name field, I would lose out on that functionality.  In my own database design, we store a 'externalID' field for this case.  I'm not super interested in storing my own state at a process level, as I'm sideloading a sync script with this in a balena install, and it crashes often.

Another possible way to implement this feature is within the PUT assets/:id call, but it seems it is hardcoded to only update (I tried that, too).  A possibility is to upsert here instead. Would you be willing to accept a PR for that?

gtrak, 2019-09-30:
BTW I _am_ using uuids as my own identifiers, not sure if this changes anything.

vpetersson, 2019-09-30:
> Another possible way to implement this feature is within the PUT assets/:id call, but it seems it is hardcoded to only update (I tried that, too). A possibility is to upsert here instead. Would you be willing to accept a PR for that?

I guess using YouTube makes it different. Yes, a pull request that makes the API use the name provided rather than the name from YouTube makes sense. I don't have any objection with this, unless @rusko124 got a good reason.

gtrak, 2019-09-30:
I'd prefer to write up the PUT /:id create (not only update) PR instead of the youtube name, in general.  We also keep an asset name, and it's useful to be able to see it in the Screenly API, and I don't want to write more string parsers.

vpetersson, 2019-11-11:
@rusko124 WDYT?

closed 2020-01-17 by stale[bot]