Screenly/Anthias #1234: Sprint 10 does not boot on RPI4

shaqaruden, 2019-10-01:
#### Overview of the Issue

Sprint 10 image will not boot on RPI4

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Write .img to microSD card
* Power up Raspberry Pi

### Environment

* Raspberry Pi Hardware Version: RPI4
* Screenly OSE Version: Sprint 10

shaqaruden, 2019-10-01:
This is kind of ridiculous that Raspberry Pi has been out for almost 4 months and still now working ose image

vpetersson, 2019-10-02:
This is because one of the core components in Screenly OSE isn't available in Raspbian Buster (the browser uzbl). We're working around this, but it's a big effort and we're aiming to address this in [Sprint 11](https://github.com/Screenly/screenly-ose/projects/4) by adopting a new browser.

closed 2019-10-02 by vpetersson

shaqaruden, 2019-10-06:
So your development branch has the new browser? Because I can install and use Screenly OSE just fine if I start from a buster NOOBS image and install ScreenlyOSE manually

vpetersson, 2019-10-07:
@shaqaruden I don't think it's possible to to do yet on Buster. Ping @rusko124.

shaqaruden, 2019-10-07:
No I'm saying that I 100% have at least 7  Raspberry Pi 4s running Raspbian NOOBS Buster Lite with screenly-ose development branch installed and running with a webpage asset working. Regardless what is the timeline on Sprint 11?

vpetersson, 2019-10-09:
That's super interesting. I was under the impression that this was not possible. 

@rusko124 / @RemingtonA Do we have a timeline set for Sprint 11?