Screenly/Anthias #1238: Mounting OSE in Balena

vhgarciaa, 2019-10-08:
Hello, I  have a problem uploading Screenly OSE to my balenacloud because it gives me an error when doing Git Push and Balena push.

The error I leave them in photos, I tried to upload it from windows and from ubuntu and I always get the same error, I don't know if it is a screenly bug.

[https://drive.google.com/open?id=1-4TkrSUhS_rTRZ7S6X-f7SxW5HaX4oDn](url)

I hope you can help me, thank you very much

vpetersson, 2019-10-09:
Attaching screenshots directly here:

[image]
[image]
[image]

Looks like it is a build issue. Unclear what's causing it. @RemingtonA can you try reproducing this?

vhgarciaa, 2019-10-14:
Any progress with this? Thank you

jcbeebe21, 2019-10-24:
Same exact issue when trying a git push or balena push with the production branch.  Any workaround for this?

vpetersson, 2019-10-24:
Yes, there's a [pull request](https://github.com/Screenly/screenly-ose/pull/1239) that should address this. Feel free to give it a spin.

tho-wa, 2019-12-13:
Is there any fix right now?

vpetersson, 2019-12-14:
> Is there any fix right now?

Should have been fixed in master.

tho-wa, 2019-12-14:
Still running in same issue.

vpetersson, 2020-01-02:
@rusko124 can you see if you can reproduce this?

rusko124, 2020-01-06:
I'll investigate the issue in the next few days

rusko124, 2020-01-10:
The issue is reproducible for RPi4.
I suggest to close the current thread and continue the discussion here #1282.

On RPi3 everything works fine.
[image]
[image]

closed 2020-03-17 by stale[bot]