Screenly/Anthias #1240: Unable to resolve local IP address

bnupen, 2019-10-16:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.
Raspebrry Pi with Screenly-OSE Sprint 10 version.  Able to run Screenly when connected on wired network.  When booting Screenly without ethernet, I get the screen showing the SSID connection and password.  I am able to connect to the Pi and input the wireless network, the screenly resets and loads and then displays "Unable to resolve local IP address".  

I have tried re-flashing the card, also tried flashing another card and using a different Pi on a home network and end up with the same results.
I tried the bash <(curl -sL https://www.screenly.io/install-ose.sh)  suggestion, I could no longer use the keyboard on the Screenly pi and when the SSID connection screen came up, I logged into another computer to select the network, but nothing changed on the Screenly pi, it stayed on the SSID screen.

I really enjoy the Screenly ose software and appreciate all the work that has gone into it.  I just can't figure out how to get on the wifi.

Raspian GNU/Linux 9

* Raspberry Pi Hardware Version: 3 Model B Version 1.2
* Screenly OSE Version: Sprint 10

the-ft, 2019-10-17:
hey there.. 
did you tried the suggestions form ealmonte32 from [here](https://forums.screenly.io/t/screenly-displays-unable-to-resolve-local-ip-address-when-moved-to-new-enviro/41/2) ?

i m struggling too, but on different errors.

ealmonte32, 2019-11-22:
@bnupen I will test this now with the latest image from here https://github.com/Screenly/screenly-ose/releases

will let you know what i find.. i will do exactly as anyone else would do step by step..

just to be confirm:

all i am doing is
- flashing the sprint10 image (2019-09-25-Screenly-OSE) into SD card
- loading SD card into a pi that **does not have ethernet** so that i get the wifi connect to come up after it creates its own ad-hoc network
- i will then join that wifi from a device with a web browser and configure the screenly pi from that device i am connected to
- i will then expect the pi to restart itself and no longer have an ad-hoc network and get a proper IP from the SSID/wifi network I just joined it since I know 100% the SSID/wifi network I selected has proper DHCP or IP reservation settings set up that the Pi upon joining will properly get an IP..

if everything i do works, the only thing i can tell you is to try and flash the SD card with raspbian lite found here since it was the last stretch version and buster is not compatible yet:
http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip

then, if you are able to connect to your SSID from  Raspbian on that same Pi, then there is something we need to look into with regards to exact settings on your wifi and maybe screenly is not able to properly join it, the only reason i am doing all this is because i once ran into an issue where the screenly image was not allowing me to properly join the wifi because it didnt even see my SSID, and i had to change the locale first and i forgot what else from the default sprint image in order for the wifi to be properly seen and joined.

ealmonte32, 2019-11-22:
Ok, I can confirm you are correct, upon giving the screenly.io screen a wifi to join and then pi starting up, I also see the Unable to load local IP address screen... however, pressing ctrl+alt+f1 and logging in as:
user: pi
pwd: raspberry

and checking ifconfig, it did displayed a proper IP address, so after that I just rebooted it and it worked, so I took the logs and will see what I find that would have caused the initial boot error.

ealmonte32, 2019-11-22:
@rusko124 
I can confirm (so far) that /lib/utils.py netifaces/default gateway section needs something that can give it time before trying to obtain ip address for devices on wlan, something like time.sleep(20) or something similar for this to not show up, i started with time.sleep(5) but that still was too fast before system saw error, 20 seconds was enough, maybe you can try different seconds, 10, 15,  etc.. but just enough for this to not show up.. i can write PR request..

ealmonte32, 2019-11-25:
@bnupen @the-ft 
I have a **temporary** solution for you guys until rusko fixes the code with a wait function.

if you have access to your Pi via ssh or pressing ctrl+alt+f1 on the keyboard, run this command which just includes my temporary time.sleep fix:
```
curl -s https://raw.githubusercontent.com/ealmonte32/screenly-ose/master/lib/utils.py > ./screenly/lib/utils.py
```

closed 2020-02-02 by stale[bot]