Screenly/Anthias #1241: Add ISO date format to Screenly settings

ajagelund, 2019-10-17:
#### ISO format missing

In the date format dropdown for Screenly settings the ISO standard format is missing.
ISO format: *yyyy-mm-dd*

### Side note

I have a Pull Request ready to push on my computer... :-)

### Environment

_Don't think this is environment related, however:_

* Raspberry Model: Model 3B Revision: 1.2 Ram: 1 GB Sony UK
* Screenly Version: production@6c2c2fd

vpetersson, 2019-10-22:
@rusko124 can you take a look at this?

ajagelund, 2019-10-23:
In settings.html:

`<option value="yyyy-mm-dd"
                                        {% if context.date_format == 'yyyy-mm-dd' %} selected="selected" {% endif %}>
                                    year-month-day
                                </option>
`

vpetersson, 2019-10-24:
Could you do a pull request for this?

rusko124, 2019-11-22:
Can be closed

closed 2019-11-22 by vpetersson