Screenly/Anthias #1242: Image Not Sizing to Screen

errn12, 2019-10-18:
I am a noob to Screenly.  Running on Pi 3.  Trying to find a good solution for our school's main board in the entry-way that someone besides me can manage.  Got everything up and running but the pictures are displaying very small (relatively speaking) but certainly not full screen.  Any solutions for this?  I found a post from 2013 but figured something would have had to been addressed since then.  Thanks.

vpetersson, 2019-10-19:
What's the resolution of the image? If it's less than 1920x1080, that's expected. Screenly never scales up images.

closed 2019-10-19 by vpetersson

errn12, 2019-10-19:
That is good to know.  No, the pics in question were not that big.  Will it scale down if it is larger than 1920x1080?

vpetersson, 2019-10-20:
Yes it should, but if possible, keep it to the native resolution as there
is a performance penalty if you exceed it.

errn12, 2019-10-21:
Awesome.  Thanks so much!