Screenly/Anthias #1243: ImportError: No module named celery

AlbertoLopSie, 2019-10-18:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

"ImportError: No module named celery" When trying to run the docker dev container

#### Reproduction Steps

 docker run --rm -it \
     --name=screenly-dev \
     -e 'LISTEN=0.0.0.0' \
     -p 8080:8080 \
     -v $(pwd):/home/pi/screenly \
     screenly/ose-dev-server

Results in:

Unable to find image 'screenly/ose-dev-server:latest' locally
latest: Pulling from screenly/ose-dev-server
092586df9206: Pull complete 
baa63ca98ea0: Pull complete 
69d6a83830a9: Pull complete 
e27f3cfa896f: Pull complete 
b0d89ad7af4f: Pull complete 
67ab4c308f78: Pull complete 
e4df6207e63d: Pull complete 
43cc66658679: Pull complete 
ce7094a59785: Pull complete 
54e307d7ab86: Pull complete 
Digest: sha256:7a74159b30852c05dbd08bfb087d5bdabacbed4ed7c2b55d304af48ba1e38d19
Status: Downloaded newer image for screenly/ose-dev-server:latest
**Traceback (most recent call last):
  File "server.py", line 18, in <module>
    from celery import Celery
ImportError: No module named celery**

### Environment

Running in a OSX terminal window

vpetersson, 2019-10-19:
I'm unable to reproduce this:

```
$ docker run --rm -it --name=screenly-dev -e 'LISTEN=0.0.0.0' -p 8080:8080 -v $(pwd):/home/pi/screenly screenly/ose-dev-server                                                
Unable to find image 'screenly/ose-dev-server:latest' locally
latest: Pulling from screenly/ose-dev-server
092586df9206: Pull complete
baa63ca98ea0: Pull complete
69d6a83830a9: Pull complete
e27f3cfa896f: Pull complete
b0d89ad7af4f: Pull complete
67ab4c308f78: Pull complete
e4df6207e63d: Pull complete
43cc66658679: Pull complete
ce7094a59785: Pull complete
54e307d7ab86: Pull complete
Digest: sha256:7a74159b30852c05dbd08bfb087d5bdabacbed4ed7c2b55d304af48ba1e38d19
Status: Downloaded newer image for screenly/ose-dev-server:latest
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
  RequestsDependencyWarning)
[2019-10-19 10:01:26 +0000] [6] [INFO] Starting gunicorn 19.8.1
[2019-10-19 10:01:26 +0000] [6] [INFO] Listening at: http://0.0.0.0:8080 (6)
[2019-10-19 10:01:26 +0000] [6] [INFO] Using worker: threads
[2019-10-19 10:01:26 +0000] [12] [INFO] Booting worker with pid: 12
```

Did you check out the git repository and were in that directory when launching the command? If not, that would explain the error.

closed 2019-10-19 by vpetersson

AlbertoLopSie, 2019-10-21:
> I'm unable to reproduce this:
>
> ```
> $ docker run --rm -it --name=screenly-dev -e 'LISTEN=0.0.0.0' -p 8080:8080 -v $(pwd):/home/pi/screenly [...]

Yes!  Triple checked!!

Running in the cloned repo folder with bra.This is the folder structure i have:
```
(venv) albertos-imac:screenly-ose alberto$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

(venv) albertos-imac:screenly-ose alberto$ python -V
Python 2.7

(venv) albertos-imac:screenly-ose alberto$ pip -V
pip 19.3.1 from /Users/alberto/dev/screenly-ose/venv/lib/python2.7/site-packages/pip (python 2.7)

(venv) albertos-imac:screenly-ose alberto$ ls -l
total 296
-rw-r--r--   1 alberto  staff  18228 Oct 17 11:39 LICENSE
-rw-r--r--   1 alberto  staff     97 Oct 17 11:39 Procfile
-rw-r--r--   1 alberto  staff   3584 Oct 17 11:39 README.md
drwxr-xr-x   4 alberto  staff    136 Oct 17 11:39 ansible
-rw-r--r--   1 alberto  staff   7912 Oct 17 11:39 auth.py
drwxr-xr-x   8 alberto  staff    272 Oct 21 15:52 bin
drwxr-xr-x  10 alberto  staff    340 Oct 17 11:39 docker
-rw-r--r--   1 alberto  staff   1285 Oct 17 11:39 docker-compose.dev.yml
-rw-r--r--   1 alberto  staff   1517 Oct 17 11:39 docker-compose.yml
drwxr-xr-x   8 alberto  staff    272 Oct 17 11:39 docs
drwxr-xr-x  13 alberto  staff    442 Oct 17 11:39 lib
-rw-r--r--   1 alberto  staff    641 Oct 17 11:39 package-lock.json
-rw-r--r--   1 alberto  staff    137 Oct 17 11:39 package.json
drwxr-xr-x   4 alberto  staff    136 Oct 17 11:39 requirements
-rwxr-xr-x   1 alberto  staff  58268 Oct 17 16:24 server.py
-rw-r--r--   1 alberto  staff   6802 Oct 17 11:39 settings.py
-rw-r--r--   1 alberto  staff   1798 Oct 17 11:39 start_resin_wifi.py
drwxr-xr-x  10 alberto  staff    340 Oct 17 11:39 static
drwxr-xr-x  13 alberto  staff    442 Oct 17 11:39 templates
drwxr-xr-x  13 alberto  staff    442 Oct 17 11:39 tests
drwxr-xr-x   3 alberto  staff    102 Oct 17 11:39 tools
drwxr-xr-x   5 alberto  staff    170 Oct 21 15:57 venv
-rwxr-xr-x   1 alberto  staff  18059 Oct 17 11:39 viewer.py
-rw-r--r--   1 alberto  staff   1597 Oct 17 11:39 websocket_server_layer.py
(venv) albertos-imac:screenly-ose alberto$ 
```

I realize you closed the issue but, could you give me any hint on this?

Thanks!

AlbertoLopSie, 2019-10-21:
For the record...

I was able to run the docker image only after rebuilding it with docker/Dockerfile.server.dev

For some reason the image pulled from docker hub is not the same I built from scratch.