Screenly/Anthias #1244: Rabbitmq keeps stopping and restarting

trustees629, 2019-10-27:
Just updated my Pi zero's to bd67ec7 and rabbitmq keeps stopping and restarting. I'm running on Balena. Has anyone else had this issue? 

Also, screenly-celery shows as downloaded but nothing else happens.

vpetersson, 2019-10-28:
@trustees629 Can you provide some logs for this?

trustees629, 2019-10-29:
Yes I can but I'm not exactly sure how to pull them from Balena. Can you help?

vpetersson, 2019-10-30:
@trustees629 You can get the container logs directly from their interface by selecting the right container.

trustees629, 2019-10-31:
Ok, I can get into screenly-websocket, screenly-viewer, and screenly-server. I looked for screenly_viewer.log and it's nowhere to be found. Thoughts?

ChuckIT, 2019-11-01:
Hi I am experiencing the same problem using Balena OS 2.43.0+rev1 on an RPi3 Screenly is only running an announcement video of still images less than two minutes in length. It runs OK for several days but then hits the RabbitMQ restart loop and the device goes offline. 
system log is at https://pastebin.com/9BymhpuE

vpetersson, 2019-11-04:
@rusko124 any clues here?

trustees629, 2019-11-04:
Here's a snip from my logs.

04.11.19 12:00:33 (-0500) <screenly-viewer> Showing asset important.jpg (image)
04.11.19 12:00:33 (-0500) <screenly-viewer> Sleeping for 10
04.11.19 12:00:38 (-0500) Restarting service 'rabbitmq sha256:3b74d2cb896b533ef4ce1c734c725caff5a3b2a7b2ddf4a81e12f275d1d2b5dd'
04.11.19 12:00:43 (-0500) Service exited 'rabbitmq sha256:3b74d2cb896b533ef4ce1c734c725caff5a3b2a7b2ddf4a81e12f275d1d2b5dd'
04.11.19 12:00:46 (-0500) <screenly-viewer> Generating asset-list...
04.11.19 12:00:46 (-0500) <screenly-viewer> Showing asset wifi.png (image)
04.11.19 12:00:46 (-0500) <screenly-viewer> Sleeping for 10
04.11.19 12:00:58 (-0500) <screenly-viewer> Generating asset-list...
04.11.19 12:00:58 (-0500) <screenly-viewer> Showing asset drinksbefore.png (image)

vpetersson, 2019-11-05:
@trustees629 Try upgrading to the latest master changes after https://github.com/Screenly/screenly-ose/pull/1248

trustees629, 2019-11-05:
@vpetersson Updated and no change.

05.11.19 18:06:21 (-0500) <screenly-viewer> Showing asset samadams.png (image)
05.11.19 18:06:21 (-0500) <screenly-viewer> Sleeping for 10
05.11.19 18:06:29 (-0500) Started service 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
05.11.19 18:06:36 (-0500) Service exited 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
05.11.19 18:06:40 (-0500) Killing service 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
05.11.19 18:06:41 (-0500) <screenly-viewer> Showing asset cornhole.png (image)
05.11.19 18:06:41 (-0500) <screenly-viewer> Sleeping for 10

rusko124, 2019-11-06:
It's very strange why the container doesn't log the reason.
I will try to reproduce the issue on my zero for the day. (Before that I ran on rpi3 without any errors)

ChuckIT, 2019-11-06:
Somewhat related I attempted to run the latest Master #1248 using redis from a clean git clone on my laptop ran docker-compose -f docker-compose.dev.yml up --build which succesfully built all the containers. However on launching celery could not talk to redis giving:
screenly-celery_1     | Trying again in 30.00 seconds...
screenly-celery_1     |
screenly-celery_1     | [2019-11-05 19:41:51,595: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
screenly-celery_1     | Trying again in 32.00 seconds...

On searching the internet I found a potential solution on stackoverflow :

(https://stackoverflow.com/questions/32615370/redisdockerdjango-error-111-connection-refused)

which recommended the replacement of localhost in the CELERY_BROKER_URL and CELERY_RESULT_BACKEND  to either "redis" or the ip of the container as localhost refers to the current container not the intended partner. Modifying the variables found in server.py I found that using "redis" failed as the hostname could not be resolved however using the ip of the redis container worked and the development version of Screenly seems stable so far.
Hope this helps

rusko124, 2019-11-06:
[image]

Hmmm, here's what I can suggest to start with:
**1.**
Change:
```
  redis:
    restart: always
    image: redis:alpine
    expose:
      - 6379
    logging:
      driver: none
```
to
```
  redis:
    restart: always
    image: redis:alpine
    expose:
      - 6379
```

**or 2.**
Use 
```
    ports:
      - 6379:6379
```
instead of
```
    expose:
      - 6379
```

vpetersson, 2019-11-06:
@rusko124 Why do we even expose the port? That feels like bad security practice. If the containers are on the same docker network, there is no need for port exposure.

rusko124, 2019-11-06:
@vpetersson To redis container we don't setup the docker networking(only for the Screenly services).
But you're right, we can change that.

rusko124, 2019-11-07:
@ChuckIT Could you pull up the latest changes and let us know if it's good to you?

ChuckIT, 2019-11-07:
@rusko124 Pulled changes specified in #1249 initial observations seem to indicate they solve the redis "error 111" connection issues involved with running screenly-ose in development mode on the laptop, I also deployed the changes to an RPi3 running balena-os connected to the BalenaCloud and no immediate broker restarting issues observed but will allow it to run overnight to verify. Thanks for the improvements.

trustees629, 2019-11-07:
Not sure if this is related but my Balena services shows screenly-celery as "downloaded" and never changes where the others show "running".

trustees629, 2019-11-07:
Ok, after updating Balena, the system is running but still getting the redis errors and now showing one with screenly-celery. Log snip below.

07.11.19 09:20:58 (-0500) <screenly-viewer> Showing asset vance1123.png (image)
07.11.19 09:20:58 (-0500) <screenly-viewer> Sleeping for 10
07.11.19 09:21:08 (-0500) Restarting service 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
07.11.19 09:21:08 (-0500) Service exited 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
07.11.19 09:21:13 (-0500) <screenly-viewer> Showing asset myster1129.png (image)
07.11.19 09:21:13 (-0500) <screenly-viewer> Sleeping for 10
07.11.19 09:21:31 (-0500) <screenly-viewer> Showing asset country12-1.png (image)
07.11.19 09:21:32 (-0500) <screenly-celery> [2019-11-07 14:21:32,701: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error while reading from socket: (104, 'Connection reset by peer').
07.11.19 09:21:32 (-0500) <screenly-celery> Trying again in 32.00 seconds...
07.11.19 09:21:32 (-0500) <screenly-celery> 
07.11.19 09:21:32 (-0500) <screenly-viewer> Sleeping for 10

vpetersson, 2019-11-07:
Ah! Yeah that's wrong. `localhost` should be the name of the container for this to work. Ping @rusko124.

trustees629, 2019-11-08:
The changes didn't work.

08.11.19 10:37:19 (-0500) <screenly-celery> [2019-11-08 15:37:18,990: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
08.11.19 10:37:19 (-0500) <screenly-celery> Trying again in 2.00 seconds...
08.11.19 10:37:19 (-0500) <screenly-celery> 
08.11.19 10:37:21 (-0500) <screenly-celery> [2019-11-08 15:37:21,136: ERROR/MainProcess] consumer: Cannot connect to redis://localhost:6379/0: Error 111 connecting to localhost:6379. Connection refused..
08.11.19 10:37:21 (-0500) <screenly-celery> Trying again in 4.00 seconds...
08.11.19 10:37:21 (-0500) <screenly-celery>

trustees629, 2019-11-08:
The latest update didn't work.

08.11.19 12:32:49 (-0500) <screenly-celery> [2019-11-08 17:32:49,824: ERROR/MainProcess] consumer: Cannot connect to redis://redis:6379/0: Error -5 connecting to redis:6379. No address associated with hostname..
08.11.19 12:32:49 (-0500) <screenly-celery> Trying again in 32.00 seconds...
08.11.19 12:32:49 (-0500) <screenly-celery> 
08.11.19 12:33:21 (-0500) <screenly-celery> [2019-11-08 17:33:21,908: ERROR/MainProcess] consumer: Cannot connect to redis://redis:6379/0: Error 111 connecting to redis:6379. Connection refused..
08.11.19 12:33:21 (-0500) <screenly-celery> Trying again in 32.00 seconds...
08.11.19 12:33:21 (-0500) <screenly-celery> 
08.11.19 12:33:22 (-0500) Service exited 'redis sha256:a6c9afd70ccbdb34137d9012f881adcca44d4e5217f1eccbbd70aae21e36be25'
08.11.19 12:33:54 (-0500) <screenly-celery> [2019-11-08 17:33:53,996: ERROR/MainProcess] consumer: Cannot connect to redis://redis:6379/0: Error -5 connecting to redis:6379. No address associated with hostname..

rusko124, 2019-11-08:
@trustees629 Could you provide a screenshot of your Balena's dashboard? And some more information about your device?

trustees629, 2019-11-08:
[image]

It's a Pi Zero.

vpetersson, 2019-11-08:
Perhaps there are no builds for the ARM builds for this particular CPU for the Pi Zero of Redis. It is after all using an older CPU than the modern Model B line.

trustees629, 2019-11-08:
UGH.... does that mean I will have to upgrade? I'm going to switch to the production version so I can at least get back up and running.

vpetersson, 2019-11-08:
I'm not saying it is the case, just wanted to flag it as a possibility.

ChuckIT, 2019-11-08:
Checking out the most recent version of the master #1252 in a couple of ways
1) The development version on my laptop 
2) The production version on an RPi3 connected to balenaCloud 

Observations

- Currently no connection issues are showing up in the logs and I am seeing Celery and Redis activity in the logs.

- Have noticed the RPi3 go offline for no discernible reason after approx 15 hours of operation. Log shows only common screenly-ose_viewer messages.

- When I reboot or restart the RPi3 I notice that the opening splash screen displays the screenly-ose_server internal docker IP and port instead of ip of the RPi3 on the local network reported as a separate issue #1253.

- Same goes for the api/docs screen.

Question:
Why is redis specified as image: redis:alpine in the development docker-compose.dev.yml and image: arm32v7/redis in production docker-compose.yml?

Thanks

vpetersson, 2019-11-09:
> Why is redis specified as image: redis:alpine in the development docker-compose.dev.yml and 
image: arm32v7/redis in production docker-compose.yml?

This is likely because there is no ARMv6/v7 version of the [official Redis image](https://hub.docker.com/_/redis). 

If we're hard coding `arm32v7`, that would indeed explain @trustees629, as the Pi Zero is ARMv6. 

@rusko124 can you look at this? Seems like it is a fairly widespread issue then.

trustees629, 2019-11-23:
I updated to the latest version and Redis seemed to be running fine. However, screenly-server wouldn't stay running and I had to revert again. I'm sorry I don't have logs but I can tell you that the error I was getting was the same as the Redis error before... restarting service sha256.... service exiting sha256...

vpetersson, 2019-11-28:
Closing this out in favor of #1261 now that we've moved to Redis.

closed 2019-11-28 by vpetersson