Screenly/Anthias #1247: Dockerized Development Environment Fails to Start

ChuckIT, 2019-11-03:
#### Overview of the Issue
Issuing the docker command detailed in the current README.md to bring up the development environment on a laptop running UbuntuStudio Xenial the following error is generated:

Traceback (most recent call last):
  File "server.py", line 18, in <module>
    from celery import Celery
ImportError: No module named celery

As a result the development version of docker is not started.
I would like this to start in order to test out some programing involving the Screenly API

#### Reproduction Steps
Remove any old Screenly source from the system using: 
    rm -r screenly-ose
Execute the command:  
    git clone https://github.com/Screenly/screenly-ose.git
Attempt to start the docker container:
    cd screenly-ose
    docker run --rm -it --name=screenly-dev \
        -e LISTEN=0.0.0.0 -p 8080:8080 \
        -v $(pwd):/home/pi/screenly screenly/ose-dev-server

### Environment

* Laptop HP Pavilion dv6

* OS UbuntuStudio 
lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

* Screenly OSE Version: Current Git Repository

* Celery version is expressed in requirements/
requirements.txt:celery==4.3.0

rusko124, 2019-11-04:
@vpetersson confirmed, I have the same error

Hey @ChuckIT!
You also could work with docker-compose.
Use the next command `docker-compose -f docker-compose.dev.yml up --build`

ChuckIT, 2019-11-06:
@rusko124 docker-compose -f docker-compose.dev.yml up --build worked thanks

closed 2019-11-06 by ChuckIT