Screenly/Anthias #1251: Create docker-compose file for demo site

vpetersson, 2019-11-07:
Since we moved to multi-container setup, the demo site broke. We need to port the configuration to the demo site by generating a new docker-compose file tailored to the demo setup.

Here is how it ran before:

```
$ cat /etc/systemd/system/ose.service
[Unit]
Description=Screenly OSE
After=docker.service
Requires=docker.service

[Service]
TimeoutStartSec=0
Restart=always
ExecStartPre=-/usr/bin/docker system prune -f
ExecStartPre=-/usr/bin/docker kill %n
ExecStartPre=-/usr/bin/docker rm %n
ExecStartPre=/usr/bin/docker pull screenly/ose-dev-server:latest
ExecStart=/usr/bin/docker run --name %n --rm -p 80:8080 -e SWAGGER_HOST=ose.demo.screenlyapp.com -e LISTEN=0.0.0.0 --memory=300m screenly/ose-dev-server:latest

[Install]
WantedBy=multi-user.target
```

Two things to keep in mind:

* It needs to have swagger working for the host specified
* Google Analytics need to work 
* It needs to listen on port 80

vpetersson, 2020-01-13:
@rusko124 This one isn't complete. [docker-compose.yml](https://github.com/Screenly/screenly-ose/blob/master/docker-compose.yml) has a number of services, whereas [docker-compose.demo.yaml](https://github.com/Screenly/screenly-ose/blob/master/docker-compose.demo.yml) only has one.

rusko124, 2020-02-05:
@vpetersson got it.
So then we need to update https://hub.docker.com/r/screenly/screenly-ose-websocket to latest
and also need to create `screenly-celery` image

PR #1279 (2019-12-17): New: docker-compose.demo.yml
Closes #1251
closed 2020-04-12 by stale[bot]