Screenly/Anthias #1255: static ip

gady87, 2019-11-13:
#### i been trying to place an static ip on the screenly but i have not been able to. i have follow most of the steps i found with no result. i cant access the raspberry by ssh its says conection refuse

vpetersson, 2019-11-14:
This should do the trick:

 * Follow [these](https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-give-your-raspberry-pi-a-static-ip-address-update) instructions for setting a static IP
 * Use `rpi-config` to enable SSH (it's disabled by default)

closed 2019-11-14 by vpetersson

gady87, 2019-11-14:
Hi so i need to install raspberrian first and the pull screenly?  Or how i can access this config with the screenly image

vpetersson, 2019-11-14:
No, you don't need to install Rasbian first. You just need to connect a keyboard (and make sure you don't have any videos enabled). With a keyboard inserted, you should be able to get to a TTY (ctlr+f2) and login.

gady87, 2019-11-14:
That is something that I didn't know thank you so much