Screenly/Anthias #1256: No module named dbus

dingetje, 2019-11-14:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

After running upgrade script on rpi-0 screenly stopped working

### Environment

* Raspberry Pi Hardware Version: raspberry Pi Zero with 512Mb RAM
* Screenly OSE Version: don't know where to find version, it's certainly not in the log file!

vpetersson, 2019-11-14:
@dingetje We need more information. What image did you use, or did you install it from scratch. If you installed from scratch, what branch etc.

dingetje, 2019-11-14:
`Nov 14 17:32:00 screenly systemd[1]: Starting Screenly Viewer...`
`Nov 14 17:32:02 screenly systemd[1]: Started Screenly Viewer.`
`Nov 14 17:32:03 screenly python[1361]: Traceback (most recent call last):`
`Nov 14 17:32:03 screenly python[1361]:   File "/home/pi/screenly/viewer.py", line 10, in <module>`
`Nov 14 17:32:03 screenly python[1361]:     from dbus import SessionBus`
`Nov 14 17:32:03 screenly python[1361]: ImportError: No module named dbus`
`Nov 14 17:32:04 screenly systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE`
`Nov 14 17:32:04 screenly systemd[1]: screenly-viewer.service: Unit entered failed state.`
`Nov 14 17:32:04 screenly systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.`
`Nov 14 17:32:04 screenly systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.`
`Nov 14 17:32:04 screenly systemd[1]: Stopped Screenly Viewer.`

vpetersson, 2019-11-14:
That looks like a corrupt installation.

dingetje, 2019-11-14:
I'm running the install script now. See if that resolves thing. Probably end-user error ;-)

dingetje, 2019-11-14:
Would it be too much to ask to print the Screenly version in the log file?

vpetersson, 2019-11-14:
We use rolling releases, so the release is really the git hash.

dingetje, 2019-11-14:
Where can I find the git hash?
All I know is that I answered 'yes' to use experimental branch

vpetersson, 2019-11-14:
```
$ cd ~/screenly-ose
$ git rev-parse --short HEAD
```

dingetje, 2019-11-14:
```
pi@screenly:~/screenly $ git rev-parse --short HEAD
1215534
pi@screenly:~/screenly $ cd ..
pi@screenly:~ $ ls
logs.txt  screenly  screenly_assets  version.md
pi@screenly:~ $ cat version.md
Screenly version: experimental@1215534
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.11 (stretch)
Release:        9.11
Codename:       stretch
```

dingetje, 2019-11-14:
I ended up renaming the old screenly folder and running the installation script as advertised:

[screenly-install.txt](https://github.com/Screenly/screenly-ose/files/3848622/screenly-install.txt)

After the reboot /var/log/syslog shows this:
```
Nov 14 22:08:12 screenly systemd[1]: Started Screenly celery worker.
Nov 14 22:08:34 screenly celery[4446]: /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
Nov 14 22:08:34 screenly celery[4446]:   RequestsDependencyWarning)
Nov 14 22:08:37 screenly celery[4446]: Traceback (most recent call last):
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/bin/celery", line 10, in <module>
Nov 14 22:08:37 screenly celery[4446]:     sys.exit(main())
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/__main__.py", line 16, in main
Nov 14 22:08:37 screenly celery[4446]:     _main()
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/bin/celery.py", line 322, in main
Nov 14 22:08:37 screenly celery[4446]:     cmd.execute_from_commandline(argv)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/bin/celery.py", line 496, in execute_from_commandline
Nov 14 22:08:37 screenly celery[4446]:     super(CeleryCommand, self).execute_from_commandline(argv)))
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/bin/base.py", line 273, in execute_from_commandline
Nov 14 22:08:37 screenly celery[4446]:     argv = self.setup_app_from_commandline(argv)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/bin/base.py", line 479, in setup_app_from_commandline
Nov 14 22:08:37 screenly celery[4446]:     self.app = self.find_app(app)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/bin/base.py", line 501, in find_app
Nov 14 22:08:37 screenly celery[4446]:     return find_app(app, symbol_by_name=self.symbol_by_name)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/app/utils.py", line 362, in find_app
Nov 14 22:08:37 screenly celery[4446]:     sym = imp(app)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/local/lib/python2.7/dist-packages/celery/utils/imports.py", line 104, in import_from_cwd
Nov 14 22:08:37 screenly celery[4446]:     return imp(module, package=package)
Nov 14 22:08:37 screenly celery[4446]:   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
Nov 14 22:08:37 screenly celery[4446]:     __import__(name)
Nov 14 22:08:37 screenly celery[4446]: ImportError: No module named celery
Nov 14 22:08:37 screenly systemd[1]: screenly-celery.service: Main process exited, code=exited, status=1/FAILURE
Nov 14 22:08:38 screenly systemd[1]: screenly-celery.service: Unit entered failed state.
Nov 14 22:08:38 screenly systemd[1]: screenly-celery.service: Failed with result 'exit-code'.
Nov 14 22:08:43 screenly systemd[1]: screenly-celery.service: Service hold-off time over, scheduling restart.
Nov 14 22:08:43 screenly systemd[1]: Stopped Screenly celery worker.
```

dingetje, 2019-11-14:
I think I must count my losses and start with a fresh SD card...

dingetje, 2019-11-14:
Running the upgrade script
```
 ./screenly/bin/run_upgrade.sh
```
and reverting to the stable release, resulting in a working screenly.
Looks like the 'experimental' version is not quite there yet?

vpetersson, 2019-11-15:
> Looks like the 'experimental' version is not quite there yet?

Yeah, it's experimental :). 

We've spent a lot of effort over the last year or so on this version to replace the browser. It's getting there, but it's not ready for prime time just yet it seems.

dingetje, 2019-11-15:
issue closed.

closed 2019-11-15 by dingetje