Screenly/Anthias #1258: Ask/suggest/recommend to users that they should change the default raspbian password for user pi

ealmonte32, 2019-11-15:
As discussed here: https://github.com/Screenly/screenly-ose/commit/01629a7ddd00e786487748e8c597edbe36b351ab#diff-2066c4d7515c02785d28a6ab3a57ace1

Instead of forcing the change of password for pi, ask/suggest/recommend instead that they should change it if they haven't already done so, because of it being an obvious security issue.

Kipjr, 2019-11-16:
Or the option to change the password or disable SSH

ealmonte32, 2019-11-16:
> Or the option to change the password or disable SSH

Ehhh...? recommending to change it is exactly that option, if user says yes to changing then they are prompted for new password.... 
..and disabling SSH is one of the most crippling decisions you can make with a raspberry pi that you need to remotely control, sorry but that is not a good suggestion at all, unless you're going to be connected directly with a keyboard/mouse to the pi all the time..

vpetersson, 2019-11-18:
@ealmonte32 I'm not sure I fully agree. In a perfect world, we would not need SSH in Screenly OSE. Everything should be configurable using the WebUI. In the future, I'd love to have a super slimmed down image, much like OpenELEC.

ealmonte32, 2019-11-18:
@vpetersson Gotcha, correct, just like LibreELEC, but remember you are only thinking along the lines of Screenly-OSE on this instance, I was thinking more along the lines of Raspbian+Pi+Screenly, meaning, there are many times where if screenly freezes or hangs, SSH'ing into the Pi is the only way to not have to go and physically reset the Pi to get it back up and running.. 

Perfect example happened just now while responding, black screen issue on master branch (posting on https://github.com/Screenly/screenly-ose/issues/1222 ), troubleshooting now, and Screenly-OSE WebUI system control allows Reboot or Shutdown, but pressing Reboot did not resolve the issue, it did not reboot the service or the Pi, and pressing shutdown will force a physical intervention of the device... but, since I could SSH into the Pi, I was able to easily run `systemctl restart screenly-viewer.service` and it came back up and running, memory was reset and it will now run until issue happens again, but this shows SSH is indeed very much needed for now.

ealmonte32, 2019-11-18:
@vpetersson What do you think of having the option of making Screenly only accept local subnet for SSH connections via `/etc/ssh/sshd_config` so that users not behind a real firewall would not have to worry that their Screenly Pi is open to any external connection? I'd rather do that then completely turn off SSH.

vpetersson, 2019-11-19:
>  there are many times where if screenly freezes or hangs, SSH'ing into the Pi is the only way to not have to go and physically reset the Pi to get it back up and running..

Sure, but we need to weed out those issues. 

> hat do you think of having the option of making Screenly only accept local subnet for SSH connections via /etc/ssh/sshd_config so that users not behind a real firewall would not have to worry that their Screenly Pi is open to any external connection? I'd rather do that then completely turn off SSH.

I don't think that's a good solution given how common it is that people just port forward from their router. In that situation. Such fix would just work if you have a public IP, which I'd argue is rare.

ealmonte32, 2019-11-20:
Hmm.. as always correct me if I'm wrong, but as far as I know port forwarding will always show up as incoming IP address of the gateway/router, thus when we deny this IP address in the sshd_config file (we get gateway address from current interface network info), we are eliminating the port forwarded connection... also, if someone says Yes to Screenly asking if such person wants to make screenly more secure by disabling SSH from external connections, and the person goes and still opens port forwarding, then what was the point of such person saying yes to disabling screenly pi from receiving such SSH connection, meaning, why say yes to trying to secure it but then go and manually enable port forward, doesnt make sense to me, the user would be literally conflicting his own decision..
I just think screenly offering these little tweaks to help certain users who are not network savvy or tech savvy secure their Pi running screenly does more good than bad, get me?

vpetersson, 2019-11-20:
> s always correct me if I'm wrong, but as far as I know port forwarding will always show up as incoming IP address of the gateway/router, thus when we deny this IP address in the sshd_config file (we [...]

The incoming connection will come from the [private CIDR](https://en.wikipedia.org/wiki/Private_network) (because of NAT'ing). Sure, we could block access from the router's IP but now we're starting to add a lot of logic that we need to manually build out into `sshd_config`.

IMHO it's better to start with a locked down state and then enable things if needed rather than the other way around. Perhaps we need the ability to toggle OpenSSH from the web interface in the future, but that would first require that we harden the web interface.

ealmonte32, 2019-11-20:
> The incoming connection will come from the private CIDR (because of NAT'ing). Sure, we could block access from the router's IP but now we're starting to add a lot of logic that we need to manually build out into sshd_config.

Actually, I kept reading and researching and found a super simple no-brainer way to do this, I am actually using it on mine now..
we do this if the user selects "yes" to question about locking down SSH to only accept LAN connections:
```
sudo sh -c "echo sshd: ALL EXCEPT 192.168.,172.16.,10. >> /etc/hosts.deny"
```
And, say the user wants to toggle it on and off from the WebUI, you simply have a button that either writes the command above for on, or clears out the /etc/hosts.deny file with 
```
sudo sh -c "> /etc/hosts.deny"
```
thoughts?

vpetersson, 2019-11-21:
No, this wouldn't work, and here's why.

Yes, it will block out access from non private CIDRs, but in what environment would that help? The only one I can think of is if the device is publicly exposed and has a public IP. Moreover, if IPv6 were to be enabled, that is still an attack vector.

Let me explain a bit more why this wouldn't work for a regular home environment (with IPv4). Assume we have three devices on the network:

Device | Public IP | Private IP | Comment
-- | -- | --| --
Router | 8.6.7.6 | 192.168.1.1 | Port 22 is forwarded to 192.168.1.5:22
Pi | N/A| 192.168.1.5 | 
Workstation | N/A | 192.168.1.6

If I were to add the rule `sshd: ALL EXCEPT 192.168.,172.16.,10.` to `/etc/hosts.deny`, I would be able to access the pi from the workstation and the router.  Moreover, since an incoming connection to 8.6.7.6:22 (the router's public IP) is NAT'd to 192.168.1.5, the pi would see this connection coming from 192.168.1.1, and thus allow the connection.

Does that make sense?

ealmonte32, 2019-11-21:
I like the challenge 😃 
[image]

Ok so i'll start with this:
>Yes, it will block out access from non private CIDRs, but in what environment would that help? The only one I can think of is if the device is publicly exposed and has a public IP

I thought the whole point of this is that someone said we should disable SSH, I disagreed due to the fact that it would cripple the Raspberry Pi, then I understood that you agreed with disabling SSH because you saw this from a Screenly point of view where the device would be protected from external connections, but now you say that blocking external IP would only help if device is publicly exposed, well if it is not publicly exposed then why disable SSH to begin with?
___

>Moreover, if IPv6 were to be enabled, that is still an attack vector.

Actually, as far as I know (from reading manual) not if we tell sshd to only listen on IPv4 by doing this:

```
example on how to get the current active IP of device:
pi@screenlypi:~ $ SRCIP=`ip route | grep via | awk '{print $7}'`
pi@screenlypi:~ $ echo $SRCIP
192.168.96.16

since the above proves to us that it works in getting the current active IP
we use the following to insert it into the sshd_config file:
sed -i "s/#ListenAddress 0.0.0.0/ListenAddress $SRCIP/g" /etc/ssh/sshd_config

this allows sshd to only listen on current IPv4 address of host..
if host has multiple active connections, then you would need to decide
whether you want it listening on wlan* or eth* as shown below with a Pi that has both:

pi@screenlypi:~ $ ip route | grep via | grep eth | awk '{print $7}'
192.168.1.17
pi@screenlypi:~ $ ip route | grep via | grep wlan | awk '{print $7}'
192.168.96.16
```
___

Now, about the port forwarding situation:

> If I were to add the rule sshd: ALL EXCEPT 192.168.,172.16.,10. to /etc/hosts.deny, I would be able to access the pi from the workstation and the router. Moreover, since an incoming connection to [...]

Actually, we could use the same method as above to deny the gateway from being a source that can connect to SSH, tested and verified:

```
the following IP would be the gateway/router IP which is what the Pi would see if NAT'd:
pi@screenlypi:~ $ ROUTE=`ip route | grep via | awk '{print $3}'`
pi@screenlypi:~ $ echo $ROUTE
192.168.99.254

Thus, if we now tell /etc/hosts.allow the following:
sshd: 192.168.,172.16.,10. EXCEPT 192.168.99.254

and tell /etc/hosts.deny the following:
sshd: ALL

then we are able to allow all local connections and block the port forwarded connections.

Confirmation: I was not able to connect through port forward, and logs showed
Nov 21 12:10:48 screenlypi sshd[25046]: pam_unix(sshd:session): session closed for user pi
Nov 21 12:10:51 screenlypi sshd[25108]: refused connect from 192.168.99.254 (192.168.99.254)
```

😃

vpetersson, 2019-11-21:
> Actually, we could use the same method as above to deny the gateway from being a source that can connect to SSH

Yes, in theory. But this assumes a) there is only one way into the network and b) we need to assume that both the default gateway of the Pi is that default gateway. Needless to say, this adds a lot of complexity for little gain.

Honestly, I don't think it's worth the effort. I rather leave SSH disabled, and let customers who really know what they are doing enable it. Hence I agree with the Pi Foundation's decision to disable it SSH by default. 

If we were to have SSH enabled by default, we would have to do one or more of the following:

* Force a password change for the user 'pi'
* Disable password logins in SSH

In case of the latter, we would then need a mechanism to populate the SSH key from the dashboard, which in turn introduces more risk on the web UI as it is now a proper attack vector into the device.

ealmonte32, 2019-11-21:
All good, I just liked the challenge of allowing SSH while at the same time trying to protect it from external connections even if we can't bullet-proof the design.
BTW, AFAIK, the `ip route` command should always return the default gateway of the Pi no matter how the Pi is connected.

Ok, so about the original purpose of this ticket:
- to ask the user if they already changed the default password on the Pi (when using the install script method), and if user says no, then it is suggested they do and passwd is issued, otherwise, it isn't
- if the user is only controlling screenly via WebUI, and we assume they are not tech savvy, would they be able to change it from web? would you need to use something like this https://xtermjs.org ?

I'm trying to figure out how to do a one liner with xargs and passwd to see if it can be changed that way..

vpetersson, 2019-11-28:
Right, so let's limit the scope of this ticket to just check if the default password has been changed. If it has, we don't show the dialogue.

In the future, embedding something like a terminal inside the dashboard would be great, but we need to step up our security first.

closed 2020-02-03 by stale[bot]