Screenly/Anthias #1259: Errors and warning while installing on RPi4 4GB

Kipjr, 2019-11-16:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

I receive messages when installing using the installation script:
 $ bash <(curl -sL https://www.screenly.io/install-ose.sh)
Will this / can this break my Screenly-OSE environment? Is there an end-of-life given to this version, especially regarding the EOL of Python2

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Install NOOBS
* Install Raspbian Lite
* Execute  $ bash <(curl -sL https://www.screenly.io/install-ose.sh)

### Environment

* Raspberry Pi Hardware Version: 4B 4GB
* Screenly OSE Version: Experimental (2019-11-16)

`
```python
root@raspberrypi:/home/pi# bash <(curl -sL https://www.screenly.io/install-ose.sh)
       _____                           __         ____  _____ ______
      / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
      \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
     ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
    /____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                     /____/
Screenly OSE requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Would you like to use the experimental branch? It contains the last major changes, such as the new browser and migrating to Docker (y/N)

Would you like to install the WoTT agent to help you manage security of your Raspberry Pi? (y/N)

Do you want Screenly to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/screenly-ose.git
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.2 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [259 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.3 MB in 7s (1,916 kB/s)
Reading package lists... Done
+ sudo apt-get purge -y python-setuptools python-pip python-pyasn1
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'python-pyasn1' is not installed, so not removed
Package 'python-setuptools' is not installed, so not removed
Package 'python-pip' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
+ sudo apt-get install -y python-dev git-core libffi-dev libssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'git' instead of 'git-core'
The following additional packages will be installed:
  git-man libcurl3-gnutls liberror-perl libexpat1-dev libpython-dev
  libpython2-dev libpython2.7 libpython2.7-dev python2-dev python2.7-dev
[... 332 lines cut ...]
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -exec rm -r '{}' ';'
+ cd /home/pi/screenly
+ git rev-parse HEAD
+ sudo chown -R pi:pi /home/pi
+ '[' experimental = master ']'
+ '[' experimental = production ']'
+ sudo tee /etc/sudoers.d/010_pi-nopasswd
+ echo 'pi ALL=(ALL) NOPASSWD: ALL'
+ sudo chmod 0440 /etc/sudoers.d/010_pi-nopasswd
+ '[' experimental = master ']'
+ '[' experimental = production ']'
++ git rev-parse --abbrev-ref HEAD
++ git rev-parse --short HEAD
++ lsb_release -a
No LSB modules are available.
+ echo -e 'Screenly version: experimental@1215534\nDistributor ID:      Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster'
+ '[' false = false ']'
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)

`

mccoy88f, 2019-11-19:
Any news? Same problem to me!

rusko124, 2019-11-19:
Hey guys!
I'm going to work on a solution tomorrow.

rusko124, 2019-11-21:
Try the next steps:
1. Install https://www.raspberrypi.org/downloads/raspbian/ Raspbian Buster Lite
2. Run `$ bash <(curl -sL https://www.screenly.io/install-ose.sh)` (Screenly merged some changes)

My final logs of installation:
```
PLAY RECAP **************************************************************************************************************************************************************************************************
localhost                  : ok=87   changed=62   unreachable=0    failed=0    skipped=11   rescued=0    ignored=0   

+ sudo apt-get autoclean
Reading package lists... Done
Building dependency tree       
Reading state information... Done
+ sudo apt-get clean
+ sudo find /usr/share/doc -depth -type f '!' -name copyright -delete
+ sudo find /usr/share/doc -empty -delete
+ sudo rm -rf /usr/share/man /usr/share/groff /usr/share/info /usr/share/lintian /usr/share/linda /var/cache/man
+ sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -exec rm -r '{}' ';'
+ cd /home/pi/screenly
+ git rev-parse HEAD
+ sudo chown -R pi:pi /home/pi
+ '[' master = master ']'
+ sudo rm -f /etc/sudoers.d/010_pi-nopasswd
+ '[' master = master ']'
+ '[' false = false ']'
+ set +e
+ passwd
Changing password for pi.
Current password: 
passwd: Authentication token manipulation error
passwd: password unchanged
+ set -e
++ git rev-parse --abbrev-ref HEAD
++ git rev-parse --short HEAD
++ lsb_release -a
No LSB modules are available.
+ echo -e 'Screenly version: master@a1a764c\nDistributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster'
+ '[' false = false ']'
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)
```

Qonstrukt, 2019-11-21:
I just tried installing on a RPi 4 with 4GB of RAM using the experimental version on a fresh installation of Raspbian Buster, but the screen stays black after reboot.

rusko124, 2019-11-25:
The experimental branch is not supported RPi 4.
Also, the points of this sprint:
- Support Rpi 4 and Raspbian Buster
- Make the experimental browser as default and abandon the experimental branch.

budbach, 2019-12-23:
Any update on this?

For now I'm reading:

1. Install https://www.raspberrypi.org/downloads/raspbian/ Raspbian Buster Lite
2. Run `bash <(curl -sL https://www.screenly.io/install-ose.sh)`
3. **Do not** select "experimental"

Any screenly ose *should* work on the pi4.  Do I need to twiddle with `Disabling the vc4-fkms-v3d overlay` still?

vpetersson, 2019-12-29:
@rusko124 could you comment on this?

rusko124, 2019-12-30:
Hey @budbach !
>Do I need to twiddle with Disabling the vc4-fkms-v3d overlay still?

If you will use the development branch then you don't need

lajtmaN, 2020-02-04:
When will the experimental browser support the RPi 4? 
There are multiple pages I want to show on my dashboards which is not supported by the old browser.

vpetersson, 2020-02-10:
I think some more work is required for that @lajtmaN.

closed 2020-04-18 by stale[bot]