Screenly/Anthias #1261: Black Screen on Pi Zero W

sawdustlegend, 2019-11-18:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

Black Screen on Pi Zero W after 2 to 4 hours

A paragraph or two about the issue you're experiencing.

I get a  black screen after 2 to 4 hours of run time on a Pi Zero. Looks like it loses the network. Can not log in over SSH. With a keyboard, terminal shows no IP address. Reboot device, and it works for another 2-4 hours

Troubleshooting steps

Screenly OSE Version - 0.18.2 --> Mounted 2019-09-25-Screenly-OSE.zip image

- Does not happen on a 3B+. Rock solid after a few days. (This is what I tested Screenly on before buying the Zeros to see if it would work for me) Pull SD card from 3B+, put it in Pi Zero, reenter Wifi credentials, runs perfect for 2 to 4 hours then black screen. 
- I have this issue with 3 different Zeros
- Tried 3 different SD cards - Latest Screenly OSE image
- Tried various power sources. 
- Tried different Wifi Networks. Thought it may be an issue with enterprise network at school, but does the exact same thing on home network. 

In reading about similar problems I ran 

$ journalctl -u screenly\* > ~/logs.txt

I attached the results. 

Thank you for any help.
Steve 
[logs2.txt](https://github.com/Screenly/screenly-ose/files/3856503/logs2.txt)

sawdustlegend, 2019-11-22:
Update - I was working with 3 Zero Ws. They all were having a similar issue. I lent one out to another school and that teacher is telling me that it is working flawlessly for him. He is still using the image I loaded. The only difference between my setups and his is the asset link. We are both running published Google slides links, but obviously different content. I am going to try and run his asset link as see if that is the issue. 

Thank you,
Steve

rusko124, 2019-11-22:
Have you tried upgrading to the latest development version?

sawdustlegend, 2019-11-22:
No, but I can. 
 I believe this is the command for the normal image
$ bash <(curl -sL https://www.screenly.io/install-ose.sh)

What is the command for the latest development version?

Also changing the asset link to another very simple google slides presentation did not help. I am getting wlan0 is disconnected.

Thank you for you help, 
Steve

ealmonte32, 2019-11-22:
@sawdustlegend 
the different versions are selected from the install menu, the command you noted is just how to start the installation process from the command line..
when you get the options, select no for experimental and yes for development one..
let us know how it goes.

sawdustlegend, 2019-11-25:
All done. I let it update over night last night.

Today I had moderate success. I started screenly up around 10am. I left everything default, except for turning off hacker news asset. Pi Zero ran until 5pm without a hiccup. At 5 pm I added my google slide presentation back in. When I checked back on it at 8:15pm it was black screened again. 

Tonight I am going to let the weather and clock assets run and see if it is still running in the morning. If it is, I will try a different, very basic, google slides presentation and see what happens. 

Thank you for the help, 
Steve

AntonyJayM, 2019-11-26:
Hi Guys,

When we upload a h.264 video to Screenly OSE we get a solid black screen. Other assets will play after however the video will not play. 

Not sure this issue relates to the above but it's been merged here, anyone got any thoughts?

vpetersson, 2019-11-26:
@AntonyJayM Please help us test out the development release. Since you're running on a Pi3, the issue should have been resolved for you.

AntonyJayM, 2019-11-26:
> @AntonyJayM Please help us test out the development release. Since you're running on a Pi3, the issue should have been resolved for you.

Where do I find the development release?

vpetersson, 2019-11-26:
Just follow the [installation instructions](https://github.com/Screenly/screenly-ose#installing-on-raspbian) and answer yes to if you want to use the developer release.

AntonyJayM, 2019-11-27:
@vpetersson Update: Installing the development version has not fixed the issue. Any other ideas?

rusko124, 2019-11-27:
@AntonyJayM 
Could you try to open video with omxplayer on Raspberry and say what happened please?

AntonyJayM, 2019-11-27:
Update: Omxplayer did not play the video. Upon further inspection Adobe Media Encoder was set to export H264 at video codec level = 4.2 not 4.1. After making this change the video now plays. 

Didn't realise the specific requirements other than H264. Thank you for your help on this!

vpetersson, 2019-11-27:
Nice! Yeah, `omxplayer` (or rather the GPU) is very picky on the codec.

sawdustlegend, 2019-11-27:
Update Time - The time to black screen seems related to length of Google Slides presentation. 

Zero with default assets (Time and weather) will run for +2 days
Zero with Time, Weather, uploaded pictures and videos will run for +2 days
Zero with Time, Weather, uploaded pictures/videos + Short (4 slide) presentation ~12 to 24 hours
Zero with Time, Weather, uploaded pictures/videos + Long (30 slide) presentation ~2 hours tops

With my test 3B+ it was able to run for 5 days with a 30 slide presentation before black screen. What is the longest anyone has has Screenly running for before a reboot? 

Thank you again, 
Steve

sawdustlegend, 2019-12-02:
After much more testing, it definitely seems to be related to the length of slide show. (Google Slides --> Publish to Web) I can turn the slides in to a video, upload it, and they will play for 48 hours without issue. 

My solution right now is to reboot the devices every 2 hours with crontab. 

Thank you
Steve

ealmonte32, 2019-12-02:
@sawdustlegend 
just for my testing purposes, you are using a screenly image on a pi zero w and using development version or production?

ealmonte32, 2019-12-03:
@sawdustlegend 
i've had the production version running with 20+ slides and video and default assets for 5+ hours without black screen..

[image]

i've just updated celery to 4.3 by removing it `sudo pip uninstall celery` and then `sudo pip install celery` and installed zram with double the normal swap memory by changing default 1024 to 2048..
give zram a try [https://github.com/novaspirit/rpi_zram](url)
and when you are about to reboot just make this small adjustment in the actual file:

[image]

using nano or your preferred editor, change the mem multiple to 2048, yours should be default at 1024..
file is `/usr/bin/zram.sh`
lets see if you get the same results that i do..

vpetersson, 2019-12-03:
@ealmonte32 Good findings. So it looks like we're consuming too much RAM. Adding swap/zram is going to kill the SD card in the long-run. Perhaps that's an acceptable price to pay for the Pi Zero, given its limited resources however.

I'm curious to see if that resolves the issue for others too. If so, we know it's a memory issue.

(Pro-tip, to upgrade a pip package, just do `pip install -U [package]`)

sawdustlegend, 2019-12-03:
I will give it a shot this weekend and see what happens. 

Thank you for the help, 
Steve

ealmonte32, 2019-12-04:
I am testing with lowering the swappiness because we would def be killing the SD card..

sawdustlegend, 2019-12-15:
Sorry about the delay. I will try and play with it this week and report back. 

Thank you,
Steve

ealmonte32, 2019-12-19:
> Sorry about the delay. I will try and play with it this week and report back.
> 
> Thank you,
> Steve

@sawdustlegend 
to be honest, that will work for you temporarily, but it's best you hold off until this memory situation is resolved in some better workaround which doesnt involve zram since we are trying to solve it with the raspbian image being default because not everyone can go around installing zram and configuring it you know..
so i think once viktor or rusko answer about the idea of killing the uzbl browser after every loop iteration as mentioned here https://github.com/Screenly/screenly-ose/issues/1222#issuecomment-567649422 , you should hold off, again, unless you want to temporarily use zram which should help get you more time out of the viewer before screen goes black..

closed 2020-02-24 by stale[bot]

ealmonte32, 2020-02-24:
Note:
This PR (https://github.com/Screenly/screenly-ose/pull/1304) should solve this.
Will do more testing tonight just to confirm.

ealmonte32, 2020-07-14:
@sawdustlegend 
This solution has been the easiest and best one I have come up with to resolve this:
https://forums.screenly.io/t/display-often-hangs/123/49

I am creating a PR that will implement this and set automatically on the system based on the total memory of the device.