Screenly/Anthias #1264: Can't install screenly anymore (pip OpenSSL Problem)

didiatworkz, 2019-11-20:
#### Overview of the Issue

Ansible Error:
```python
TASK [screenly : Install pip dependencies] **************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "/usr/local/bin/pip2 freeze", "msg": "\n:stderr: Traceback (most recent call last):\n  File \"/usr/local/bin/pip2\", line 8, in <module>\n    sys.exit(main())\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/main.py\", line 45, in main\n    command = create_command(cmd_name, isolated=(\"--isolated\" in cmd_args))\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/commands/__init__.py\", line 96, in create_command\n    module = importlib.import_module(module_path)\n  File \"/usr/lib/python2.7/importlib/__init__.py\", line 37, in import_module\n    __import__(name)\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/commands/freeze.py\", line 12, in <module>\n    from pip._internal.operations.freeze import freeze\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/operations/freeze.py\", line 17, in <module>\n    from pip._internal.req.constructors import (\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/req/__init__.py\", line 11, in <module>\n    from .req_file import parse_requirements\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/req/req_file.py\", line 20, in <module>\n    from pip._internal.download import get_file_content\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_internal/download.py\", line 14, in <module>\n    from pip._vendor import requests\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/__init__.py\", line 97, in <module>\n    from pip._vendor.urllib3.contrib import pyopenssl\n  File \"/usr/local/lib/python2.7/dist-packages/pip/_vendor/urllib3/contrib/pyopenssl.py\", line 46, in <module>\n    import OpenSSL.SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/__init__.py\", line 8, in <module>\n    from OpenSSL import crypto, SSL\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/crypto.py\", line 15, in <module>\n    from OpenSSL._util import (\n  File \"/usr/local/lib/python2.7/dist-packages/OpenSSL/_util.py\", line 152, in <module>\n    with ffi.from_buffer(b\"\"):\nTypeError: from_buffer() cannot return the address of the raw string within a str or unicode object\n"}
```

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Install Screenly

### Environment

* Raspberry Pi Hardware Version: 3B+
* Screenly OSE Version: 0.18.2

vpetersson, 2019-11-20:
Most likely a broken SD card. Try swapping it out.

closed 2019-11-20 by vpetersson

didiatworkz, 2019-11-20:
I tried it with two different Pis and different SD Cards...

rusko124, 2019-11-21:
Hmm the same issue on Raspbian Buster.
Please follow the progress here.

At the moment I can suggest you to return to Raspbian Stretch.
I have no problem on the same SD.

rusko124, 2019-11-21:
@didiatworkz 
Try the next steps:
`sudo rm -rf /usr/local/lib/python2.7/dist-packages/OpenSSL/`(if we have the broken pip)
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`(Screenly merged some edits).

didiatworkz, 2019-11-21:
Hey @rusko124,

thanks - that solved the problem for me - But is it a python problem?