Screenly/Anthias #1275: URI of Asstes get´s encoded wrong

beemeeup, 2019-11-28:
#### Overview of the Issue
When adding a new asset, the URI gets encode wrong in the screenly.db.

http://monitoring.domain.com/client/check_mk/login.py?_username=admin&_password=12345678&_login=1

get´s encoded to:
http://monitoring.domain.com/client/check_mk/login.py?_username=admin&amp;amp;_password=12345678&amp;amp;__login=1

#### Reproduction Steps
If I add a new asset this behaviour repeats.

I have edited the URI Fiel via sqlite3 CLI and changed the URL to my first version by hand. This worked for me .

### Environment

* Raspberry Pi Hardware Version: 3B
* Screenly OSE Version: master@06acc42

vpetersson, 2019-11-28:
Very interesting. Good find @beemeeup

rusko124, 2019-11-28:
@vpetersson 
This is relevant after #1275
Any ideas how to avoid some? Explicitly set these characters? Not sure it sounds good

didiatworkz, 2019-12-05:
Same problem here :/

rusko124, 2019-12-05:
@vpetersson Could you reverse https://github.com/Screenly/screenly-ose/commit/4f422eae402e7a956736609ea3abc0c6deb5d6a5 and re-open #1254 until we find a better solution, please?

vpetersson, 2019-12-05:
@rusko124 should we really revert a security fix instead of fixing the issue?

didiatworkz, 2019-12-05:
Hi,
i think yes.. When it make product not usable... 
Or fix the security fix ;)

vpetersson, 2019-12-05:
I really don't like the idea of reverting the security fix.

didiatworkz, 2019-12-05:
Ok, but why do you escape URLs?
The problem is, that the GET Parameter have no function after that.

vpetersson, 2019-12-05:
I am pretty sure that it had something to do with UZBL or it's interface.

didiatworkz, 2019-12-05:
In my opinion, it's an interface problem. - It should also have priority number 1, because URL assets are currently not usable!
Today I put on seven players which all have the problem :(
So please don't do it first in Sprint 12 (upcoming) ;)

rasqal, 2019-12-19:
I don't know if this will add anything to the discussion but I too recently ran into this problem. We have two instances of Screenly at work, both running in identical 3B pies. The one I set up was running **production@6c2c2fd** while the other one was running **master@[some other number I do not remember]**, the one running the production branch worked fine, but the master branch did not. We tried to reinstall/flash the malfunctioning pie with the same production branch but it still resulted in the same error. Is this something to do with the hardware maybe? Or how's the progress on fixing this?

didiatworkz, 2019-12-22:
Hey @rasqal 
I solved it by taking the source code from the release:
https://github.com/Screenly/screenly-ose/archive/0.18.2.zip

Or swap the server.py file from this: https://github.com/Screenly/screenly-ose/blob/b9a2e96c16fe19325cfe5fede15d6a4dd8519e60/server.py

Merry X-Mas
didiatworkz

beemeeup, 2020-02-20:
I did an update in my screenly raspberry...
Problem is still there.

Will this be fixed?

abrenoch, 2020-03-18:
I'm running into this issue over here too #1308 

The current build is literally unusable for us because it turns ampersands in the URL into `&amp;` and then subsequent edits into `&amp;amp;`. The flask [documentation ](https://tedboy.github.io/flask/generated/flask.escape.html) clearly states this is to convert characters for displaying in HTML... So why is it saving the converted version rather than the original?

ealmonte32, 2020-03-18:
@abrenoch 
Thanks for pointing to this issue here as well.
Is this only in balena?
Why not also use the replace here ? Let me know if this only happens in balena so I can prepare a testing environment for it.

abrenoch, 2020-03-18:
@ealmonte32 though I have only tested it there, I don't see how this issue wouldn't effect all platforms. I tried reversing the changes in [this commit](https://github.com/Screenly/screenly-ose/commit/b699d1999c6a826224339868d1a8239ca93a44a5) and got the behavior I was expecting. I understand this was implemented to protect against xss attacks, but it seems to have broken some functionality.

Maybe a replacement of [reserved characters](https://tools.ietf.org/html/rfc3986#section-2.2) before performing the escape would protect against the buggy behavior we are seeing here?

beemeeup, 2020-05-17:
Will this be fixed? Kind of unusable nö.

ealmonte32, 2020-07-13:
Fix has been done for this.
I am doing PR now.

closed 2020-09-20 by stale[bot]

ealmonte32, 2020-09-20:
PR for this has been waiting for approval... https://github.com/Screenly/screenly-ose/pull/1344