Screenly/Anthias #1276: Raspberry Pi still to weak to View Youtube Camera streams?

hostheimer, 2019-12-10:
#### Overview of the Issue

More and more camera's viewing live streams via YouTube.
For example: https://www.youtube.com/watch?v=CByh5BIdgik

Is Screenly/Raspberry Pi 3B still  not able to view this is the performance of the newer Raspberry PI's still to low?
Is there maybe a work arround to view this kind of video streams?

#### Reproduction Steps

n/a

### Environment

* Screenly OSE Version: Sprint 10

vpetersson, 2019-12-10:
I don't think that's supported, unless YouTube produces a pure h264 (or h265 for pi4) video feed.

ealmonte32, 2020-01-15:
@hostheimer The RPI does seem to support it, see screenshot below, the problem is most likely the browser, I tested using the chromium-browser and RPi Model 3B+ Revision: 1.3:

Same pi running screenly, but for test environment I simply stop the screenly service and run the chromium-browser from terminal..

[image]

ealmonte32, 2020-01-15:
@vpetersson  I forgot to say, I am running into the character encoding issue? for youtube videos:

[image]

vpetersson, 2020-01-16:
Judging by the screenshot, that's a live stream. That will not work. We can only download non-streams (for obvious reasons). The only live streams that we can support are RTSP feeds (see [here](https://www.screenly.io/blog/2017/07/05/screenly-ose-0-10-is-out/)). To my knowledge, YouTube does not support this format.

closed 2020-01-23 by vpetersson